zigpy/zha-device-handlers #3847: [BUG] Tuya TRV _TZE204_rtrmfadk attribute running_state heating and idle are mixed up

toalchers, 2025-02-10:
### Bug description

`converter=lambda x: 0x01 if not x else 0x00,  # Heat, Idle`
https://github.com/prairiesnpr/zha-device-handlers/blob/8c3f178b9d210cf5b1c70969f1994306d7800078/zhaquirks/tuya/ts0601_trv.py#L1837

`[6, 'running_state', tuya.valueConverterBasic.lookup({heat: 1, idle: 0})],`
https://github.com/Koenkk/zigbee-herdsman-converters/blob/05865e983e39ce03578a6bfdb40099dffe041537/src/devices/tuya.ts#L5134

### Steps to reproduce

Nothing special

### Expected behavior

In heating - show heating, in idle - show idle.

### Device signature

### Diagnostic information

maciej-wilczynski, 2025-02-10:
I have the same issue with `_TZE200_0dvm9mva TS0601`. It shows heat while idle and idle while heating

mayraost, 2025-02-11:
Same here _TZE204_rtrmfadk HLK ist reversed

prairiesnpr, 2025-02-12:
@maciej-wilczynski Are you able to set the system mode on your `_TZE200_0dvm9mva`, does everything else work?

PR #3855 (2025-02-12): Invert running state, fix local temp cal for `_TZE204_rtrmfadk`
## Proposed change

Invert running state for `_TZE204_rtrmfadk`, use `tuya_number` for `local_temperature_calibration`.

Closes: #3847

## Additional information

## Checklist

- [ ] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works
closed 2025-02-12 by TheJulianJES