zigpy/zha-device-handlers #3850: [Device Support Request] Energy Meter automatic reclosing Tuya TS0601 _TZE204_fhvdgeuh

jairbj, 2025-02-11:
### Problem description

Trying to make it works. Not recognized by ZHA.

[image]

The device has already been supported by Z2M.

```
{
        fingerprint: tuya.fingerprint('TS0601', ['_TZE204_fhvdgeuh', '_TZE200_abatw3kj']),
        model: 'TS0601_din_4',
        vendor: 'Tuya',
        description: 'Din rail switch with power monitoring and threshold settings',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        configure: async (device, coordinatorEndpoint) => {
            await tuya.configureMagicPacket(device, coordinatorEndpoint);
            // Required to get the device to start reporting
            await device.getEndpoint(1).command('manuSpecificTuya', 'dataQuery', {});
        },
        whiteLabel: [tuya.whitelabel('RTX', 'TS0601_RTX_DIN', 'Din rail switch', ['_TZE200_abatw3kj'])],
        exposes: [
            e.switch().setAccess('state', ea.STATE_SET),
            e.power(),
            e.current(),
            e.voltage(),
            e.energy(),
            e.numeric('temperature', ea.STATE).withUnit('°C').withDescription('Current temperature'),
            e.numeric('leakage', ea.STATE).withUnit('mA').withDescription('Current leakage'),
        ],
        meta: {
            tuyaDatapoints: [
                [16, 'state', tuya.valueConverter.onOff],
                [1, 'energy', tuya.valueConverter.divideBy100], // Total forward energy
                [6, null, tuya.valueConverter.phaseVariant2], // Phase A voltage and current
                // [9, 'fault', tuya.valueConverter.raw], // no expose
                // [11, 'switch_prepayment', tuya.valueConverter.raw], // no expose
                // [12, 'clear_energy', tuya.valueConverter.raw], // no expose
                // [14, 'charge_energy', tuya.valueConverter.raw], // no expose
                [15, 'leakage', tuya.valueConverter.raw],
                // [102, 'reclosing_allowed_times', tuya.valueConverter.raw], // no expose
                [103, 'temperature', tuya.valueConverter.raw],
                // [104, 'reclosing_enable', tuya.valueConverter.raw], // no expose
                // [105, 'timer', tuya.valueConverter.raw], // no expose
                // [106, 'cycle_schedule', tuya.valueConverter.raw], // no expose
                // [107, 'reclose_recover_seconds', tuya.valueConverter.raw], // no expose
                // [108, 'random_timing', tuya.valueConverter.raw], // no expose
                // [109, 'switch_inching', tuya.valueConverter.raw], // no expose
                // [119, 'power_on_delay_power_on_time', tuya.valueConverter.raw], // no expose
                // [124, 'overcurrent_event_threshold_time', tuya.valueConverter.raw], // no expose
                // [125, 'time_threshold_of_lost_flow_event', tuya.valueConverter.raw], // no expose
                // [127, 'status', tuya.valueConverter.raw], // no expose
                // [134, 'relay_status_for_power_on', tuya.valueConverter.raw], // no expose
            ],
        },
    },
```

### Solution description

Device should have an on/off switch, energy and temperature reporting.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>
[... 740 lines cut ...]
</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

jairbj, 2025-02-16:
I made a partial quirk.

As this device reports 3 attributes in 1 data point I'm waiting for https://github.com/zigpy/zha-device-handlers/pull/3643 to be merged. Now it's missing voltage and power measurement.

``` python3
from zhaquirks.tuya.builder import TuyaQuirkBuilder
import zigpy.types as t
from zigpy.quirks.v2 import SensorDeviceClass, SensorStateClass
from zigpy.quirks.v2.homeassistant import UnitOfElectricCurrent, UnitOfEnergy, UnitOfPower, UnitOfElectricPotential

(
    TuyaQuirkBuilder("_TZE204_fhvdgeuh", "TS0601")
    .applies_to("_TZE204_fhvdgeuh", "TS0601")
    # Main switch
    .tuya_onoff(dp_id=16)
    # # Voltage
    # .tuya_sensor(
    #     dp_id=6,
    #     attribute_name="voltage",
    #     type=t.uint32_t,
    #     converter=lambda x: (x[1] | (x[0] <<8)) / 10,
    #     state_class=SensorStateClass.MEASUREMENT,
    #     device_class=SensorDeviceClass.VOLTAGE,
    #     unit=UnitOfElectricPotential.VOLT,
    #     translation_key="voltage",
    #     fallback_name="Voltage",
    # )
    # Current
    .tuya_sensor(
        dp_id=6,
        attribute_name="current",
        type=t.uint32_t,
        converter=lambda x: (x[4] | (x[3] << 8)) / 1000,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.CURRENT,
        unit=UnitOfElectricCurrent.AMPERE,
        translation_key="current",
        fallback_name="Current",
    )
    # # Power
    # .tuya_sensor(
    #     dp_id=6,
    #     attribute_name="power",
    #     type=t.uint32_t,
    #     converter=lambda x: x[7] | (x[6] << 8),
    #     state_class=SensorStateClass.MEASUREMENT,
    #     device_class=SensorDeviceClass.POWER,
    #     unit=UnitOfPower.WATT,
    #     translation_key="power",
    #     fallback_name="Power",
    # )
    # Energy
    .tuya_sensor(
        dp_id=1,
        attribute_name="energy_consumed",
        type=t.uint32_t,
        divisor=100,
        state_class=SensorStateClass.TOTAL_INCREASING,
        device_class=SensorDeviceClass.ENERGY,
        unit=UnitOfEnergy.KILO_WATT_HOUR,
        fallback_name="Total energy",
    )
    # Leakage measurement
    .tuya_sensor(
        dp_id=15,
        attribute_name="leakage",
        type=t.uint32_t,
        divisor=1,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.ENERGY,
        unit=UnitOfElectricCurrent.MILLIAMPERE,
        translation_key='power',
        fallback_name="Leakage",
    )
    # Temperature
    .tuya_temperature(dp_id=103)
    .skip_configuration()
    .add_to_registry()
)
```

asoliverez, 2025-03-02:
I tested the above quirk, and leakage and energy measurement works (it sums up correctly in the energy dashboard).

The one thing that doesn't work is the switch. The switch shows always off. When I switch it on, it does nothing. When I switch it if off, it does turn off (but then I have to go and manually turn it on again

jairbj, 2025-03-02:
> I tested the above quirk, and leakage and energy measurement works (it sums up correctly in the energy dashboard).
>
> The one thing that doesn't work is the switch. The switch shows always off. When I [...]

I tested it few minutes ago and to me it worked. I turned it on and off from homeassistanta and when turned on/off manually it sync the new state to homeassistant.

asoliverez, 2025-03-07:
> > I tested the above quirk, and leakage and energy measurement works (it sums up correctly in the energy dashboard).
> > The one thing that doesn't work is the switch. The switch shows always off. When I switch it on, it does nothing. When I switch it if off, it does turn off (but then I have to go and manually turn it on again
> 
> I tested it few minutes ago and to me it worked. I turned it on and off from homeassistant and when turned on/off manually it synced the new state to homeassistant.

You're right, the on switch works. I think last time, my UPS didn't come up and the HA server went down when I switched the mains off. hehe

So, the only thing missing then is the initial sync of the on/off status. Any idea how that might be done?

asoliverez, 2025-03-22:
I found a small issue, leakage should of type Current, not energy

```
    # Leakage measurement
    .tuya_sensor(
        dp_id=15,
        attribute_name="leakage",
        type=t.uint32_t,
        divisor=1,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.CURRENT,
        unit=UnitOfElectricCurrent.MILLIAMPERE,
        translation_key='current',
        fallback_name="Leakage",
    )
```

creepyhauz, 2025-05-22:
"""Tuya Power Meter."""

from collections.abc import ByteString

from zigpy.quirks.v2 import SensorDeviceClass, SensorStateClass
from zigpy.quirks.v2.homeassistant import (
    UnitOfElectricCurrent,
    UnitOfEnergy,
    UnitOfPower,
)
import zigpy.types as t
from zigpy.zcl.clusters.general import LevelControl, OnOff
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement

from zhaquirks.tuya import DPToAttributeMapping, TuyaLocalCluster
from zhaquirks.tuya.builder import TuyaQuirkBuilder

def dp_to_power(data: ByteString) -> int:
    """Convert DP data to power value."""
    # From https://github.com/Koenkk/zigbee2mqtt/issues/18603#issuecomment-2277697295
    power = int(data)
    if power > 0x0FFFFFFF:
        power = (0x1999999C - power) * -1
    return power

def multi_dp_to_power(data: ByteString) -> int:
    """Convert DP data to power value."""
    # Support negative power readings
    # From https://github.com/Koenkk/zigbee2mqtt/issues/18603#issuecomment-2277697295
    power = data[7] | (data[6] << 8)
    if power > 0x7FFF:
        power = (0x999A - power) * -1
    return power

def multi_dp_to_current(data: ByteString) -> int:
    """Convert DP data to current value."""
    return data[4] | (data[3] << 8)

def multi_dp_to_voltage(data: ByteString) -> int:
    """Convert DP data to voltage value."""
    return data[1] | (data[0] << 8)

class Tuya3PhaseElectricalMeasurement(ElectricalMeasurement, TuyaLocalCluster):
    """Tuya Electrical Measurement cluster."""

    _CONSTANT_ATTRIBUTES = {
        ElectricalMeasurement.AttributeDefs.ac_current_multiplier.id: 1,
        ElectricalMeasurement.AttributeDefs.ac_current_divisor.id: 1000,
        ElectricalMeasurement.AttributeDefs.ac_voltage_multiplier: 1,
        ElectricalMeasurement.AttributeDefs.ac_voltage_divisor.id: 10,
    }

(
    TuyaQuirkBuilder("_TZE204_fhvdgeuh", "TS0601")
    .tuya_temperature(dp_id=133, scale=10)
    .tuya_onoff(dp_id=16)
    # Energy
    .tuya_sensor(
        dp_id=1,
        attribute_name="energy",
        type=t.int32s,
        divisor=100,
        state_class=SensorStateClass.TOTAL,
        device_class=SensorDeviceClass.ENERGY,
        unit=UnitOfEnergy.KILO_WATT_HOUR,
        fallback_name="Total energy",
    )
    .tuya_sensor(
        dp_id=9,
        attribute_name="power",
        type=t.int32s,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.POWER,
        unit=UnitOfPower.WATT,
        fallback_name="Total power",
        converter=dp_to_power,
    )
    .tuya_dp_multi(
        dp_id=6,
        attribute_mapping=[
            DPToAttributeMapping(
                ep_attribute=Tuya3PhaseElectricalMeasurement.ep_attribute,
                attribute_name="active_power",
                converter=multi_dp_to_power,
            ),
            DPToAttributeMapping(
                ep_attribute=Tuya3PhaseElectricalMeasurement.ep_attribute,
                attribute_name="rms_voltage",
                converter=multi_dp_to_voltage,
            ),
            DPToAttributeMapping(
                ep_attribute=Tuya3PhaseElectricalMeasurement.ep_attribute,
                attribute_name="rms_current",
                converter=multi_dp_to_current,
            ),
        ],
    )
    
    .adds(Tuya3PhaseElectricalMeasurement)
    .removes(LevelControl.cluster_id)
    .skip_configuration()
    .add_to_registry()
)

franortiz, 2025-07-24:
Here you can find my [WIP quirk gist](https://gist.github.com/franortiz/f0253489098c2d5b1ccd89de661202d9) 
Need to copmlete the over/under mapping settings

Has some annotations on missing DPs, please, review and comment