zigpy/zha-device-handlers #3854: [Device Support Request] Parasoll Ikea door/window sensor

FrankTub, 2025-02-12:
### Problem description

When trying to start using my [Parasoll Ikea door/window sensor](https://www.ikea.com/nl/nl/p/parasoll-deur-raamsensor-smart-wit-80504308/) I experience irregular responses from them.

It is hard to pair them with my Sonoff zigbee dongle in HA, while other zigbee devices are handled correctly (even if they are very close to the door sensor).  Even if I succesfully paired them following [these instructions](https://community.home-assistant.io/t/ikea-parasoll-vallhorn-devices-not-working-zha/668852/54) these sensors are not reliable. After some time they stop reporting their status and become unavailable. Why this happens is still a black box to me.

Given this HA community [thread](https://community.home-assistant.io/t/ikea-parasoll-vallhorn-devices-not-working-zha/668852) I'm sure a lot of people would benefit from making sure that this sensor could properly communicate with HA.

### Solution description

I'm not completely sure if this will work but I guess it would make sense to make a custom quirk for the parasoll sensor so that it matches zigbee standards.

### Screenshots/Video

N/A

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4476,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0103",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0b05",
        "0x1000",
        "0xfc7c",
        "0xfc81"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0019",
        "0x1000"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0402",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0500"
      ],
      "output_clusters": [
        "0x0003"
      ]
[... 38 lines cut ...]
[36m2025-02-12 13:18:03.935 DEBUG (MainThread) [zha.application.discovery] 'sensor' platform -> 'LQISensor' using ['basic'][0m
[36m2025-02-12 13:18:03.935 DEBUG (MainThread) [zha.application.discovery] Discovering entities for device: d4:48:67:ff:fe:d5:a9:0d-IKEA of Sweden PARASOLL Door/Window Sensor[0m
[36m2025-02-12 13:18:03.935 DEBUG (MainThread) [zha.application.discovery] Attempting to discover quirks v2 entities for device: d4:48:67:ff:fe:d5:a9:0d-IKEA of Sweden PARASOLL Door/Window Sensor[0m
[36m2025-02-12 13:18:03.935 DEBUG (MainThread) [zha.application.discovery] Device: d4:48:67:ff:fe:d5:a9:0d-IKEA of Sweden PARASOLL Door/Window Sensor is not a quirks v2 device - skipping discover_quirks_v2_entities[0m
[36m2025-02-12 13:18:03.935 DEBUG (MainThread) [zha.application.gateway] [0x0189](IKEA of Sweden PARASOLL Door/Window Sensor) restored as 'available', last seen: 0:01:13 ago, consider_unavailable_time: 21600 seconds[0m
[36m2025-02-12 13:18:03.938 DEBUG (MainThread) [zha.application.gateway] Platform entity data: BinarySensorEntityInfo(fallback_name=None, unique_id='d4:48:67:ff:fe:d5:a9:0d-2-1280', platform=<Platform.BINARY_SENSOR: 'binary_sensor'>, class_name='IASZone', translation_key=None, device_class=<BinarySensorDeviceClass.OPENING: 'opening'>, state_class=None, entity_category=None, entity_registry_enabled_default=True, enabled=True, cluster_handlers=[ClusterHandlerInfo(class_name='IASZoneClusterHandler', generic_id='cluster_handler_0x0500', endpoint_id=2, cluster=ClusterInfo(id=1280, name='IAS Zone', type='server', commands=[ZCLCommandDef(id=0x00, name='enroll_response', direction=<Direction.Server_to_Client: 1>, schema=<class 'zigpy.zcl.foundation.enroll_response'>, is_manufacturer_specific=None), ZCLCommandDef(id=0x01, name='init_normal_op_mode', direction=<Direction.Client_to_Server: 0>, schema=<class 'zigpy.zcl.foundation.init_normal_op_mode'>, is_manufacturer_specific=None), ZCLCommandDef(id=0x02, name='init_test_mode', direction=<Direction.Client_to_Server: 0>, schema=<class 'zigpy.zcl.foundation.init_test_mode'>, is_manufacturer_specific=None)]), id='2:0x0500', unique_id='d4:48:67:ff:fe:d5:a9:0d:2:0x0500', status='CREATED', value_attribute=None)], device_ieee=d4:48:67:ff:fe:d5:a9:0d, endpoint_id=2, available=True, group_id=None, attribute_name='zone_status')[0m
[36m2025-02-12 13:18:03.942 DEBUG (MainThread) [zha.application.gateway] Platform entity data: CommandButtonEntityInfo(fallback_name=None, unique_id='d4:48:67:ff:fe:d5:a9:0d-1-3', platform=<Platform.BUTTON: 'button'>, class_name='IdentifyButton', translation_key=None, device_class=<ButtonDeviceClass.IDENTIFY: 'identify'>, state_class=None, entity_category=<EntityCategory.DIAGNOSTIC: 'diagnostic'>, entity_registry_enabled_default=True, enabled=True, cluster_handlers=[ClusterHandlerInfo(class_name='IdentifyClusterHandler', generic_id='cluster_handler_0x0003', endpoint_id=1, cluster=ClusterInfo(id=3, name='Identify', type='server', commands=[ZCLCommandDef(id=0x00, name='identify', direction=<Direction.Client_to_Server: 0>, schema=<class 'zigpy.zcl.foundation.identify'>, is_manufacturer_specific=None), ZCLCommandDef(id=0x01, name='identify_query', direction=<Direction.Client_to_Server: 0>, schema=<class 'zigpy.zcl.foundation.identify_query'>, is_manufacturer_specific=None), ZCLCommandDef(id=0x40, name='trigger_effect', direction=<Direction.Client_to_Server: 0>, schema=<class 'zigpy.zcl.foundation.trigger_effect'>, is_manufacturer_specific=None)]), id='1:0x0003', unique_id='d4:48:67:ff:fe:d5:a9:0d:1:0x0003', status='CREATED', value_attribute=None)], device_ieee=d4:48:67:ff:fe:d5:a9:0d, endpoint_id=1, available=True, group_id=None, command='identify', args=[5], kwargs={})[0m
[36m2025-02-12 13:18:03.966 DEBUG (MainThread) [zha.application.gateway] Platform entity data: SensorEntityInfo(fallback_name=None, unique_id='d4:48:67:ff:fe:d5:a9:0d-1-1', platform=<Platform.SENSOR: 'sensor'>, class_name='Battery', translation_key=None, device_class=<SensorDeviceClass.BATTERY: 'battery'>, state_class=<SensorStateClass.MEASUREMENT: 'measurement'>, entity_category=<EntityCategory.DIAGNOSTIC: 'diagnostic'>, entity_registry_enabled_default=True, enabled=True, cluster_handlers=[ClusterHandlerInfo(class_name='PowerConfigurationClusterHandler', generic_id='cluster_handler_0x0001', endpoint_id=1, cluster=ClusterInfo(id=1, name='Power Configuration', type='server', commands=[]), id='1:0x0001', unique_id='d4:48:67:ff:fe:d5:a9:0d:1:0x0001', status='CREATED', value_attribute='battery_voltage')], device_ieee=d4:48:67:ff:fe:d5:a9:0d, endpoint_id=1, available=True, group_id=None, attribute='battery_percentage_remaining', decimals=1, divisor=1, multiplier=1, unit='%')[0m
[36m2025-02-12 13:18:03.966 DEBUG (MainThread) [zha.application.gateway] Platform entity data: SensorEntityInfo(fallback_name=None, unique_id='d4:48:67:ff:fe:d5:a9:0d-1-0-rssi', platform=<Platform.SENSOR: 'sensor'>, class_name='RSSISensor', translation_key='rssi', device_class=<SensorDeviceClass.SIGNAL_STRENGTH: 'signal_strength'>, state_class=<SensorStateClass.MEASUREMENT: 'measurement'>, entity_category=<EntityCategory.DIAGNOSTIC: 'diagnostic'>, entity_registry_enabled_default=False, enabled=True, cluster_handlers=[ClusterHandlerInfo(class_name='BasicClusterHandler', generic_id='cluster_handler_0x0000', endpoint_id=1, cluster=ClusterInfo(id=0, name='Basic', type='server', commands=[ZCLCommandDef(id=0x00, name='reset_fact_default', direction=<Direction.Client_to_Server: 0>, schema=<class 'zigpy.zcl.foundation.reset_fact_default'>, is_manufacturer_specific=None)]), id='1:0x0000', unique_id='d4:48:67:ff:fe:d5:a9:0d:1:0x0000', status='CREATED', value_attribute=None)], device_ieee=d4:48:67:ff:fe:d5:a9:0d, endpoint_id=1, available=True, group_id=None, attribute=None, decimals=1, divisor=1, multiplier=1, unit='dBm')[0m
[36m2025-02-12 13:18:03.966 DEBUG (MainThread) [zha.application.gateway] Platform entity data: SensorEntityInfo(fallback_name=None, unique_id='d4:48:67:ff:fe:d5:a9:0d-1-0-lqi', platform=<Platform.SENSOR: 'sensor'>, class_name='LQISensor', translation_key='lqi', device_class=None, state_class=<SensorStateClass.MEASUREMENT: 'measurement'>, entity_category=<EntityCategory.DIAGNOSTIC: 'diagn [... cut ...]
[36m2025-02-12 13:19:15.424 DEBUG (MainThread) [zha] Emitting event cluster_handler_attribute_updated with data ClusterAttributeUpdatedEvent(attribute_id=2, attribute_name='zone_status', attribute_value=<ZoneStatus: 0>, cluster_handler_unique_id='d4:48:67:ff:fe:d5:a9:0d:2:0x0500', cluster_id=1280, event_type='cluster_handler_event', event='cluster_handler_attribute_updated') (1 listeners)[0m
```

</details>

### Custom quirk

Not sure where to start but if pointed to the right location I would be open for contributing it myself

FrankTub, 2025-02-12:
What I noticed in the diagnostics is that it appears that the on/off status is communicated in endpoint_attribute `0x0006` and more specific in attribute `0x0000`. 

See:

```json
          "0x0006": {
            "endpoint_attribute": "on_off",
            "attributes": {
...
              "0x0000": {
                "attribute": "ZCLAttributeDef(id=0x0000, name='on_off', type=<enum 'Bool'>, zcl_type=<DataTypeId.bool_: 16>, access=<ZCLAttributeAccess.Read|Report|Scene: 25>, mandatory=True, is_manufacturer_specific=False)",
                "value": null
              },
```

Note sure if it helps but ChatGPT created for it. 

```py
"""Custom device handler for the IKEA PARASOLL Door/Window Sensor."""

import logging

from zhaquirks import CustomDevice
from zigpy.profiles import zha
from zigpy.zcl.clusters.general import Basic, Identify, OnOff

_LOGGER = logging.getLogger(__name__)

class CustomOnOffCluster(OnOff):
    """Custom On/Off cluster to interpret the door sensor's status."""
    
    def attribute_updated(self, attrid, value):
        _LOGGER.debug("CustomOnOffCluster: attribute 0x%04x updated with value %s", attrid, value)
        if attrid == 0x0000:  # The on_off attribute
            # Interpret the Boolean value as door status.
            # Adjust the mapping if needed. Here we assume:
            #   True  => door is open
            #   False => door is closed
            if value:
                _LOGGER.debug("Door sensor reports: OPEN")
                # Here you might want to call a method to update HA state,
                # e.g., self.endpoint.device.handle_door_opened()
            else:
                _LOGGER.debug("Door sensor reports: CLOSED")
                # Similarly, you might call: self.endpoint.device.handle_door_closed()
        # Always call the superclass to ensure any built‐in behavior is preserved.
        super().attribute_updated(attrid, value)

class ParasollDoorSensor(CustomDevice):
    """Custom device handler for IKEA PARASOLL Door/Window Sensor using On/Off cluster."""
    
    # The signature below should match what the device reports.
    # You can adjust the manufacturer, model, device_type, and clusters based on your logs.
    signature = {
        1: {
            "manufacturer": "IKEA of Sweden",
            "model": "PARASOLL Door/Window Sensor",
            "profile_id": zha.PROFILE_ID,
            "device_type": 0x0105,  # Example device type; use the correct value from your logs
            "input_clusters": [
                Basic.cluster_id,
                Identify.cluster_id,
                OnOff.cluster_id  # 0x0006: This is the On/Off cluster reporting the door state
            ],
            "output_clusters": [],
        },
        # If your device has additional endpoints that you want to ignore,
        # include them in the signature so that the device matches, but omit them in the replacement.
        2: {
            # This endpoint may be reporting other data that you don't need.
            "profile_id": zha.PROFILE_ID,
            "device_type": 0x0101,  # Example; adjust as needed
            "input_clusters": [OnOff.cluster_id],
            "output_clusters": [],
        },
    }

    replacement = {
        1: {
            "manufacturer": "IKEA of Sweden",
            "model": "PARASOLL Door/Window Sensor",
            "profile_id": zha.PROFILE_ID,
            "device_type": 0x0105,
            "input_clusters": [
                Basic,
                Identify,
                CustomOnOffCluster,  # Use the custom cluster in place of the standard On/Off
            ],
            "output_clusters": [],
        },
        # Endpoint 2 is omitted so it will be ignored by ZHA.
    }

```

MattWestb, 2025-02-12:
You dont need doing some magic things only have one network that is working well.
I suppose you have one TI coordinator and its not one good Zigbee 3 router so try paring it to one good Zigbee 3 router and hope its staying on it.
The device is relaying on Zigbee 3 pull control is working in its parent and if not the network can getting problem sending the commands is using.
The OnOff cluster is sending light commands that can being used for truing light on / Off.
The ISAZone is used and intrusion (that you see in ZHA) but need being registered OK for working and also updating the status of the zone (it can being missing updates if changing status very fast).
If its parent router or one in the path to the coordinator is not relaying the commands OK you is getting problems and parasol is working very good if having one stable Zigbee 3 network.

MattWestb, 2025-02-12:
By the was is your controller doing checkins every 50 minutes ?

[image]

FrankTub, 2025-02-12:
@MattWestb perhaps my issue is confusing. Maybe my issue suggests that I'm unable to get the parasoll to work in zha, but that is indeed not the case. So let me retry;

I have 30 zigbee devices paired with my `Sonoff Zigbee 3.0 USB Dongle Plus`, I have the [E version](https://www.bol.com/nl/nl/p/sonoff-zbdongle-e-zigbee-3-0-usb-dongle/9300000124030418/?Referrer=ENTcli_order_confirmation_c1a36889-9b55-49df-a257-27494535a28c). The firmware in HA is shown as `Firmware: 6.10.3.0 build 297`.  I use a raspberry pi 5 and have an usb 2.0 extension cable plugged into my usb 2.0 port and that part is connect to the dongle If that is not good enough I'm still open for suggestions, but with some googling this should work for almost all zigbee devices.. Of those 30 devices, only the Parasolls are instable. Even in the same rooms where I experience the issues with the parasolls I have literally 3 other zigbee devices (withing 2 meter range from the dongle that all work without any issues. I have devices that are at least 8 meter from the dongle and 2 thick walls in between and that all works great. The pain with these parasolls started the moment you try to add them to zha, for all my other devices that works perfectly fine. But for the parasolls I have to go through the [following steps](https://community.home-assistant.io/t/ikea-parasoll-vallhorn-devices-not-working-zha/668852/54), the highlights:

1) Get a parasoll into pairing mode by pressing the button at the back 4 times in 5s and add device through zha.
1) Quite quickly it shows `Interview complete > Configuring`. It stays in this forever and won't continue.
1) Next you will have to reload/restart the zha integration and then it shows a new device. The device often lacks attributes like `Firmware` or `Battery`. 
1) If you are lucky it will add it after some random time, but it also might stay like this for at least weeks.

This in itself is something I would like to report and if possible have an expert look at why this is happening. I would guess the most relevant lines in my log would be:

``` 
[36m2025-02-12 13:18:03.973 DEBUG (MainThread) [zha.application.gateway] Platform entity data: UpdateEntityInfo(fallback_name=None, unique_id='d4:48:67:ff:fe:d5:a9:0d-1-25-firmware_update', platform=<Platform.UPDATE: 'update'>, class_name='FirmwareUpdateEntity', translation_key=None, device_class=<UpdateDeviceClass.FIRMWARE: 'firmware'>, state_class=None, entity_category=<EntityCategory.CONFIG: 'config'>, entity_registry_enabled_default=True, enabled=True, cluster_handlers=[ClusterHandlerInfo(class_name='OtaClientClusterHandler', generic_id='cluster_handler_0x0019', endpoint_id=1, cluster=ClusterInfo(id=25, name='Ota', type='client', commands=[ZCLCommandDef(id=0x03, name='image_block', direction=<Direction.Client_to_Server: 0>, schema=<class 'zigpy.zcl.clusters.general.ImageBlockCommand'>, is_manufacturer_specific=None), ZCLCommandDef(id=0x04, name='image_page', direction=<Direction.Client_to_Server: 0>, schema=<class 'zigpy.zcl.clusters.general.ImagePageCommand'>, is_manufacturer_specific=None), ZCLCommandDef(id=0x01, name='query_next_image', direction=<Direction.Client_to_Server: 0>, schema=<class 'zigpy.zcl.clusters.general.QueryNextImageCommand'>, is_manufacturer_specific=None), ZCLCommandDef(id=0x08, name='query_specific_file', direction=<Direction.Client_to_Server: 0>, schema=<class 'zigpy.zcl.foundation.query_specific_file'>, is_manufacturer_specific=None), ZCLCommandDef(id=0x06, name='upgrade_end', direction=<Direction.Client_to_Server: 0>, schema=<class 'zigpy.zcl.foundation.upgrade_end'>, is_manufacturer_specific=None)]), id='1:0x0019', unique_id='d4:48:67:ff:fe:d5:a9:0d:1:0x0019', status='CREATED', value_attribute=None)], device_ieee=d4:48:67:ff:fe:d5:a9:0d, endpoint_id=1, available=True, group_id=None, supported_features=<UpdateEntityFeature.INSTALL|SPECIFIC_VERSION|PROGRESS: 7>)[0m
[36m2025-02-12 13:19:05.211 DEBUG (MainThread) [bellows.ezsp.protocol] Sending command  getExtendedTimeout: () {'remoteEui64': d4:48:67:ff:fe:d5:a9:0d}[0m
[36m2025-02-12 13:19:15.415 DEBUG (MainThread) [zha] Emitting event cluster_handler_attribute_updated with data ClusterAttributeUpdatedEvent(attribute_id=2, attribute_name='zone_status', attribute_value=<ZoneStatus: 0>, cluster_handler_unique_id='d4:48:67:ff:fe:d5:a9:0d:2:0x0500', cluster_id=1280, event_type='cluster_handler_event', event='cluster_handler_attribute_updated') (1 listeners)[0m
```

This is the period when I restart the ZHA integration. But based on this log I still have no clue why it is not added like other devices in the first place.

After the pairing the sensor works as it should. I can create automations based on Closed/Open state of the device and that is all great. But then after couple of weeks some devices just stops randomly reporting its state. Again, this only happens for the parasoll devices. 

Hope that makes my issue a lot clearer, if not just let me know.

> By the was is your controller doing checkins every 50 minutes ?

Currently it is. But can't trace my logbook far enough back to see what was going on just before one of my devices stopped working.

FrankTub, 2025-02-12:
But isn't there some way to make the pairing of these ikea devices more hufterproof in the sense that anyone without an IT background should be able to do it?

MattWestb, 2025-02-12:
Normal its only putting the system in paring mode and then the device and if possible waking up little if its taking long time and all is working well.
If having not good routers try paring thru one good by from its device card three dots and "add device via this device" then the system is only sending open for pairing to that router.

I also having:
```
zha:
  custom_quirks_path: /config/custom_zha_quirks/
  zigpy_config:
    source_routing: false
    ezsp_config:
      CONFIG_MAX_END_DEVICE_CHILDREN: 0
```
That making EZSP not using source routing and the coordinator is refusing having end devices so they must using other routers = less problems.

> hufterproof 

Its not easy but all systems have there "personality" and some combinations is better other worse and no is having all good and no bad sides.
ZHA with ZESP is normally very good combination but one Zigbee network need good routers and also low radio interference for working well and one bad device can killing one well working network (add one old OSRAM plug and you have all problems you can imaged of) completely !!!

FrankTub, 2025-02-13:
Okay thanks for your input. Any recommendations on dedicated zigbee routers?

And just for my understanding, are you also saying that parasoll devices is causing my zigbee network to perform worse? If so, I would like to know and next time just buy a better device 😄

MattWestb, 2025-02-13:
Normally all mains powered Zigbee devices (not all, some tuya and aqara is not) is routers. Then you have end devices that can being real sleepers with radio of 99.9% of time and with radio on (IKEA blinds is end device with radio on then idle).
I cant see and use of dictated router then they is normally expensive and not working well.
I was building my backbone of me production network with IKEA outlet in every room and then putting Aqara weather sensors in every room (they is very picky and never jumping to on new parent they leaving silent). Use god Zigbee 3 router devices shall being OK but look that they is certificated (not only zigbee logo on the box/device).
I think IKEA tretakt is one good aspirant for 7€ / peace but you can finding other but not for that low price and easy mounting. 
I dont have real network problems with IKEA controllers but some time Badring and Prasoll can draining battery fast but i think its handshaking with its parent that making it and putting in new battery normally fixing it (i still have some not very good tuya routers).

nraeth, 2025-02-14:
I operate some Parasoll sensors without issues on my network with a Sonoff Zigbee 3.0 USB Dongle Plus P.

I need to set the timeout for battery powered devices to a much higher value, because otherwise HA treats the sensors as unavaiable much too soon.

Keep in mind, that door sensors usually only report there state when they are triggered.

FrankTub, 2025-02-14:
@nraeth in which way do you modify the timeout settings these sensors?

nraeth, 2025-02-14:
> in which way do you modify the timeout settings these sensors?

In the ZHA global network settings there are timeouts for marking as unavaiable for mains powered and battery powered devices.

FrankTub, 2025-02-16:
@MattWestb , one of my parasoll devices has issues with reporting it's state. One of the thing I noted was that this sensor `Checkin event was fired` does not have the event in the logbook. I have two parasoll sensors that are relatively close to it and they have that event and have been up and running for at least 2 weeks. While for this device where the checking event is missing it is like the third time it has become irregular reporting it's state.

Any suggestions on how to solve that?

MattWestb, 2025-02-16:
If you is not getting checkins ever 55 minutes its the device not OK configured or the network is losing some message from the device. 
First try do one reconfigure from the device card and waling up the device so its not sleeping then you is sending things to it.
IKEA controllers can being little tricky if they is not OK configured and some time its needed taking out the battery and putting it back and its working and some time restart of HA can helping or more try and error,
As soon getting reconfigure all OK (green for all cluster its configuring) it shall doing the checking OK.
But Parasoll, Badring and Vallhor also need registering one IAS Zone that also can having problems but normally its being fixed with battery in and out and waiting little (IAS Zone is little different and its working like one Ethernet multicast so more device can using the same channel but must registration OK).

Also if the problematic device is having one bad router as parent (ole OSRAM plug) that is not forwarding the commands from its chides you must trying getting it using one good parent (best only having good routers).

FrankTub, 2025-03-02:
@MattWestb I finally tried your suggestion:

```yml
zha:
  zigpy_config:
    source_routing: false
    ezsp_config:
      CONFIG_MAX_END_DEVICE_CHILDREN: 0
```

And I still don't really understand the end result. See below my topology. All my end devices that don't act as routers are still connected, however the ones that do act as router have become unavailable. 

[image]

Is that expected behavior of this configuration? Or am I missing something now?

MattWestb, 2025-03-02:
That is little strange but the visualization is not updates its links so oft and routers have faster time for being flagged offline then end devices so you can being between both but i dont knowing. Also i think the map is not updated then many end devices is connected to the coordinator.

If setting up one new system then need adding at least one router before blocking end devices or no router can being added but you shall not have that problem then have many working routers.

Try if some lights or plugs (device that cab changing state) is reacting on commands and if not try repower the device and see if its coming online (if link counters is out of sync then routers trying finding new routs and updating the network ( self healing).

That worrying me is that you looks having very low signal quality on nearly all links in your network (out grayed or red / orange links shall being green or orange for being OK).
My test network for IKEA controller is all  green / orange and have lowest LQI of 194 for the moment (normally always over 95).

[image]

FrankTub, 2025-03-02:
> Try if some lights or plugs (device that cab changing state) is reacting on commands and if not try repower the device and see if its coming online (if link counters is out of sync then routers trying finding new routs and updating the network ( self healing).

They are unavailable and thus not reacting to commands. For now removed it completely out of my config and even though things are back to working again as it was, many of my links still remain greyed out or red. What is interesting to me is that even the end devices that react very well and are very close to the controller are greyed out in this vis. My current graph:

[image]

But what I can't seem to figure out is that even the one that is literally less than 0,5 meter in the same room away from the controller is shown as greyed out:

[image]

Just double checked and turning it on and off is literally instant. So I'm having a hard time understanding how I should interpret this visualization.

MattWestb, 2025-03-02:
The system is looking for all communication and setting "last seen" for the device in the system and if the time is "running out" for the device its flagged offline (the time is set  in the configure zigbee / network tab).

The visualization / map is using network scans (pulling the neighbor table in the routers so getting the links and quality of then) that the system is doing in the background and some links can being some hours old.

Your out grayed Sonoff have communicating but its parent (the coordinator) was having it as child then the network scan was done but very likely it was one old record.
One bad thing for one end device having the coordinator as parent (technically its one normal router with trust center for manage the network security) is that its doing more things then routing and then children is getting problems it not repositioning on pulls and command frames its trying sending and is its not getting one ack its shall (some device dont do) looking for one better parent and using that. I was going on Holiday for 3 weeks to Spain and was power down 2 test zigbee networks and coming home all was online in one hour and no lost end devices then they was not having the missing coordinator as parent only real good routers and if having the coordinator as parent and its offline (updating HA next week = ZHA is going off line) theny trying jumping and if not  working they is leaving the network = bad for short or long un/planed coordinator outages for the network.

So the visualization is **not** one picture how its now its how it was for some time ago (can being some minutes or hours) and can being times for different routers. Let the network setting for some hours and the looking on the visualization and you shall having one more realistic view.

If you is not having very long distance between your routers and normal surroundings you shall heaving OK links between all (green, Orange or yellow) between all routers and combinations of them = working mesh network and only if it very long between routers or mounted very badly (in earthed metal casing and so on).
Wot brand / model is your routers in your network then its looks they is not having antennas at  all or somthing else is very wrong ?

FrankTub, 2025-03-03:
I left it for some time and just checked again. And the strangest thing to me is still the behavior the same end devices are giving me. So in picture below you see `Schakelaar Gang`, which is like 2 meters away from the central controller choosing a direct connection to the controller with a poor connection. But `Schakelaar Trapkast` is something like 30 cm away from the central controller and chooses to connect through `Kantoor docking station voeding`, this device is a little bit further away to the controller (2,5 m or something) but has to go to a single wall. Now I'm so clueless why `Schakelaar Trapkast` does not prefer a direct connection to the controller, while for `Schakelaar Gang` it might make more sense to connect to the router `Kantoor docking station voeding`.

[image]

In the example above I use a mmWave sensor [SNZB06P](https://sonoff.tech/product/gateway-and-sensors/snzb-06p/) with firmware `0x00001006` with quirk `zhaquirks.sonoff.snzb06p.SonoffPresenceSenorSNZB06P`. For the rest I'm mostly using some smart plugs. Four of them are `Tuya Smart Plug TS011F _TZ3000_5f43h46b` with quirk `zhaquirks.tuya.ts011f_plug.Plug_v2` (I wasn't aware I had bought tuya shit..). I also have some [ZB-SW01 by eWeLink](https://www.zigbee2mqtt.io/devices/ZB-SW01.html) that does not have a quirk and is not handled as switch but as a light. And the other ones I have are two [S26R2ZB by SONOFF](https://www.sonoff.nl/a-68364640/zigbee-producten/sonoff-s26-r2zb-zigbee-wandcontactdoos/#description). Perhaps the problem is that the ones close to my controller are all shitty ass routers?

MattWestb, 2025-03-03:
You still have **very bad** signal / Link Quality between nearly all devices.
Look in the Zigbee configuration witch channel is on and then use one mobile and installing one WiFI scanner app and look with channels is in use and beware of Zigbee and WiFi is using the same frequencies but have different channel numbering (see Zigpy wiki).

In RF (Radio Frequencies) one device can being near one other and have very high signal strings but is getting interference from other transmitters then you is getting very low LQI (Link Quality). Also reflections can facing out nearly 100% of one sent signal to the receiver and moving one device some mm or turning the antenna some ° in one other direction can helping or making thing worse. But then only some devices have problem but your network looks having very bad RF environment for the moment.

Little "Zigbee things" one device can finding one link with very good signal strengths than its near the other device not being good and using one with lower strengths but better quality but also every hope is giving penalties then its making delays but its one part of selecting the best way in the mesh network.
I dont going in the problematic with asymmetric RF links its out of your problem and normally eZSP coordinators is not having that problem (as deconz is having) and in the end is not getting one mesh network but one star with on single point of failure.

FrankTub, 2025-03-05:
I really appreciate all your help so far @MattWestb ! Also tried your suggestion to do a wifi scan and you see that the strongest signals are on wifi channels 1 and 6. And my zigbee channel is on channel 25, which makes sense than I guess.

I double checked and I have a philips Hue bridge with some bulbs connected and this one is also on channel 25.

I downloaded the diagnostics of ZHA several times today and looked at `energy_scan`. There it however appears that channel 25 is often very busy. So I reasoned it makes sense for me to try channel 20? Example given:

```json
    "energy_scan": {
      "11": 3.6632469452765037,
      "12": 88.70042934643088,
      "13": 96.64469941013013,
      "14": 95.69133648577223,
      "15": 88.70042934643088,
      "16": 36.830390267097734,
      "17": 89.93931580241996,
      "18": 88.70042934643088,
      "19": 25.74050169409602,
      "20": 6.011489450827149,
      "21": 59.15797905332195,
      "22": 68.14622793558128,
      "23": 55.9836862725909,
      "24": 6.789392891308996,
      "25": 92.0598007161209,
      "26": 94.48255331375627
    },
```

I did so and most of my devices kept working, but my closest parasoll device stopped working. To get a little back on track on the why I raised the initial issue is that I saw some behavior that still made no sense to me.

So I removed the device from ZHA and added the device again following [these instructions](https://community.home-assistant.io/t/ikea-parasoll-vallhorn-devices-not-working-zha/668852/54). Now the thing is that directly after pairing the device it does not properly report its state back to ZHA. In print screen below you can see that after pairing it, the state was only `Closed`.

[image].

However looking at the Logbook you can see I opened and closed the door several times:

[image]

The strange thing is that looking at the logbook in Home Assistant side tab it does not show all this activity:

[image]

My conclusion would be that in some cases the parasoll devices communicates zigbee messages that ZHA just simply ignores, while it would be relevant to use these messages. Do you agree with my observation? And if so, do you need more info?

MattWestb, 2025-03-06:
### One step in the right direction !!!
But more strange things is going on.

(Z)HA have one hockup and need restarting for getting the logging working OK.
Its normally happening then paring devices and cant getting any things from the device but its sending it OK like paring one emote and nothing is happening but its not the truth.
Also very likely HA have renaming the old name for the device ID in the system and you is seen the old and not the current but shall being fixed with restart but loosing the old historical status (but you shall see it in the logbook).

After restart is your Parasoll doing checking every 55 minutes if looking in the mini log on the device card ?

Have you getting better link quality after moving to CH 21 ?

The energy scan is what the coordinators radio receiving without checking if its Zigbee protocol so it can being Bluetooth or other RF interference on that channel but the normal problem is WiFi but its very mush we cant see with easy tools.

I dont have HUR hub but it can being that some device in it is spamming its network and if its on the same channel it can more or less blocked (ZHA was doing scans for it before but our **Puddly** was taking it away then many users was complaining of it) but it shall not getting the LQI very low but slowing the network down.

Also can you looking on the visualization / Zigbee Map how the numbers is on the links between the routers. If one link is symmetric (the sending and reviving is around the same quality) its good but if its large different its bad and ZHA is marking it red / yellow then its cant being used goof in both directions and the network must routing sending and receiving on different links.
I was putting in 2 tuya dimmer / relay and they is very asymmetrical compared to IKEA standard routers:

[image]
coordinator <> 203/88. Signal repeater <> 136/85 and most IKEA <> IKEA is around the same (not 100% the RF is not doing it like that but around = good).
PS: All devices in my test network is located in 1.5 meters radius but the coordinator is 4 meters away from them.

FrankTub, 2025-03-06:
> (Z)HA have one hockup and need restarting for getting the logging working OK. Its normally happening then paring devices and cant getting any things from the device but its sending it OK like paring one emote and nothing is happening but its not the truth.

Not sure if I understand what you mean here?

> Also very likely HA have renaming the old name for the device ID in the system and you is seen the old and not the current but shall being fixed with restart but loosing the old historical status (but you shall see it in the logbook). 

I'm 100% sure that this is **not** the case. I have opened and closed the door like 5 times standing there looking at the Logbook (and at that time no one else was at home). And seeing the events in the logbook being added as I opened and closed the door without it affecting the state of the device in HA. So that left me still a bit puzzled and I could imagine a custom quirk for parasoll add value for some users.

> After restart is your Parasoll doing checking every 55 minutes if looking in the mini log on the device card ?

It only did that once random like 16 hours later. So that is probably not a good sign 👎 

> Have you getting better link quality after moving to CH 21 ?

No my current setup is worse in experience(FWIW I changed it to CH 20). And looking at the graph this makes sense, I see less green lines than I used to and other than that there are no real improvementa. I looked at the energy scan of the diagnostics again, but currently that is not a reason to change it back:

```json
    "energy_scan": {
      "11": 17.086630587133605,
      "12": 82.35373987514762,
      "13": 93.76433891498253,
      "14": 95.69133648577223,
      "15": 95.12234809209261,
      "16": 13.711043742539033,
      "17": 12.244260188723507,
      "18": 23.33483723001185,
      "19": 65.26028270288712,
      "20": 55.9836862725909,
      "21": 65.26028270288712,
      "22": 28.30261646762903,
      "23": 43.057636198227904,
      "24": 4.15070068297423,
      "25": 85.82097888710312,
      "26": 95.69133648577223
    },
```

> I was putting in 2 tuya dimmer / relay and they is very asymmetrical compared to IKEA standard routers:

I see what you mean and most of my routers are quite asymmentrical, so often times the first value (sending?) is a lot larger than the second value. I think this is why they are probably doing a bad job at playing their role as a router? For example:

[image]

So perhaps my easiest fix would be to simply replace all my crappy ass plugs and buy something from ikea?

FrankTub, 2025-03-07:
To add to this, the same thing happened when repairing a vallhorn motion sensor of ikea:

[image]

This thing however does change the attribute `Openen` when motion was detected, so at least this one can be (ab)used 😄

MattWestb, 2025-03-07:
Both Parasoll and Vallhorn have one OnOff klsuter that sending light commands Oh/Off and one IAS zone. The later is little triky then it must registrating and setting up the zone and can failing some times, Restart HA and look if its being OK.

I can only recommending baying 2 INSPELING (2X10€) and making one new network with only them as router and adding some end device and waiting little and see how its working.
IKEA is normally not going max what the chip can if sending power for getting the RF links balanced (some other is using max and getting problems also with the stability of the chip then being warm) but some is not so good antennas (some E27 its in the metal socket) but normally OK.

I have never see om so bad behaving Zigbee network as yours but its in the ground bad devices that making the most of the problems.

Also updating your HA then Vallhorn have getting more functionality like setting the time lights is being on from ZHA and its from 10 seconds to days and not the default on the back of the sensor!
And your parasol need new battery (can being way its not doing the ISA zone) and both need checking for firmware updates (can being tricky if the network not working OK).

MattWestb, 2025-03-07:
One Vallhorn at latest ZHA and good working network:

[image]
Its making nice checkins every 55 minutes and both sensors is working OK plus the light.

cianiandreadev, 2025-03-08:
Omg, I just bought today a new Ikea Parasoll and I'm having exactly the same issue. 
Reading to this issue I have the feeling I won't make it work. 
What is the "easy solution"? Buy a different brand? Buy the Ikea bridge and use that to find the device?
😬

MattWestb, 2025-03-08:
With all Zigbee network you must have good routers or devices is miss behaving.
And parasoll is one Zigbee 3 end device and working best if the network is supporting that functionality that many no named or old not updated devices is doing it OK.
So first by only Zigbee certificated routers and not no name from china that is not have implanting basic functionality in there firmware.

I have some Parasoll from day one (then was getting them in Vienna) and they is working OK in one good working zigbee network.

Start looking on the Zigbee map (visualization) and see if your network looks working OK and if you have routers that is blocking the network traffic (bad links between routers).

FrankTub, 2025-03-11:
@MattWestb I took your advice and went to ikea to buy some of their smart plugs, only tretakt was available in the Netherlands so had to do it with that. I replaced most of my smart plugs and my visualization starts to look a lot better, will continue to replace and repair devices to get it even better.

[image]

I tried to pair my parasoll and vallhorn sensor via a ikea tretakt and I'm now able to move through the whole flow of adding the device by zha without having to reload the zha integration. However I keep having the issue that the device is not integrated properly into zha like my other parasoll and vallhorn devices are. 

So the same parasoll device that was already very rebellious it is now added as:

[image]

I've added it directly next to my sonof dongle, but also via a ikea tretakt that has 255/255 connection with the dongle and in both situations it is not integrated properly into home assistant.. It does however show up in the logbook that it communicates to HA about it's state, but the binary sensor is not updated. Any further ideas what I can do to get my device to work?

MattWestb, 2025-03-11:
Now its looks _LITTLE_  better but you is not having one good link between all routers and the coordinator (some can being they is far away).
If you is getting one good stable link between all routers so they can communicating well (but not more routes possible) its can start working well but you need more stable links then possible for redundancy so the network can finding one working routes if getting interment interference (lik using the microwaveowen).
But now you have one Zigbee network but with little problems in it.

ZHA can being **very** tricky then adding devices and need restart for applying all functions and my feeling is that is some lock in HA that blocking it (also oft getting time out then ZHA is checking for OTA updates / devices doing checkins).

Try reading the battery of the Parasoll then the GUI is showing 0% and is its true its not like being _very_ communicative at all (my experience is that they start being bad around 30-40% battery) by reading battery renaming attribute on the power cluster and waking it up then sending commands to it.

I have installing one X64 Linux machine and installed KVM and spinning up one HAOS and was having problem getting bridged network working OK but after one day it was online and killing my production system (Matter server is fighting with all other in the system but its somthing more its broadcasting to other HA instances that blocking all) and was spending 2 more days before taking down the HAOS on the KVM for getting the production system being online longer then 30 minutes (normally never down / restart).

I was starting building my production network with one TF plug and one TF dimer switch and Aqara T/H sensor (they is very picky) in every room and working great only the entree its little bad then its one level down and have very much reinforced concrete to forcing but have taking out all LIDL (tuya Zigbee certificated devices) that i can getting away and the same with normal tuya but i cant getting all functions i need but its much better without trubbling routers.
The last was my WC and bath fans that i was baying Askväder from IKEA and open them (dremmel) and direct mounting cables for the fans that is triaged of motion sensors sending On with time (most tuya have not implanting THAT command) and with binding its working then the system is down and is getting better routers.

Its always two steps forward and one back but by learning its going in the right direction.

FrankTub, 2025-03-12:
So by now I have replaced all my plugs with the ones from ikea and the visualization now looks like:

[image]

Not really sure why ZHA draws so many lines, even if some routers are really far away from another it is added and displayed as red here. The ones relatively close to another (<5m)  are all at least with an orange line now.

I also added below configuration again:

```yml
zha:
  zigpy_config:
    source_routing: false
    ezsp_config:
      CONFIG_MAX_END_DEVICE_CHILDREN: 0
```

When I now add a parasoll device that I had lying around it does not matter if I'm really far away from the dongle or not, it simply get's added without me having to do any special thing. The same is true for a vallhorn I had lying around, no issues there and I can simply add it.

The thing remains however that for the two devices I already had issues with, I keep having difficulties (One Parasoll and one Vallhorn). It just looks like when you add them they fall back to all the history that zha knows of that device and just keeps the bad behavior that I used to have. The pairing it self is really simple now, but the result remains the same and still not as expected.

Is there some way to do a hard delete from all the history of these devices so that they can pair a blanco device?

MattWestb, 2025-03-12:
Now its looks like you is having one real **Zigbee Mesh Network** !!!!
Most of the routers is having green lines so its strong stable links.
One router is saving the strongest neighbors (depends of the device config how many it can handle) and is using the best with less hope (little more but east say), If the active link its trying the next god one and if its OK its deleting the old good and using the new one and if all links its bad its start re scanning the radio Chanel (takes long time). That is the working model of one mesh network and its working in real if the routers is good behaving.
Then some routers is fare and the device can still communicating with them its still in the routing table but never used if all is being OK. Allso its important that links is symmetrical = can sending and receiving on them in both direction. IKEA is using Sirlabs recommendations and is not using the maximal sending power (20 db) then all can herring it but it cant herring them then the receiver cant herring the week signal then so mush strong sending all the time.
(tuya, Sonoff, Drestens CornBee and so on have not understanding how RF links is working).
Also sleeping end device is using 1/10 or lesser power then routers so they being jammed of routers far away and getting problems (you see Sirlabs chips have very good radio receivers => many week links) but with balanced links its possible using them if needed.

Best way getting old data deleted is deleting the device and waiting one minute and HA **shall** cleaning it out from the database (cashing of data) before adding it agen (added as new) but sometime its not working then its best restarting HA before start adding it (you see if its getting the old name you have getting it or one standard name from the device).

Forcing the coordinator not having direct children (end device) is only good if having one working mesh network then if the coordinator / host system is not working OK they still communicating with its parent and not leaving the network (old aquara end device is doing it if the parent is not  responding and they dont try finding on other parent like they shall do as Zigbee end devices).
Sourcerouting is one other thing and its working well if having good routers and is good for large networks but can also destroying the mesh functionality if not all is playing well (like some bad  t..a  routers).

One interesting thing is how energy scan is looking now with not many "shouting" routers around !!

FrankTub, 2025-03-13:
> Best way getting old data deleted is deleting the device and waiting one minute and HA shall cleaning it out from the database (cashing of data) before adding it agen (added as new) but sometime its [...]

I'm still struggling with this. This single sensor (100% sure that it has worked before) keeps giving me all sorts of shit. I've removed the device from ZHA yesterday and rebooted HA several times since. I added the device back again (with a freshly charged battery) on 12:41, but as you can see in the printscreen below it just takes all history from the old entity it used to have in HA.

[image]

In the logs I see:

```log
Log details (ERROR)
Logger: homeassistant.components.recorder.table_managers.statistics_meta
Source: components/recorder/table_managers/statistics_meta.py:314
integration: Recorder ([documentation](https://www.home-assistant.io/integrations/recorder), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+recorder%22))
First occurred: 12:41:01 PM (1 occurrences)
Last logged: 12:41:01 PM

Cannot rename statistic_id `sensor.ikea_of_sweden_parasoll_door_window_sensor_batterij` to `sensor.deursensor_trapkast_batterij` because the new statistic_id is already in use

Log details (WARNING)
Logger: homeassistant.components.recorder.entity_registry
Source: components/recorder/entity_registry.py:76
integration: Recorder ([documentation](https://www.home-assistant.io/integrations/recorder), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+recorder%22))
First occurred: 12:41:01 PM (4 occurrences)
Last logged: 12:41:01 PM

Cannot migrate history for entity_id `binary_sensor.ikea_of_sweden_parasoll_door_window_sensor` to `binary_sensor.deursensor_trapkast` because the new entity_id is already in use
Cannot migrate history for entity_id `sensor.ikea_of_sweden_parasoll_door_window_sensor_batterij` to `sensor.deursensor_trapkast_batterij` because the new entity_id is already in use
Cannot migrate history for entity_id `update.ikea_of_sweden_parasoll_door_window_sensor_firmware` to `update.deursensor_trapkast_firmware` because the new entity_id is already in use
Cannot migrate history for entity_id `button.ikea_of_sweden_parasoll_door_window_sensor_identificeren` to `button.deursensor_trapkast_identificeren` because the new entity_id is already in use
```

I guess it is just keeping some data of the device that is not removed that is now giving me all these issues. I then tried `Developer Tools` &rarr; `Actions` &rarr; `Recorder: Purge entities` &rarr; all related entities. This ran succesfully. Then rebooted HA and tried to add the device once more. And again it still adds it in the same way as before. It migrates all the history along with it and just returns the same error. Do you have any more ideas how to make ZHA fully forget about this device?

> One interesting thing is how energy scan is looking now with not many "shouting" routers around !!

Haven't really played around with it but currently it is like below (I'm using channel 25 again):

```json
    "energy_scan": {
      "11": 23.33483723001185,
      "12": 46.26944564832987,
      "13": 89.93931580241996,
      "14": 87.33047519856483,
      "15": 84.164247274957,
      "16": 3.2311094587038967,
      "17": 17.086630587133605,
      "18": 19.00785284282869,
      "19": 3.2311094587038967,
      "20": 4.69985354430736,
      "21": 19.00785284282869,
      "22": 17.086630587133605,
      "23": 12.244260188723507,
      "24": 15.32285793082191,
      "25": 43.057636198227904,
      "26": 92.0598007161209
    },
```

MattWestb, 2025-03-13:
Was you giving it on new name then still having the paring windows open ?
I think the system is technical saving the name from the default and then renaming it then you putting in one new and the system is getting little between old and new name is being in the database for it.
If yes deleting it and restart HA and adding  it new (shall have the standard name `ikea_of_sweden_parasoll_door_window_sensor` or if not real deleted is showing up with your old name `deursensor_trapkast`) and dont  change the name only changing aria if you like.
Then going back (not kicking on the device) to integration and ZHA and then the sensor and looking if its working OK. If it dos try change its name.
I have seen the logger complaining many times but hvae not paying any attention to it but i have also deleting  old trach from the recorder like you have doing.

43% is not bad i was having around 85% the ZHA was doing it then starting before and it  was not one big problem but your network is not more balanced then send and reviving its not  equal but also think that all devices is still chatting in the network then the chip is listening on the channel so it OK.

Also try reading the battery renaming of the device and see if ZHA is updating it (three dots right  from reconfigure and then change  to power cluster and battery renaming attribute and wake uo the device then clicking on send.

FrankTub, 2025-03-13:
Thank you very much for all your help @MattWestb it is now finally working! In my complete house I have perfectly working zigbee devices and devices are recovering themselves when they were in a limbo state for some time without me having to do something.

Still not really sure how I finally resolved it but my guess for anyone stumbling across this I guess this did the final trick:

- Remove the device from ZHA
- Restart HA
- Without me having to do anything the device was automatically added again in ZHA after the restart but still unresponding.
- It is now added as `ikea_of_sweden_parasoll_door_window_sensor`, open the device in zha &rarr; click on the sensor &rarr; click on the gear icon &rarr; rename the `Entity_id` to something that you have never used before. In my case it was still showing as on and off, this can be solved by setting `Show as` to `Door`.
- Verify that you sensor is updated in HA again.
- If so, now rename it and test again. Everything should be OK now.

### Energy scan

I did a few energy scans afterwards and in the other ones the channel is always > 95%, while channel 20% is in the last few hours always < 10%. Would it make sense for me to try out to switch to another channel (I'm not able to set my routers wifi channel to a fixed channel (Deco)).

MattWestb, 2025-03-14:
Then deleting one sleeping end device you must waking it up after clicking on the delete or its very likely the device is not getting the command (leave the network = deleting the network key and go in pairing mode) and if its happening its still in the network and later ZHA if "finding" it and adding it but it looks going OK in your case then it was happening after the restart of HA.

FrankTub, 2025-03-17:
@MattWestb , I went completely through this [rant against Parasoll devices](https://github.com/Koenkk/zigbee2mqtt/issues/22579) and it is not conclusive, but their one of the suggestions is to use a [customized solution for Ikea devices](https://gist.github.com/Koenkk/551ae5fda75e295356599b56af1adfe3). Do you think something could be applied to ZHA and be helpfull?

MattWestb, 2025-03-17:
Z2M must configuring all things for one device or there system is knowing nothing.
ZHA is trying do all things 110% Zigbee compatible so if one device is good behaving its working  out of the box. And by Zigbee standard its also possible implanting private functions on manufacturer cluster that other devices ignoring.
IKEA device is one of the best Zigbee compatible devices but some have extended functionality (shortcut button and Symfonisk 2 and so on) that must having extra code for working.
Most is working out of the box and can need little cosmetic fixes for being good in ZHA.
Parsoll is very standard and is using on/off cluster for sending On/Off commands to lights and one IAS Zine for larm functionality. Its need registration for working and i have seen problem devices doing it OK and if not the status is not working well.
The Z2M code is standard iasZoneAlarm, battery reporting and checkin that all is being done in ZHA then its reading the device configuration so we dont  need do any of this things.

I think it can being interesting if you putting debug log on for ZHA and doing test so can see if the IAS is working OK then i  think that is the problem is.

And if the device is not doing checkins its not getting the configuration from ZHA and need being reconfigured. If doing checkins is the network working OK but it can still being other problems on the road.

FrankTub, 2025-03-27:
I tried several things by now and these ikea sensors are still kicking my ass. I did the following.

- Moved to channel 20
- Repaired 1 or 2 devices where needed

Concluded that the device that was kicking my ass the most was now doing checkin events without me having to do something. However, the result was that the sensor `Deursensor Toilet` with `IEEE: d4:48:67:ff:fe:d4:43:95` has now the same behavior as the sensor that used to kick my ass. That is it is reporting the state, but it is not doing checkin events, while it used to do so before migrating to channel 20.

I enabled zha debug mode and removed this device from the network. Then I added it via some router and tried it out, the sensor did not report it's state in zha. Also I don't see any events in the logbook.  Then I renamed the sensor name from `binary_sensor.deursensor_toilet` to `binary_sensor.deursensor_toilet_openen` (this is the name in dutch that properly setup devices have by default). And then I clicked on reconfigure the device and now when I check the state is properly displayed in the logbook and in the binary sensor. See logbook info below if you want to understand a bit more of when I did what and compare it to the debug log:

[image]

My adventure with the debug log can be found here: [zha-parasoll-rename-required.log](https://github.com/user-attachments/files/19493400/zha-parasoll-rename-required.log). Tried to keep it as clean as possible, but when searching for `d4:48:67:ff:fe:d4:43:95` it still returns a lot of results. It's not really giving me that much info, could you please take a look and point me into the direction why this device is still behaving so buggy?

MattWestb, 2025-03-27:
Doing checkings = its communicating OK its parent router and that can sending the commands to the coordinator OK.
Next thing its it have 2 cluster that sending status. First the On/Off for searing light and the second is IAS Zone (alarm) that is sending Open / Closed but its need being registration OK with the alarm zone for working OK.
And the On/Off need getting OK reporting configuration for ZHA shall getting the commands it sending. 
So i always do reconfiguring of end devices for getting the configuration working OK (some time its looks working OK but i have it as standard).
So for the moment its looks like both network and the reporting is working OK for both devices.
Sadly some time HA is blocking status updates (or losing them) and then devices is being paired its very much talking in the system so and all must being sent to the device OK or its not working OK and if missing some reporting configuration or the registration of the IAS Zone its not OK  the system is not getting all status updates OK.
The blocking / lock in the (Z)HA can doing much problems and the best is restarting HA and letting it stabilizing and then doing reconfigure and if its working OK you is getting all green,
Was fighting with some Styrbar remotes for over one years ago and then was very nasty but in the end they is working well but must taking out the battery if they is not getting the configuration OK then they is going in sleep very fast if not getting the binding and reporting OK and is not  trying sending anything then pressing one button so its sleeping well.

If i getting time i shall testing little and doing paring with IKEA Dirigera and ZHA and sniffing and see if i can see somthing strange but its needs time for digging in the sniffs then Zigbee have many layers and must analyzing all of then to see if somthing its going wrong so must do it on my test network but i must killing the Inspelning then its fuddling the network with its reporting.

PS: The IEEE (long network address) (used mostly in erarly stage of paring or network management) is always the same and you see it then the device is joining  the network.
But the short address (MKW) (is used for most normal communications) can / is changing then the device i joining the network and can happening then all is working OK if the network / device dont like somthing.

Keep looking if you is getting checking every 55 minutes from your sensors and then i think the status reporting shall working OK.

MattWestb, 2025-03-28:
Have doing little sniffing.
Diridera is only having one end point and doing minimal setup (normal IKEA) so only IAS Zone and no On/Off (DG is not using binding of light commands so no use but they is using it in direct pairing with lights).
Then doing one change the device is sending one zone change command alarm and the coordinator is sending one aps ack and one default response = 3 commands sent and is very fast.
Was looking if changing status very fast (1-2 seconds) is device not sending the change and the system is keeping the zone in the "old" status.

ZHA is having 2 endpoints and its setting up the IAS Zone (EP2 in device) OK but not OnOff then repairing it  (EP 1 in device) but the status is working OK on the device card if not changing it to fast.
Doing reconfigure ZHA is re init the ISA Zone on EP2 and EP0 (ZDO EP) and also the On/Off is being OK.
Then the device is sending one open to EP 1 and one to EP2 plus one On to EP1 plus ack and default response for all commands but the ack and default response for the 2 IAS commands is sent to EP2 and not one to EP2 and one to EP1.
The good thing is the On/Off is very fast and if doing it very fast is being sent both the on and off before the parent have getting the IAS commands so that is no problems (like some tuya scene switches that  is not sending more commands if not getting ack and default response and the last in the wrong direction).
In the end one change of status is in ZHA 10 commands sent !!!!
That is no problem if the network is working stable with unicasts and dot have routing problems.
After letting the system setteling down and was looking on the live sniff the device was doing checking but HA was not updating it in the device long. Was looking in the HA logbook and it was 2 there on the same timestamp. Back to the device card and reloading the window i is also having the checking and also to times.

So the device and the network is working OK but the device and ZHA have little different "personalities" but all is very Zigbee standard as normal  IKEA.

Ping our @puddly and @TheJulianJES !!
Little thinking from my side.
First do you both having one Parasoll (10€ and its one MG24 chip) ?
Its looks ZHA is setting up the IAS Zone little strange then in Dirigera its only sending zone changing to one end point (its only having one EP). I was only see that reconfigure was also setting it up on EP0 on the coordinator and its sending the change to EP 1 and 2 (2 commands with the same app sequence number).
Is it somthing strange ZHA is doing then having 2 end points that is comfusin the device ?? ZHA is sending the ack of the same SQN to EP2 2 times and no to EP1.
On Off looks working  very well but is not shown on the device card but in the log = good for automation.

And for @TheJulianJES you is having one bug hunt on log problems and other strange  things and i think we see little of it here 2 then if the device is not working OK then being OK paired and reconfigured is it being fixed with restart of HA and all is working OK.

And for @puddly Was looking how ZHA is handling checkings and the response is fast pull 8 (quarter seconds =2 seconds) and after done sending fast pull stop.
The device is pulling its parent 2 more times after getting the command in 1 second interval = 3 second alive before going to sleep. I think its little short then if the network is having problem (making popcorn in the micro or some HA integration is doing some work) is it very likely its timing out and we is getting problems.
If i remember right is gen 3 IKEA device requesting 8 minutes end device time out = 8 minutes deep sleep (and i think the parent  shall holding the command to it until its waking up). If setting the checking response longer is not problem then the system is sending one fast pull en in some milliseconds if all if OK and if the device is not getting it its going to sleep after the timeout.
Way not 8 (normal end device pulling for no Zigbee 3 devices) or 12 seconds and we is not getting problems with timeout the doing OTA checks. Also 20 or 30 seconds shall being OK but can making some sleepers burning batteries if the network is not working OK and the stop fast pull is not arriving OK.

More thinking pleas !!!

FrankTub, 2025-03-31:
I double checked and the devices that were succesfully setup with checkins still have checking events in the log. I still have one parasoll where I'm not sure how I can get this device to have the checkin events working. This device however is working as expected, probably because we open/close that door quite frequently. 

If there is anything for me to test or deliver more info please let me know.

MattWestb, 2025-03-31:
If the device is working well without doing checkins then its OK.
Its normal if it parent is not supporting real Zigbee 3 and is working in legacy Zogbee HA mode.
You can trying recofiguring it and also taking out the battery and putting it back after some seconds so its rejoining the network.

I hope Puddly is taking one look how the IAS Zone is being setup and if its some problem how ZHA is doing it but in the ground its looks working OK but its little messy from my point of view how ZHA is doing it comparing with how Dirigera is doing it the looking in the sniffings i was doing.