zigpy/zha-device-handlers #3857: [BUG] Innr SP240 newest firmware not compatible anymore with the quirk

Skaars-fr, 2025-02-12:
### Bug description

It seems that newest firmware behavior changed

With the current zha quirk and the newest firmware the reported consumption of the plug is far too high
The 1.7 firmware was working perfectly but the 1.9 firmware introduced a change in the divisors and multipliers behaviour.

https://github.com/Koenkk/zigbee-herdsman-converters/pull/8715#issuecomment-2640979923

### Steps to reproduce

1 update Innr SP240 to 1.9.30 (or 1.9.28)
2 check the energy consumption

### Expected behavior

According to what was reported on z2m github space, the new value is 10 times higher than the expected value

https://github.com/Koenkk/zigbee-herdsman-converters/issues/8712

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

[{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4454,
    "maximum_buffer_size": 74,
    "maximum_incoming_transfer_size": 80,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 80,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x010a",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0702",
        "0x0b04",
        "0x1000",
        "0xe001"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "innr",
  "model": "SP 240",
  "class": "zhaquirks.innr.innr_sp240_plug.SP240"
}]

```

</details>

### Diagnostic information

[zha-01J6W5FKBBNTRHTDK6VTSD41ZW-innr SP 240-995ccf6fb61ac727b9784daf8291c58f.json](https://github.com/user-attachments/files/18774448/zha-01J6W5FKBBNTRHTDK6VTSD41ZW-innr.SP.240-995ccf6fb61ac727b9784daf8291c58f.json)

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

nraeth, 2025-02-14:
I can confirm that.

Started with the FW update to 1.9.30

pajtaz, 2025-03-22:
I have the same issue with SP 240 firmware 1.9.30. It is 10 times larger value being reported. I am monitoring 3D printers that are drawing about 200W right now but the total energy consumed went from 5.91 kWh to 6.23 kWh in 10 minutes. That is 320 Wh in 10 minutes or 1920 Wh = 1.92 kWh in 1 hour, where it should be 192 Wh.

AlexanderLeporiday, 2025-03-22:
If you update to the latest firmware you also need to update Z2M to a newer version that has the new converter. You then need to reconfigure the device to load the new converter as the old values will not work for the latest firmware. The latest firmware actually provides its divisors on setup, whereas previously they were hard-coded into the converter.
It appears the converters are cached even through updates so reconfiguring the device will load the correct values. It will reset them back to factory default so any customizations will need to be reapplied such as reporting.

nraeth, 2025-03-22:
> If you update to the latest firmware you also need to update Z2M to a newer version that has the new converter. 

This here is the ZHA device handler, not Z2M though.

AlexanderLeporiday, 2025-03-22:
My apologies. I was getting notifications for this topic from GitHub for some reason. It seems so wonderful linked it to the Z2M topic where the firmware changes were discussed in great length.

You are correct that ZHA will need to be updated in a similar way that Z2M was updated, to get the device values which have been added to latest version of the firmware. Avoid hard coding them if you can get them from the device during set up.

pajtaz, 2025-03-22:
One thing I don't understand, why can't I just change the value being reported in the "Sensors" section of the "Device" so it is divided by 10? I cannot find anywhere how to do this myself. I am newish to Home Assistant and Zigbee and I thought I can simply modify these kind of things as I wish.

[image]

jo8be, 2025-08-26:
I just updated the firmware of my SP 240 from 1.7.17-3.6.85 to 1.9.1e-3.6.85.

Current is now reported with a factor of 1000.

What is now shown below as 45 A was before 0,045 A.    My fault or known bug? 

[image]

Basvg, 2025-08-26:
> I just updated the firmware of my SP 240 from 1.7.17-3.6.85 to 1.9.1e-3.6.85.
> 
> Current is now reported with a factor of 1000.
> 
> What is now shown below as 45 A was before 0,045 A. My fault or known bug?
> 
> <img alt="Image" width="491" height="522" src="https://private-user-images.githubusercontent.com/143457228/482237578-f6242794-51c5-4859-9a02-5e9f591a26cc.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTYyMzIyMzIsIm5iZiI6MTc1NjIzMTkzMiwicGF0aCI6Ii8xNDM0NTcyMjgvNDgyMjM3NTc4LWY2MjQyNzk0LTUxYzUtNDg1OS05YTAyLTVlOWY1OTFhMjZjYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgyNlQxODEyMTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wNTUwOWE5MjMzYmQ5NWU4MWY3NzM2ZWE5ZWUyZWNiOWMyNWNmYmRjZWM2MTkzNTA1NTU4NjUzYWU4M2EwZGMzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.7pa-XJr2a9ysfmrFuGX2A5_keEFEme3MOVlhCvSo1eY">

When I updated the firmware previously, I had to reconfigure the device in ZHA to load the new configuration. Or alternatively delete it and re-pair. Could that help?

TheJulianJES, 2025-09-22:
I'm trying to get all relevant attributes per version, so we ideally have a quirk that works with all plug models and versions. See this comment here for more info: https://github.com/zigpy/zha-device-handlers/issues/2781#issuecomment-3234748088

But yeah, this still not fixed. Hopefully, I can look into this more soon.

bart1970, 2025-10-18:
Just removed the SP240 plugs from the zigbee network and re-paired them.
Summation delivered nicely started at 0 kwh, but still counting a factor 10 times too much.
Version HA:2025.10.3
Firmware SP-240: 0x191d3685

Dirk77777, 2025-12-10:
> Just removed the SP240 plugs from the zigbee network and re-paired them. Summation delivered nicely started at 0 kwh, but still counting a factor 10 times too much. Version HA:2025.10.3 Firmware SP-240: 0x191d3685

Same for Firmware 0x191e3685 with SP240
Re-paired and reconfigured the device, but summation is still increasing by factor 10 too fast.

TheJulianJES, 2025-12-30:
There still seems to be an issue with the firmware version filter used for quirks due to some attributes being loaded too late. We'll [fix that](https://github.com/zigpy/zigpy/pull/1735) and move forwards with [the PR](https://github.com/zigpy/zha-device-handlers/pull/4592) that should fix this for all SP 240 firmware versions.

@GerryInnr If you're around, I have a couple of questions for you:
1. What is the first firmware's `FileVersion` that changed/fixed the divisor for each of these plugs: SP 120, SP 234, SP 240?
  For the SP 240, it seems to be `0x191E3685` (corresponding to `1.9.30`). Or is there already an earlier version that fixed this before `0x191E3685`/`1.9.30`? Having this information for all of the plug models would be really helpful to determine the cutoff point on when to apply a quirk or not.
2. On the old firmware versions, the quirk needs to change the divisor to `100`. For the newer firmware versions, the devices all have `1000` as the divisor. Is that always the case for all newer firmware versions of these three plug models or do some newer firmware versions need a different divisor than `1000`?
  The reason I'm asking this is because the old quirk will have persisted the `100` divisor into the zigpy database, so re-pairing (with a fixed quirk) would be required to correctly use the device's divisor again. However, if we can always override the divisor to `1000` for newer firmware versions, we'd avoid users needing to re-pair the devices.
3. We've recently launched the [zigpy-ota repo](https://github.com/zigpy/zigpy-ota/discussions/2). All updates in that repo will automatically be shown to ZHA users with Home Assistant Core 2026.1.0 or later (already available in beta). When the quirk is fixed, it might be nice to roll-out the Innr firmware update to all users.
  Can you submit the latest OTA(s) in that repo? No PR or Git usage required, just filling out the GitHub issue form for adding an OTA is enough to automatically have the bot create a PR with the relevant metadata for HA users. We'd only merge that after a fixed quirk for all firmware versions is included in HA, but it would be really nice if you could already submit the images.

Thanks!

GerryInnr, 2026-01-05:
Hi @TheJulianJES here are the answers to your questions:

- The SP 120 and SP 234 are older plugs, for which no new firmware will be made available anymore. The last FileVersions available are resp. 0x11040002 and 0x31016610. Both these plugs do not support the Divisor and Multiplier for the CurrentSummationDelivered attribute. That attribute provides kWh readings in 2 decimals, in other words in units of 10 Wh, which would be Divisor 100 and Multiplier 1.
- For the SP 240/242/244 (EU/UK/US), the first FileVersion that supported the Divisor and Multiplier for the CurrentSummationDelivered attribute is 0x191b3685 (SWBuildID "1.9.27"). This is also the first version that had CurrentSummationDelivered in 3 decimals, in other words in units of 1 Wh (Divisor 1000 and Multiplier 1). All versions before that provided 2 decimals and did not support the Divisor and Multiplier.
- Due to some Zigbee systems not handling kWh readings in 3 decimals well, we were forced to make different firmware versions for different systems. Some versions have Divisor 100, and other versions have Divisor 1000. We roll out the right firmware version OTA files to the right systems; for HA and Z2M we will only provide OTA firmwares with 3 decimal kWh / Divisor 1000. The factory only produces plugs with Divisor 1000. However, someone could switch from one system to another, so you cannot rely on all versions >= 1.9.27 having Divisor 1000.
- You *can* rely on all firmware versions >= 1.9.27 to provide a correct Divisor.
- I will upload the latest images with 3 decimal kWh / Divisor 1000 to the zigpy-ota repo this week.
- The SP 120, SP 234, SP 240, SP 242, SP 244, and OSP 240 (outdoor plug) are the only power monitoring plugs we brought to market. SP 120, SP 234, and SP 24x have been discussed above. The OSP 240 follows the same firmware versions as the SP 24x, but was released at FileVersion 0x19223685 (SWBuildID "1.9.34") so there are no older versions without Divisor.

Hope this helps! If you have any further questions let me know, I'll gladly assist.

TheJulianJES, 2026-01-07:
Thank you for that detailed answer! Very much appreciated. Status update on this issue:
- For now, we've merged https://github.com/zigpy/zha-device-handlers/pull/4592 which overrides the divisor to `100` for older firmware versions and `1000` for newer ones (SP 240 only). It's only done for a slightly newer version (`0x191E3685` instead of `0x191B3685` you mentioned), but we can adjust that for a future HA release.
- I'm aware we can't count on all newer firmware versions needing `divisor=1000`, but from your comment, it should apply to the vast majority of plugs. This is currently needed to fix the issue without re-pairing and allows OTA firmware upgrades for old plugs without re-pairing (though a ZHA reload or HA restart would still be needed for OTA upgrades at the moment).
- For SP 240 plugs on newer firmware from other systems that provide and need `divisor=100` (few hopefully), users should be able to upgrade the firmware to an even newer version once that's later uploaded to the [zigpy-ota repo](https://github.com/zigpy/zigpy-ota) that then uses `divisor=1000`, working properly with the quirk override for new versions.
- In the future, I hope that we can remove the `1000` divisor override quirk for newer SP 240 firmware versions when some other underlying ZHA changes have been made and then only rely on the device-provided `divisor` attribute again.
For now, I think the current solution is fine and much better than all plugs that need `divisor=1000` showing invalid values.
- It looks like we need to make no changes for other plugs, though we could add a fix for SP 242/244 plugs with old firmware. I'd probably not do it now, but instead wait on the OTA upgrades being available. This will likely require removing and re-pairing the plugs to read the updated `divisor` attribute then (if coming from an old version without `divisor`).
- The PR mentioned above will be available with Home Assistant Core `2026.1.0b5` and later, with the full release coming later today/tomorrow. This should *finally* get the SP 240 working properly for basically everyone. 🎉

jasperkuperus, 2026-01-07:
Thank you both @GerryInnr and @TheJulianJES for your swift and adequate responses. Being new to Home Assistant, I now only wonder what the update process will look like. Is it simply a Home Assistant update? Or should ZHA be updated independently? Can we expect this fix to land in the next HA version. I'll have to undo my workarounds when I install this update.

AllardKatan, 2026-01-07:
@TheJulianJES are you sure it is in b5? I just updated to this version and switched my espresso machine on.
As you can see it reports 1kW for 3 mins as 0.5 kwH, still a factor 10 too high.
[image]

AllardKatan, 2026-01-07:
Note it is neither E nor B which are the versions mentioned above.

Dirk77777, 2026-01-11:
I can confirm now that with the latest firmware from pre-release my plug is working fine. I reconfigured it once. Consumption is fine, it also updates nicely in small increments.
This nice incremental update now even works fine again for the old firmware 0x16183682 after reconfigure, thanks for the hint. However I will update this plug to latest firmware soon, of course. 

I will now update a few more of my plugs with the pre-release version.

Dirk77777, 2026-01-14:
@TheJulianJES: Will the new firmware appear automatically in HA out of the box as an available update, or do we need to configure a separate update channel for zha just like I did earlier with this configuration?

`zha:
  zigpy_config:
    ota:
      extra_providers:
        - type: z2m`

TheJulianJES, 2026-01-14:
@Dirk77777 The updates will show up when running HA 2026.1.0 or later in a bit, with no configuration required.
We want to wait for HA 2026.1.1 to have rolled out further, as it includes a bugfix for the SP 240 v1. Then, the updates will be pushed to the stable channel shortly, so everyone will get them out of the box.

For now, they're only on the zigpy-ota beta channel, which can be activated like this (as [described in the zigpy-ota README](https://github.com/zigpy/zigpy-ota#using-pre-release-channels)):
```yaml
zha:
  # custom_quirks_path: /config/zha_custom_quirks
  zigpy_config:
    ota:
      extra_providers:
        - type: zigpy_ota
          channel: beta
          override_previous: true
```

You can add that to your `configuration.yaml` (or merge with existing `zha` config), restart Home Assistant, check for updates in HA settings (multiple times), and the updates should show up if your plugs are on an older firmware version.
(We aim to expose this beta channel option in the ZHA UI with the next major HA release.)

But in the next few days, they'll be truly available just out of the box.

TheJulianJES, 2026-01-21:
Update: The [zigpy-ota 2026.1.1 release](https://github.com/zigpy/zigpy-ota/releases/tag/2026.1.1) with all provided Innr smart plug images has now been pushed to the `stable` channel.

If you **restart Home Assistant Core** (or wait 24 hours for the OTA index cache to expire) and **then check for updates** (Settings -> System -> Updates -> refresh button), you should see new the Innr firmware updates if you have outdated plugs in your network. You may need to check for updates multiple times, depending on the size of your network. If you bought the plug recently, it's most likely already on the newest firmware version, so you won't see any updates then.

- Due to the quirk for the Innr SP 240 plug, HA Core can just be restarted to correct the "Summation delivered" value when coming from an old version.
- For the SP 242 and SP 244, there is no quirk that re-sets the divisor based on the firmware version. As it's currently not re-read after a firmware update (will be improved in the future), SP 242 and SP 244 users may need to remove and re-pair the device, or re-read the attribute manually. I'm hoping this won't be a big issue. Otherwise, we'll need to improve the release notes for that and/or pull the update until ZHA can re-read the attribute automatically.
For the SP 242 and SP 244, [instructions for those possible update issues are linked here](https://github.com/zigpy/zigpy-ota/pull/14#pullrequestreview-3640008450).

Otherwise, **SP 240 users should all get the updated firmware automatically now, with no configuration required**.
Running a quick reconfiguration through the UI to re-set up attribute reporting is recommended for these plugs.

renatoyamane, 2026-02-11:
> Can you explain why I'm not being offered a firmware upgrade on 2 of my Firmware: 0x191e3685 devices?

I also have the SP242 on firmware 0x191e3685 and I can't find any update to them.