zigpy/zha-device-handlers #3866: [BUG] _TZE200_hr0tdd47 Tuya CO sensor entities not showing values

bbkr, 2025-02-14:
### Bug description

Device was added in: https://github.com/zigpy/zha-device-handlers/pull/3635

However only CO binary state and mute siren entities work. CO level, battery level and self test (I clicked self test button several times) are not showing values. Screenshot here: https://bbkr.org/_TZE200_hr0tdd47.png

BTW: This device was added to my HA instance before it was supported. Does it matter? Should newly added entities work out of the box after update?

### Steps to reproduce

Entities do not work on their own, no user actions are required to trigger this bug.

### Expected behavior

CO level, battery level and self test values to be read correctly by entities.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[https://bbkr.org/_TZE200_hr0tdd47.png]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

[Paste the device signature here]

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

prairiesnpr, 2025-02-14:
Enable debug logging and check if you see any unmatched DPs in the logs. Z2m does enable the magic packet on this, you could clone it as a custom quirk, add `tuya_enchantment()`, then remove and re-add the device and see if anything changes. Some devices won't report without that, although z2m seems to add it to everything so I doubt it will change anything. 

Also, sometimes DPs just don't report, I've added devices where they Tuya dev console shows the DP, yet it never actually reports out and isn't used in the app. It's possible that this specific device just doesn't report those DPs. Without comparing to the Tuya app or a joined device in z2m, we can't be certain. I just based this on the z2m converter and have to trust that's accurate.

bbkr, 2025-02-14:
Thanks for prompt reply.

I'm using HaOS (this bundle of HA and operating system) on bare metal and I added my devices through Zigbee integration. From what I read it is different from Zigbee2MQTT. So I'll figure out how to enable debugging and activate this quirk on my instance and I'll get back to you ASAP.

prairiesnpr, 2025-02-14:
Start with the debug logs, if you see warnings about unmatched Datapoints then your device is reporting, but it's on the wrong DP and we need to adjust the quirk.

bbkr, 2025-02-16:
I've enabled Zigbee integration debugging and extracted lines by device identifier or mac address
```
grep -i '0xDBBF\|a4:c1:38:20:6d:bf:2d:54' home-assistant_zha_2025-02-16T14-15-22.860Z.log > TS0601.log
```

[TS0601.log](https://github.com/user-attachments/files/18815000/TS0601.log)

Here is current device signature:

```
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0500",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_hr0tdd47",
  "model": "TS0601",
  "class": "zigpy.quirks.v2.CustomDeviceV2"
}
```

And here is Zigbee info section.

[image]

Also I'm not currently on-site so I cannot currently trigger self-test or unpair/pair device because it requires physical access to a device. I did not have success with triggering it through `init_test_mode (id: 0x0002)` command.

bbkr, 2025-02-17:
Thanks for checking. I won't be able to re-add device to a network for about 2 weeks (as I mentioned I'm not on site). I'm closing the issue - if re-adding and/or applyin tuya_enchantment won't work I'll restore this issue with more data.

Thanks again.

closed 2025-02-17 by bbkr