zigpy/zha-device-handlers #3869: [BUG] S520567W Cover inverted

bouboun59, 2025-02-15:
### Bug description

The cover information is inverted with ZHA, with Z2M it's ok.
Can you explain why and solve the problem ?

### Steps to reproduce

When you see the cover is open but the shutter is close and inversely when the cover is closed the sutter is open.
%tage are inverted too.

### Expected behavior

I wwish when the shutter is open the rller is open etc.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

the shutter is closed... but the cover is open
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

[Paste the device signature here]

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

bouboun59, 2025-02-17:
Hi

For your information I created a quirks with the code ans it's ok
this quirk inspired by the quirk « »« 1GANG/SHUTTER/1 from Schneider Electric

`"""Quirks for Schneider Electric shutters."""

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.closures import WindowCovering
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Identify,
    LevelControl,
    OnOff,
    Ota,
    Scenes,
)
from zigpy.zcl.clusters.homeautomation import Diagnostic

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.schneiderelectric import (
    SE_MANUF_NAME,
    SEBasic,
    SESpecific,
    SEWindowCovering,
)

class NhpbShutter1(CustomDevice):
    """NHPB/SHUTTER/1 from Schneider Electric."""

    signature = {
        MODELS_INFO: [
            (SE_MANUF_NAME, "NHPB/SHUTTER/1"),
        ],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=5, profile=260, device_type=514,
            # device_version=0,
            # input_clusters=[0, 3, 4, 5, 258, 2821],
            # output_clusters=[25]>
            5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    WindowCovering.cluster_id,
                    Diagnostic.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=21, profile=260, device_type=260,
            # device_version=0,
            # input_clusters=[0, 3, 2821, 65303],
            # output_clusters=[3, 4, 5, 6, 8, 258]>
            21: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.DIMMER_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Diagnostic.cluster_id,
                    SESpecific.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    WindowCovering.cluster_id,
                ],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [33],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    SEBasic,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    SEWindowCovering,
                    Diagnostic.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
            21: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.DIMMER_SWITCH,
                INPUT_CLUSTERS: [
                    SEBasic,
                    Identify.cluster_id,
                    Diagnostic.cluster_id,
                    SESpecific,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    SEWindowCovering,
                ],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [33],
            },
        }
    }`

Possible to merge with zha ?

uvNikita, 2025-09-10:
This issue has been fixed in https://github.com/zigpy/zha-device-handlers/pull/4130