zigpy/zha-device-handlers #3871: [BUG] Tuya TRV: Unable to enter negative values for local temperature calibration

kai-rdh, 2025-02-15:
### Bug description

I have several Tuya TRVs (_TZE200_h4cgnbzg) which failed after updating to 2025.2.0, similar to #3837 

2025.2.4 was supposed to resolve the issue, but it still doesn't work properly. After deleting and re-pairing the TRVs, the offset is visible, but when I try to enter a negative value, i get the error message 

> Failed to convert attribute local_temperature_calibration from -6 (<class 'int'>) to type <class 'zigpy.types.basic.uint32_t'>

[image]

Manually sending a negative value to the device works fine, the entity also updates with the new value:

[image]

After reviewing some of the recent changes (#3853, #3856), I noticed the following ([tuya_trv.py, Line 111](https://github.com/zigpy/zha-device-handlers/blob/ba3c7c7190a060e1a2fa57821b39631704dbe7f9/zhaquirks/tuya/tuya_trv.py#L111C1-L120C55)):

```python
    .tuya_number(
        dp_id=27,
        attribute_name=TuyaThermostatV2.AttributeDefs.local_temperature_calibration.name,
        type=t.uint32_t,
        min_value=-6,
        max_value=6,
        unit=UnitOfTemperature.CELSIUS,
        step=1,
        translation_key="local_temperature_calibration",
        fallback_name="Local temperature calibration",
``` 

I'm not an expert, but as far as I understand the code, uint32 is an unsigned integer, which can only take positive values. Shouldn't it be a signed integer instead? I think that this could explain the error message that I'm getting.

Assuming this is in fact a bug and not caused by me, is there any way I can fix this until it is resolved in the next update? Thanks in advance.

I'm happy to provide additional information if needed.

### Steps to reproduce

1. Update to 2025.2.4
2. Delete device
3. Re-pair device
4. Try to set local temperature calibration to negative value

### Expected behavior

The negative value should be accepted and sent to the device without throwing an error.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
[... 682 lines cut ...]
            },
            "unsupported_attributes": []
          }
        },
        "out_clusters": {}
      }
    }
  }
}

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

TheJulianJES, 2025-02-15:
Nice catch! Download that quirk file (raw), then try to change the type to `int32s`, and load it as a custom quirk.

If you need more instructions, I can provide them later. If it works though, feel free to PR the change (otherwise, we can do it later).

kai-rdh, 2025-02-15:
Did as you said, works perfectly, thank you. I also created pull request.

TheJulianJES, 2025-02-15:
Awesome! Feel free to use the custom quirk until there's a HA release with the PR. If it's not in 2025.2.5, it should be in 2025.3.0, so you can/should remove the custom quirk then.

kai-rdh, 2025-02-15:
Will do that, thanks for your guidance. This was my first time contributing to a software project, so I appreciate your help. Definitely won't be the last time :)

PR #3873 (2025-02-15): Fix int data type for local calibration value of Tuya TRV
## Proposed change

The value can be set from -6 to 6 so a signed integer is needed. Change type from uint32_t to int32s.

## Additional information

Fixes #3871

## Checklist

- [x] The changes are tested and work correctly
- [ ] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works
closed 2025-02-15 by kai-rdh