zigpy/zha-device-handlers #3872: [QUESTION] In a quirk how can I return multiple entities from a single Tuya Datapoint?

jairbj, 2025-02-15:
Hi there, I'm trying to make a quirk for a din rail power switch (_TZE204_fhvdgeuh) but I couldn't figure out how can I return 3 entities for a single datapoint.

If I try to create multiple tuya_sensor() with the same dp_id I get an error DP 6 is already mapped.

Here is the quirk I'm trying to achieve (note multiple dp_id=6, they work independently but not together).

``` python
from zhaquirks.tuya.builder import TuyaQuirkBuilder
import zigpy.types as t
from zigpy.quirks.v2 import SensorDeviceClass, SensorStateClass
from zigpy.quirks.v2.homeassistant import UnitOfElectricCurrent, UnitOfEnergy, UnitOfPower, UnitOfElectricPotential

(
    TuyaQuirkBuilder("_TZE204_fhvdgeuh", "TS0601")
    .applies_to("_TZE204_fhvdgeuh", "TS0601")
    # Main switch
    .tuya_onoff(dp_id=16)
    # Voltage
    .tuya_sensor(
        dp_id=6,
        attribute_name="voltage",
        type=t.uint32_t,
        converter=lambda x: (x[1] | (x[0] <<8)) / 10,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.VOLTAGE,
        unit=UnitOfElectricPotential.VOLT,
        translation_key="voltage",
        fallback_name="Voltage",
    )
    # Current
    .tuya_sensor(
        dp_id=6,
        attribute_name="current",
        type=t.uint32_t,
        converter=lambda x: (x[4] | (x[3] << 8)) / 1000,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.CURRENT,
        unit=UnitOfElectricCurrent.AMPERE,
        translation_key="current",
        fallback_name="Current",
    )
    # Power
    .tuya_sensor(
        dp_id=6,
        attribute_name="power",
        type=t.uint32_t,
        converter=lambda x: x[7] | (x[6] << 8),
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.POWER,
        unit=UnitOfPower.WATT,
        translation_key="power",
        fallback_name="Power",
    )
    # Energy    
    .tuya_sensor(
        dp_id=1,
        attribute_name="energy_consumed",
        type=t.uint32_t,
        divisor=100,
        state_class=SensorStateClass.TOTAL_INCREASING,
        device_class=SensorDeviceClass.ENERGY,
        unit=UnitOfEnergy.KILO_WATT_HOUR,
        fallback_name="Total energy",    
    )
    # Leakage measurement
    .tuya_sensor(
        dp_id=15,
        attribute_name="leakage",
        type=t.uint32_t,
        divisor=1,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.ENERGY,
        unit=UnitOfElectricCurrent.MILLIAMPERE,
        translation_key='power',
        fallback_name="Leakage",
    )
    # Temperature
    .tuya_temperature(dp_id=103)
    .skip_configuration()
    .add_to_registry()
)
```

Thank you for your help.

TheJulianJES, 2025-02-15:
Should be possible in HA 2025.3.0 when https://github.com/zigpy/zha-device-handlers/pull/3643 is merged.

jairbj, 2025-02-15:
> Should be possible in HA 2025.3.0 when [#3643](https://github.com/zigpy/zha-device-handlers/pull/3643) is merged.

Thank you for answering. Does the **.3** in the HA version mean March?