zigpy/zha-device-handlers #3874: [BUG] Tuya _TZE200_h4cgnbzg and _TZE200_9xfjixap error when setting local temperature calibration

CodeWanderer7331, 2025-02-16:
### Bug description

Hi dear, in Home Assistant 2025.1.4 there was introduced the newest zigpy release, which added and fixed a lot of tuya trvs native.

I noticed a problem with my TRVs which are added without custom quirk.

It's not possible to adjust local_temperature_calibration, since it throws this error:

ValueError: Failed to convert attribute local_temperature_calibration from -1 (<class 'int'>) to type <class 'zigpy.types.basic.uint32_t'>

Kind Regards

### Steps to reproduce

1. add _TZE200_h4cgnbzg or _TZE200_9xfjixap via ZHA
2. Head to Devices -> the device
3. Change Slider of Local Temperature Calibration
4. Error Message appears

### Expected behavior

No Error Message appears and value is getting set.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

[Paste the device signature here]

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

TheJulianJES, 2025-02-16:
Thanks for the issue. It'll be fixed in a future HA version, but you can already temporarily try the [this file](https://raw.githubusercontent.com/zigpy/zha-device-handlers/refs/heads/dev/zhaquirks/tuya/tuya_trv.py) as a custom quirk. Please remove it when the fix is included in HA though (likely with 2025.3.0). Duplicate issue of:
- https://github.com/zigpy/zha-device-handlers/issues/3871

TheJulianJES, 2025-02-16:
Actually, I'll leave this open for now. It's already fixed upstream, but since the fix hasn't made its way to HA yet and the device seem to be somewhat popular, other people will probably face the same issue, so let's leave it open for visibility. We can close it, as soon as the fix is in HA (and remind others to remove the custom quirk then 😄).