zigpy/zha-device-handlers #3892: [BUG] Aqara+ ZHA+ ZBT-1 FAILS; Aqara+ ZHA+ Sonoff WORKS

FloatingBoater, 2025-02-18:
### Bug description

Hi,

**TL;DR**: ZHA +  Aqara Zigbee seems to fail with Yellow/ SkyConnect/ ZBT-1 but work with a Sonoff 3.0 (TI CC2652) coordinator. 

**Anyone else seen this before I go into full logs / microscope / GitHub issue mode please?**

I recently bought a few Aqara devices from the nice folks at [EST](https://shop.everythingsmart.io/), and have been going through troubleshooting endless Zigbee issues:
* Aqara kit struggled to pair with a well populated Zigbee mesh.
  * Tried pairing 0.5m from the Zigbee coordinator.
  * Added IKEA 13A UK mains socket switches to give a very close routing device.
  * Factory reset -> Add Device -> fail, rinse, repeat.

* After pairing, all of the entity vales go to `Unavailable` greyed out
  * Pairing works, for a second, device entity values populate, then to to `Unavailable`
  * Zigbee Device **RECONFIGURE** starts, but doesn't ever complete.
  * Adding to the mesh coverage, waiting, Factory reset - no difference.
  * I tried a custom "quirks" file for the Aqara E1 curtain driver - looked visually better, but still really `Unavailable` so removed it.
  GitHub seemed to have some Aqara quirk updates in the last few months, but merged and (presumably) released.

I then repeated the tests on a pre-production HA install (spare kit, same software) with an old Sonoff Zigbee coordinator - **all my Aqara products worked first time!**
The test system is not perfect (T1 Battery `Unknown`, Curtain Luminance `0` - it is dark though!), however controls work and the main sensors report (better than `Unavailable`).

Now, I've looked about the forum and found ZHA issues, Z2M issues, specific coordinator issues, specific coordinator firmware updates, but not quite this combination of OK/ FAIL.

### Steps to reproduce

## System 1 - FAILS :x: :red_circle:
* NC Yellow +  RPiCM4
* SkyConnect / ZBT-1  Firmware: 6.10.3.0 build 297
* HAOS 14.2 / Core 2025.2.1
* ZHA       "zha": "0.0.48" /  "zhaquirks": "0.0.132"
* All controls and sensors greyed out or `Unavailable`

## System 2 - WORKS :heavy_check_mark: :green_circle:
* RPi3b (a minimal pre-production test rig, so fits memory OK)
* Sonoff 3.0 (TI CC2652)  Firmware: Z-Stack 20210708 
* HAOS 14.2 / Core 2025.2.1
* ZHA       "zha": "0.0.48" /  "zhaquirks": "0.0.132"
* AOK

Note: Both systems were tested under 0.0.48m but are not running 0.0.49.

## Aqara kit
* 2x Aqara Zigbee Curtain Driver E1 Rod version
* 1x Aqara Zigbee Temperature and Humidity Sensor T1 (also includes pressure)

### Other details
* UK 240V 50Hz, no known RF issues.
* Stable, working HAOS install.
* HACS limited to Spook, Dyson and a front-end Horizon card.
* Lots of Z-Wave, Lots of Zigbee, decent WLAN, ESPhome, Tasmota, custom kit.
* [EST](https://shop.everythingsmart.io/) deliver very quickly (when Scotland isn't in the middle of a 1:100 year storm!).
* Engineer :mage:, years of HA/ openHAB/ MisterHouse, etc.
* No hair (pulled out after too many 5-second button holds for factory reset on a precarious chair! :man_bald: :face_with_symbols_over_mouth:).
* Don't want to move from ZHA to Z2M, especially now the coordinator hardware seems to be a possible root-cause (and the Yellow has been otherwise great).
* No delays nor latency on either Zigbee mesh (so far...) so the expected interference isn't showing up.
* The test System 2 only has the 3x Aqara devices on the mesh, and paired immediately from many meters away (unlike the production System 1). Yes, there's probably more mesh traffic, but also a load more mains routers.

Thanks for any insight you may have!

### Expected behavior

ZHA works (fairly) independently of coordinator hardware:
 * Yellow/ SkyConnect/ ZBT-1
 * Sonoff 3.0 (TI CC2652)

### Screenshots/Video

Detailed debug available on request - help to narrow down test scenarios would be appreciated.

### Device signature

I decided not to include screeds of logs at this stage in case issue triage spotted a known issue, and you might be able to point at specific test scenarios.

Would be pleased to capture specifically useful scenarios - 
* Pairing with ZBT-1?
* Re-interview with ZBT-1? (can make entity states worse than initial pairing)
* Is successful pairing with the Sonoff 3.0 (TI CC2652) useful?
* Anything else?

(software / microelectronics engineer)

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Additional information

Note: Both systems were tested under 0.0.48m but are not running 0.0.49.

Would be happy to re-test with debug logging under 0.0.49.

The Yellow SilLabs is my production system, the Sonoff a test rig (although suspect you'll need SilLabs logs form the Yellow).

FloatingBoater, 2025-03-13:
After some research, the Yellow / ZBT1 / Skyconnect coordinator appears to NOT upgrade firmware automatically, and the shipped 6.10.3.0 build 297 is very old compared to the current 7.4.0.0, and from my testing DOES NOT WORK with Aqara kit in my home.

**ISSUE: Many Nabu Casa Yellow devices shipped with the `6.10.3.0` firmware and may be unaware the device requires an upgrade.**

Here's more detail, and a Step-by-step how-to upgrade:
https://community.home-assistant.io/t/aqara-zha-zbt-1-fails-aqara-zha-sonoff-works-why/844159/2

## FAILS with all Aqara kit tested
INFO Detected ApplicationType.EZSP, version '6.10.3.0 build 297' (6.10.3.0.297) at 115200 baudrate (bootloader baudrate 115200)
## WORKS as expected, and as the other Sonoff 3.0 coordinator
INFO Detected ApplicationType.EZSP, version '7.4.4.0 build 0' (7.4.4.0.0) at 1152

TheJulianJES, 2025-04-01:
The next HA version (2025.4.0), releasing later today/tomorrow, should allow you to update the ZBT-1 (and other radios used in Nabu Casa devices) directly from Home Assistant Core.

closed 2025-10-07 by github-actions[bot]