zigpy/zha-device-handlers #3895: [Device Support Request]  TS0601 TZE204 fan and light switch

SilverDante, 2025-02-20:
### Problem description

I have been able to successfully add the ZigBee device to HA, but it does not give me anything in the controls section. 

The device info is "TS0601
by _TZE204_lawxy9e2"

### Solution description

It seems to need a control for the light. The on/off/toggle. And a control for the fan, on/off/+/-.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

[Paste the device signature here]

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

AlexZT2000, 2025-06-20:
"""Quirk for _TZE204_lawxy9e2 TS0601."""

from zigpy.quirks.v2 import EntityType
import zigpy.types as t

from zhaquirks.tuya.builder import TuyaQuirkBuilder

class TuyaSpeedFan(t.enum8):
    
    level_1 = 0x00
    level_2 = 0x01
    level_3 = 0x02
    level_4 = 0x03
    level_5 = 0x04
    
class TuyaLightMode(t.enum8):
    
    none = 0x00
    relay = 0x01
    pos = 0x02

(
    TuyaQuirkBuilder("_TZE204_lawxy9e2", "TS0601")
    .tuya_switch(
        dp_id=1,
        attribute_name="on_off_1",
        entity_type=EntityType.STANDARD,
        translation_key="on_off_1",
        fallback_name="Switch 1",
    )
    .tuya_switch(
        dp_id=5,
        attribute_name="on_off_2",
        entity_type=EntityType.STANDARD,
        translation_key="on_off_2",
        fallback_name="Switch 2",
    )
    .tuya_enum(
        dp_id=3,
        attribute_name="Speed_Fan",
        enum_class=TuyaSpeedFan,
        translation_key="Speed_Fan",
        fallback_name="Speed fan",
    )
    .tuya_enum(
        dp_id=12,
        attribute_name="Ligth_Mode",
        enum_class=TuyaLightMode,
        translation_key="Ligth_Mode",
        fallback_name="Ligth mode",
    )
    .skip_configuration()
    .add_to_registry()
)

AlexZT2000, 2025-06-20:
I searched for a long time myself, using trial and error I wrote Quirk for v2 It's work

AlexZT2000, 2025-12-15:
I don't use Z2MQTT, I can't help, I only use ZHA. In ZHA light turns on. I wrote the code above.

.tuya_enum(
dp_id=12,
attribute_name="Ligth_Mode",
enum_class=TuyaLightMode,
translation_key="Ligth_Mode",
fallback_name="Ligth mode",
)

rrooggiieerr, 2026-01-16:
Also, there's a typo in your entity names, Lig**th**_Mode should be Lig**ht**_Mode