zigpy/zha-device-handlers #3903: [BUG] The mode of the GL-C-003P controller is incorrect

Aogu181, 2025-02-25:
### Bug description

In ZHA GL-C-003P mode is CCT not RGB.

please look : https://github.com/Koenkk/zigbee2mqtt/issues/24091

### Steps to reproduce

1. Switch the indicator light to blue
2. connect

### Expected behavior

It should be displayed in RGB mode

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

[Paste the device signature here]

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

Vitaliy86, 2025-07-20:
Problem not solved

seflm, 2025-12-27:
Having the same issue.