zigpy/zha-device-handlers #3914: [BUG] issue with tuya_rain ZHA quirk

pragyandeb, 2025-03-02:
### Bug description

Hello. I am trying to use the Tuya ZHA quirk for the rain sensor. However, it’s not working and checking logs, I get the following error. Grateful for any help. 

Logger: zhaquirks
Source: /usr/local/lib/python3.13/site-packages/zhaquirks/__init__.py:481
First occurred: 12:07:15 PM (1 occurrences)
Last logged: 12:07:15 PM

Unexpected exception importing custom quirk 'tuya_rain'
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/zhaquirks/__init__.py", line 479, in setup
    spec.loader.exec_module(module)
    ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_zha_quirks/tuya_rain.py", line 61, in <module>
    .tuya_dp_multi(
     ^^^^^^^^^^^^^
AttributeError: 'TuyaQuirkBuilder' object has no attribute 'tuya_dp_multi'

### Steps to reproduce

Logger: zhaquirks
Source: /usr/local/lib/python3.13/site-packages/zhaquirks/__init__.py:481
First occurred: 12:07:15 PM (1 occurrences)
Last logged: 12:07:15 PM

Unexpected exception importing custom quirk 'tuya_rain'
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/zhaquirks/__init__.py", line 479, in setup
    spec.loader.exec_module(module)
    ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_zha_quirks/tuya_rain.py", line 61, in <module>
    .tuya_dp_multi(
     ^^^^^^^^^^^^^
AttributeError: 'TuyaQuirkBuilder' object has no attribute 'tuya_dp_multi'

### Expected behavior

Quirk not working. 

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

[Paste the device signature here]

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

TheJulianJES, 2025-03-04:
They’re included in HA 2025.3.0b0 and later. Currently, still in beta but already seems stable. Full release should be out in about two days.

alainprofe, 2025-03-15:
I left this comment as a comment to the conflicting line:

As someone said in issues conversation (#3914), yes, there is also a bug for me.
I'm a programmer so I speak humbly but with some knowledge. I'm just starting in HA world.
I added this quirk for ts0601 _TZE204_qyr2m29i TRV valves ("MOES Tuya whatever). It didn't work and gave me the same "AttributeError: 'TuyaQuirkBuilder' object has no attribute 'tuya_dp_multi'"as mentioned before.
In a desperate move I commented this whole tuya_dp_multi block and... IT WORKS! I'm not using system modes because I feel them funny, but I'm just starting to use this kind of TRV so I'm not fully able to say whether this is an abnormal situation. But at least the basics do work. I have thermostats and I can change the target temperature and most if not all of the other parameters.
As a programmer, I feel like maybe this tuya_dp_multi is something about a newer version of TuyaQuirkBuilder that my installation doesn't have. It's a one month old, updated installation on a Docker container, on Raspberry Pi OS, on a RasPi 5.
THANKS A LOT FOR ALL THE EFFORT! 😃

closed 2025-06-24 by TheJulianJES