zigpy/zha-device-handlers #3915: [Device Support Request] Fingerbot _TZ3210_j4pdtz9v TS0001 USB-C

antoniovj1, 2025-03-03:
### Problem description

## Issue with the mode of the _TZ3210_j4pdtz9v TS0001 device

Hi! I'm trying to update the mode of the **_TZ3210_j4pdtz9v TS0001** device, but it doesn't work.

The device is recognized, and I can see that the mode attribute is written and can be read properly. However, the movement behavior doesn't change.

### Device signature:

<details>
  <summary>View the signature JSON</summary>

  ```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0006",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3210_j4pdtz9v",
  "model": "TS0001",
  "class": "zhaquirks.tuya.builder.EnchantedDeviceV2"
}
```
</details>

[image]

### Solution description

The mode is changed in the device

### Screenshots/Video

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 2,
[... 928 lines cut ...]
}

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

https://github.com/zigpy/zha-device-handlers/blob/b87ac8907765303c915b19d915a1ca5ea9a8bdae/zhaquirks/tuya/tuya_fingerbot.py#L27

</details>

jack5mikemotown, 2025-03-08:
This device [*should* be supported](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/tuya_fingerbot.py) so i'm not working. but as such, i think this would be a bug

TheJulianJES, 2025-03-08:
Update to 2025.3.0 (or later) and try again with the built-in entities.

pha0001, 2025-03-08:
> Hi! I'm trying to update the mode of the **_TZ3210_j4pdtz9v TS0001** device, but it doesn't work.
> 
> The device is recognized, and I can see that the mode attribute is written and can be read properly. However, the movement behavior doesn't change.

@antoniovj, I suspected that if you remove the battery from the TZ3210 device and then put it back in, a subsequent read attribute request would show that it has reverted back to FingerBotMode.CLICK. This is exactly what happened to me.

pha0001, 2025-03-08:
> Update to 2025.3.0 (or later) and try again with the built-in entities.

I’m experiencing the same problem with two out of my three Tuya FingerBot devices.

My HA is running the latest version, 2025.3.1, and I have tested it with the new built-in entities, but the issue persists.

It seems to be a widespread issue with Tuya FingerBot devices. Does anyone else have this problem where the device fails to register a change request from FingerBotMode.CLICK to FingerBotMode.SWITCH?

jack5mikemotown, 2025-03-08:
> Does anyone else have this problem where the device fails to register a change request from FingerBotMode.CLICK to FingerBotMode.SWITCH?

For me all possible configuration options fail to have any effect on the devices physical functions despite the changes being accurately displayed in the Zigbee cluster attributes. I'm also on 2025.3.0 and later in testing this device

antoniovj1, 2025-03-08:
@jack5mikemotown I have exactly tha same issue. Changes appears to be saved properly but not doing anything.

@TheJulianJES Feel free to ask for anything that can help to debug the issue 👍

aleksey925, 2025-03-14:
Hi, I use Home Assistant 2025.3.1, and I have encountered the same issue.
I bought https://www.aliexpress.com/item/1005007233648891.html?spm=a2g0o.order_list.order_list_main.5.f9341802XgyJ47; it was recognized as `TS0001` by _TZ3210_j4pdtz9v, and I have a default menu for customizing the device mode, but the changes do not have any effect. The device always makes a press and immediately lifts its finger.

Signature
```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0006",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3210_j4pdtz9v",
  "model": "TS0001",
  "class": "zhaquirks.tuya.builder.EnchantedDeviceV2"
}
```

[image]

[image]

aleksey925, 2025-03-14:
If you need any assistance from me to debug the device, write to me, I will help as much as I can.

Nafania, 2025-03-18:
Same issue here, while with tuya zigbee gateway and and tuya app it works and options can be used properly

philiplb, 2025-03-23:
Having the same Issue with a TS0001.

TheJulianJES, 2025-03-26:
Should hopefully be fixed in the HA beta later today/tomorrow: 2025.4.0b0 (and later).

antoniovj1, 2025-03-27:
Since the update the buttons seems to be disabled:

[image]

This is the only change I noticed, but the problem persists. Here is the diagnostic information:

<details>
  <summary>Diagnostic JSON</summary>

  ```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2025.4.0b1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.13.2",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Madrid",
    "os_name": "Linux",
    "os_version": "6.6.74-haos-raspi",
    "supervisor": "2025.03.3",
    "host_os": "Home Assistant OS 15.0",
    "docker_version": "28.0.1",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "xiaomi_home": {
      "documentation": "https://github.com/XiaoMi/ha_xiaomi_home/blob/main/README.md",
      "version": "v0.2.3",
      "requirements": [
        "construct>=2.10.56",
        "paho-mqtt",
        "numpy",
        "cryptography",
        "psutil"
      ]
    },
    "meross_cloud": {
      "documentation": "https://www.home-assistant.io/components/meross_cloud",
      "version": "1.3.6",
      "requirements": [
        "meross_iot==0.4.9.0"
      ]
    },
    "hacs": {
      "documentation": "https://hacs.xyz/docs/use/",
      "version": "2.0.5",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "after_dependencies": [
      "hassio",
      "onboarding",
      "usb"
    ],
    "codeowners": [
      "dmulcahey",
      "adminiuga",
      "puddly",
      "TheJulianJES"
    ],
[... 806 lines cut ...]
                "attribute": "ZCLAttributeDef(id=0x0001, name='time_status', type=<flag 'TimeStatus'>, zcl_type=<DataTypeId.map8: 24>, access=<ZCLAttributeAccess.Read|Write_Optional: 5>, mandatory=True, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0002": {
                "attribute": "ZCLAttributeDef(id=0x0002, name='time_zone', type=<class 'zigpy.types.basic.int32s'>, zcl_type=<DataTypeId.int32: 43>, access=<ZCLAttributeAccess.Read|Write: 3>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              },
              "0x0009": {
                "attribute": "ZCLAttributeDef(id=0x0009, name='valid_until_time', type=<class 'zigpy.types.named.UTCTime'>, zcl_type=<DataTypeId.UTC: 226>, access=<ZCLAttributeAccess.Read|Write: 3>, mandatory=False, is_manufacturer_specific=False)",
                "value": null
              }
            },
            "unsupported_attributes": []
          }
        }
      }
    }
  }
}
```
</details>

Can I provide any useful information?

thanks!

TheJulianJES, 2025-03-27:
Thanks! I've just had another look and I made an obvious mistake: https://github.com/zigpy/zha-device-handlers/pull/3980#discussion_r2015938047

I'll take a look at fixing this tomorrow or so.

TheJulianJES, 2025-03-27:
Hmm, not sure that would cause the entities to not register at all though. So yeah, interested in any errors/warnings in your logs. Feel free to PM them to me on Discord or send via email if you don't wanna post them publicly here.

antoniovj1, 2025-03-27:
@TheJulianJES I have removed the device to get fresh logs when adding it, and now the UI is enabled. I don't see any error logs. I might have done something wrong the first time, but the issue is still there. Are the logs still useful for you?

TheJulianJES, 2025-03-27:
Probably not for now, thanks. Just to confirm, the entities still don't work, right?

Anima1337, 2025-03-31:
Finally I found someone with the exact same behaviour! D:
If I can help with any logs whatsoever, please tell me. I got the exact same issue. I cannot even overwrite the quirk "zhaquirks.tuya.builder.EnchantedDeviceV2" to try and fix it myself. Moe's (aka Tuya) Zigbee Fingerbot. 

[image]

[image]

[image]

psromsch, 2025-04-01:
Same Issue with a TS0001.

bencolemann, 2025-04-01:
Issue not fixed with the latest update, same behaviour, says it's written but no change

closed 2025-04-01 by TheJulianJES

Anima1337, 2025-04-02:
Issue was fixed in Beta 2025.4.0b15
Thank you for your contribution TheJulianJES! :)

antoniovj1, 2025-04-03:
Yeah!! It's working perfectly! Thank you very much @TheJulianJES 🎉🎉🎉

pha0001, 2025-04-03:
Legend! Thank-you

mg-beep, 2025-04-04:
Thanks a lot, it worked!!!

rcuijpers, 2025-04-05:
Oh lord I need this fix too! How does one install the core beta? Or should I wait a few days?

TheJulianJES, 2025-04-05:
[2025.4.0](https://www.home-assistant.io/blog/2025/04/02/release-20254/) is already released, no beta needed anymore.

rcuijpers, 2025-04-05:
Oh mannnnn, got it. You made my evening. 🙏🏻

ryrun, 2025-04-28:
> Finally I found someone with the exact same behaviour! D: If I can help with any logs whatsoever, please tell me. I got the exact same issue. I cannot even overwrite the quirk "zhaquirks.tuya.builder.EnchantedDeviceV2" to try and fix it myself. Moe's (aka Tuya) Zigbee Fingerbot.

Is there a way to properly read the battery status? I have the exact same Fingerbot.