zigpy/zha-device-handlers #3917: Tuya TRV Setpoint setting is broken

therealprof, 2025-03-03:
Testing the current quirks on my Nedis TRV I noticed that the thermostat setting using the thermostat UI in ZHA doesn't work, any attempt to change the temperature makes it jump to 160C and issue a warning that this temperature is out of range:

[image]

Comparing with a configuration from the forum (https://community.home-assistant.io/t/nedis-smartlife-zbhtr20wt-not-supported/796424/14), it turns out that these two lines are wrong:

https://github.com/zigpy/zha-device-handlers/blob/b87ac8907765303c915b19d915a1ca5ea9a8bdae/zhaquirks/tuya/tuya_trv.py#L116-L117

Should be this

```
-        Thermostat.AttributeDefs.abs_min_heat_setpoint_limit.id: 500,
-        Thermostat.AttributeDefs.abs_max_heat_setpoint_limit.id: 3000,
+        Thermostat.AttributeDefs.min_heat_setpoint_limit.id: 500,
+        Thermostat.AttributeDefs.max_heat_setpoint_limit.id: 3000,
```

instead.

TheJulianJES, 2025-03-04:
According to ZCL spec, the "abs" values are absolute min/max values (normally set by the manufacturer).
The non-abs values are meant to be changed by the user. That's why we create configuration entities for them, if supported. But they're not actually user configurable for this device, so we correctly use the absolute values for the limits.

Did you use a custom quirk before (that set the non-abs values)? That would cause HA to go into this buggy behavior, as the non-abs values will now be set to some unexpected values (and not be `None`).

Please revert your changes and read the `min_heat_setpoint_limit` and `max_heat_setpoint_limit` attributes through the debug UI on the device page: three dots on the left -> Manage Zigbee device -> Clusters -> select thermostat cluster UI -> select attribute -> press read, wait and note the output, then select the other attribute and press read again.

You can also try to completely remove the device from ZHA/HA and then re-pair it (with no custom quirks). That'll cause the attribute cache to be cleared and reinitialized for the device.

TheJulianJES, 2025-03-04:
Also, you are currently testing this with Home Assistant Core 2025.3.0 betas, right? And did not modify the dependencies manually?

therealprof, 2025-03-04:
I'm using the docker image with Home Assistant Core 2025.2.5 with a manually installed (via pip3) zha-device-handlers-0.0.134.tar.gz package.

The read value are:
min_heat_setpoint_limit (id: 0x0015) -> 20000
max_heat_setpoint_limit (id: 0x0016) -> 16000

... and the UI has reverted to being broken again, even after reading the values. I did indeed experiment with a custom quirk before but that was based on the included zha-quirks in Core 2025.2.5 which was very different and did never properly work so it's unlikely I had those variables before.

Here's the debug output from before:

[zha-01JMF0VNMCQRC7B9FTDY8VATZV-_TZE284_ne4pikwm TS0601-4ed2d4aa3edbdac082002a17425dd8d9.json](https://github.com/user-attachments/files/19062910/zha-01JMF0VNMCQRC7B9FTDY8VATZV-_TZE284_ne4pikwm.TS0601-4ed2d4aa3edbdac082002a17425dd8d9.json)

However I've just re-paired the TRV and now it's working properly, so that's that (i.e. the reading of the values did somehow not clear it). Anyway here's another debug output from after the re-pairing:

[zha-01JMF0VNMCQRC7B9FTDY8VATZV-_TZE284_ne4pikwm TS0601-4ed2d4aa3edbdac082002a17425dd8d9 (1).json](https://github.com/user-attachments/files/19062941/zha-01JMF0VNMCQRC7B9FTDY8VATZV-_TZE284_ne4pikwm.TS0601-4ed2d4aa3edbdac082002a17425dd8d9.1.json)

I guess this issue can be closed then. Sorry about the noise... 🙏🏻

TheJulianJES, 2025-03-04:
Thanks for the different files. We'll leave it open for now. There is one change we could do to automatically clear old attribute state.

But I'm not sure where the `20000` and `16000` values come from. The values should return `None`.. I'm hoping they come from some old quirk and not from the device somehow. If you read the values now, after re-pairing with the original/new quirk, is `None` returned for the non-abs values now?

therealprof, 2025-03-04:
Now they return `None`.