zigpy/zha-device-handlers #3929: [Device Support Request] TS110E by _TZ3210_k1msuvg6 - on/off works, not the dimmer level

AlenHodnik, 2025-03-04:
### Problem description

Device TS110E by _TZ3210_k1msuvg6 (dimmer with neutral, 1 gang) is found in ZHA, but not dimmer function. It doesn't load ts110e.py quirk. Is there a way to make dimmer function work?

### Solution description

Dimmer function to work.

### Screenshots/Video

[image]

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0101",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3210_k1msuvg6",
  "model": "TS110E",
  "class": "zigpy.device.Device"
}

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

mesut3000, 2025-03-09:
Hi,

I also have TS110E - by _TZ3210_ngqk6jia Tuya dimmer.
On/off works but dimming doesn't respond.

bennierex, 2025-04-03:
Same thing here; the device is reported in ZHA as `TS110E
by _TZ3210_pagajpog`

Ulf-Yngwe, 2025-04-05:
Xactly the same for me with _TZ3210_3mpwqzuu