zigpy/zha-device-handlers #3938: [Device Support Request]  TS0201 by _TZ3000_dowj6gyi

ajbathe, 2025-03-07:
### Problem description

Hello,

I bought this Tuya (compatible) temperature & humidity sensor:
https://de.aliexpress.com/item/1005007755890938.html

Device info is "TS0201 by  _TZ3000_dowj6gyi"

I haven't found the string  "_TZ3000_dowj6gyi" within the source of zha-device-handlers. Could you please add this to the Tuya TS0201 devices? Thanx alot.

Andreas

### Solution description

see screenshot

### Screenshots/Video

[image]

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

[Paste the device signature here]

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

lxvdev, 2025-09-09:
I bought the exact same device (hasn't arrived yet), is it compatible with ZHA?

samtrot, 2025-12-25:
Same for me - would love to see it added.