zigpy/zha-device-handlers #3944: [Device Support Request] TNCE RMDZB-1PNL63 - TS0601 _TZE204_m64smti7

Koterak, 2025-03-09:
### Problem description

Zigbee DIN single phase energy meter - unfortunatly quirk not available. 

I have code from zigbee2mqtt:

`    {
        fingerprint: tuya.fingerprint("TS0601", ["_TZE204_m64smti7"]),
        model: "RMDZB-1PNL63",
        vendor: "TNCE",
        description: "Zigbee DIN single phase energy meter",
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        configure: tuya.configureMagicPacket,
        exposes: [
            tuya.exposes.switch(),
            e.energy(),
            e
                .enum("fault", ea.STATE, [
                    "clear",
                    "short_circuit_alarm",
                    "surge_alarm",
                    "overload_alarm",
                    "leakagecurr_alarm",
                    "temp_dif_fault",
                    "fire_alarm",
                    "high_power_alarm",
                    "self_test_alarm",
                    "ov_cr",
                    "unbalance_alarm",
                    "ov_vol",
                    "undervoltage_alarm",
                    "miss_phase_alarm",
                    "outage_alarm",
                    "magnetism_alarm",
                    "credit_alarm",
                    "no_balance_alarm",
                ])
                .withDescription("Fault status of the device (clear = nothing)"),
            tuya.exposes.voltageWithPhase("a"),
            tuya.exposes.powerWithPhase("a"),
            tuya.exposes.currentWithPhase("a"),
            e.temperature(),
            e
                .binary("over_current_breaker", ea.STATE_SET, "ON", "OFF")
                .withDescription("OFF - alarm only, ON - relay will turn off when threshold reached"),
            e
                .numeric("over_current_threshold", ea.STATE_SET)
                .withUnit("A")
                .withDescription("Setup the value on the device")
                .withValueMin(1)
                .withValueMax(63),
            e
                .binary("over_voltage_breaker", ea.STATE_SET, "ON", "OFF")
                .withDescription("OFF - alarm only, ON - relay will turn off when threshold reached"),
            e
                .numeric("over_voltage_threshold", ea.STATE_SET)
                .withUnit("V")
                .withDescription("Setup value on the device")
                .withValueMin(250)
                .withValueMax(300),
            e
                .binary("under_voltage_breaker", ea.STATE_SET, "ON", "OFF")
                .withDescription("OFF - alarm only, ON - relay will turn off when threshold reached"),
            e
                .numeric("under_voltage_threshold", ea.STATE_SET)
                .withUnit("V")
                .withDescription("Setup value on the device")
                .withValueMin(150)
                .withValueMax(200),
[... 806 lines cut ...]
</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

frankii91, 2025-03-19:
I need this too, I bought 50 pieces and it would come in handy....

frankii91, 2025-03-29:
I have copies or very similar ones as in the photo.

However, I am new and I do not know if it is a bug or not working properly because after restarting Home Assistant does not remember the state of the main switch, it sees "?"

**"state": null**

`
{
    "current_a": 0,
    "energy": 0.01,
    "high_temperature_breaker": "OFF",
    "high_temperature_threshold": 80,
    "last_seen": "2025-03-29T23:36:39.030Z",
    "linkquality": 21,
    "over_current_breaker": "OFF",
    "over_current_threshold": 16,
    "over_voltage_breaker": "OFF",
    "over_voltage_threshold": 275,
    "power_a": 0,
    "temperature": 21,
    "under_voltage_breaker": "OFF",
    "under_voltage_threshold": 175,
    "voltage_a": 229.3,
    "device": {
        "applicationVersion": 74,
        "dateCode": "",
        "hardwareVersion": 1,
        "ieeeAddr": "0xa4c1380a1dabb904",
        "manufacturerID": 4417,
        "manufacturerName": "_TZE204_m64smti7",
        "model": "RMDZB-1PNL63",
        "networkAddress": 21323,
        "powerSource": "Mains (single phase)",
        "stackVersion": 0,
        "type": "Router",
        "zclVersion": 3
    },
    "fault": null,
    "state": null
}
`

111lisu, 2025-09-09:
Has anyone managed to get this thing working? The relay works but the rest is unavailable.

 ZBN-JT-63
```
"""Tuya TNCE ZBN-JT-63 Zigbee DIN single phase energy meter quirk for ZHA v2."""

import logging
from typing import Final, Any, Dict
from zigpy.quirks.v2 import QuirkBuilder
import zigpy.types as t
from zigpy.zcl.foundation import ZCLAttributeDef
from zigpy.zcl.clusters.general import OnOff
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.measurement import TemperatureMeasurement
from zigpy.zcl.clusters.smartenergy import Metering
from zigpy.quirks.v2.homeassistant import EntityPlatform, EntityType
from zhaquirks.tuya import (
    TuyaLocalCluster,
    TuyaPowerConfigurationCluster2AA,
)
from zhaquirks.tuya.mcu import TuyaMCUCluster, DPToAttributeMapping

_LOGGER = logging.getLogger(__name__)

class FaultStatus(t.enum32):
    """Fault status values."""
    clear = 0x00
    short_circuit_alarm = 0x01
    surge_alarm = 0x02
    overload_alarm = 0x04
    leakagecurr_alarm = 0x08
    temp_dif_fault = 0x10
    fire_alarm = 0x20
    high_power_alarm = 0x40
    self_test_alarm = 0x80
    ov_cr = 0x100
    unbalance_alarm = 0x200
    ov_vol = 0x400
    undervoltage_alarm = 0x800
    miss_phase_alarm = 0x1000
    outage_alarm = 0x2000
    magnetism_alarm = 0x4000
    credit_alarm = 0x8000
    no_balance_alarm = 0x10000

class TuyaEnergyMeterOnOffCluster(OnOff, TuyaLocalCluster):
    """Tuya local OnOff cluster for relay control."""
    
    async def command(
        self,
        command_id,
        *args,
        manufacturer=None,
        expect_reply=True,
        tsn=None,
        **kwargs,
    ):
        """Handle on/off commands."""
        _LOGGER.debug("OnOff command: %s, args: %s", command_id, args)
        
        if command_id in (0x00, 0x01, 0x02):  # Off, On, Toggle
            if command_id == 0x00:  # Off
                switch_value = False
            elif command_id == 0x01:  # On  
                switch_value = True
            else:  # Toggle
                current_state = self.get("on_off", 0)
                switch_value = not bool(current_state)
            
            # Send to MCU cluster
[... 395 lines cut ...]
        TuyaEnergyMeterMCU.cluster_id,
        fallback_name="Under Voltage Breaker",
        translation_key="under_voltage_breaker",
    )
    # High temperature threshold
    .number(
        TuyaEnergyMeterMCU.AttributeDefs.high_temperature_threshold.name,
        TuyaEnergyMeterMCU.cluster_id,
        step=1,
        min_value=40,
        max_value=100,
        unit="°C",
        fallback_name="High Temperature Threshold",
        translation_key="high_temperature_threshold",
    )
    # High temperature breaker
    .switch(
        TuyaEnergyMeterMCU.AttributeDefs.high_temperature_breaker.name,
        TuyaEnergyMeterMCU.cluster_id,
        fallback_name="High Temperature Breaker",
        translation_key="high_temperature_breaker",
    )
    .add_to_registry()
)
```