zigpy/zha-device-handlers #3950: [Device Support Request] TS000F Zigbee switch with humidity sensor Manf: _TZ3218_7fiyo3kv

Kuro507, 2025-03-11:
### Problem description

Device is discovered in ZHA ok, but its only seen as a light switch. In fact this particular model has a external Humidity sensor probe. (I believe same model is available with temperature probe instead)

More features are show in the app [https://community.home-assistant.io/t/zigbee-temperature-sensor-switch-only-detected-as-light/748950/28?u=blue407](url)

### Solution description

Please create device support for this model with all the other available data fields, also change from light switch to normal switch.

Some people have already created custom files for this, so maybe referencing these will help [https://community.home-assistant.io/t/zigbee-temperature-sensor-switch-only-detected-as-light/748950](url)

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
[... 830 lines cut ...]
<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

### Additional information

Thanks in advance for this, happy to test it once a handler has been created :)

Kuro507, 2025-04-14:
Any news on this issue?

McGyver88, 2025-04-16:
I would love to know as well. I just bought one of these to replace the thermostat on my older hot tub. I'm learning how to make a custom quirk in the mean time.

Kuro507, 2025-04-22:
> I would love to know as well. I just bought one of these to replace the thermostat on my older hot tub. I'm learning how to make a custom quirk in the mean time.

Could you share your workaround, if you get it working?

closed 2025-10-27 by github-actions[bot]