zigpy/zha-device-handlers #3957: [Device Support Request] Aqara Light Switch H2 EU

kokrp, 2025-03-12:
### Problem description

Hi requesting to enhance support for the Aqara Light Switch H2 EU [(link)](https://www.aqara.com/eu/product/light-switch-h2-eu/).

_First time posting something like this. I hope I got it right._

[image]

I can add it to ZHA and it's recognized as lumi.switch.agl010, but seems there are issues with the current config/support:

Power reading has ~40sec delay, not sure if intended/expected
Sumation delivered is stuck on 0.0 kWh
Bottom switches are not visible in HA

Only exposed entities are: switches (1, 2), Power, Summation delivered.
Looking at the Zigbee2MQTT [page](https://www.zigbee2mqtt.io/devices/WS-K08E.html) it seems there are many other available options: current, energy, led_indicator, flip_indicator_light, switch (state), device_temperature, power_outage_count, power_on_behavior, operation_mode, lock_relay, multi_click, action

Home Assistent version:
Core: 2025.3.2
Supervisor: 2025.03.2
Operating System: 14.2
Frontend: 20250306.0

### Solution description

Full support of device/fix existing bugs. Possibly mirror Zigbee2MQTT level of support and options.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4447,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0012",
        "0x0702",
        "0x0b04",
        "0xfcc0"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0012",
        "0xfcc0"
      ],
      "output_clusters": []
    },
    "3": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0xfcc0"
      ],
      "output_clusters": []
    },
    "4": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0012",
        "0xfcc0"
      ],
      "output_clusters": []
    },
    "5": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0012",
        "0xfcc0"
      ],
      "output_clusters": []
    },
    "21": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x000c"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "Aqara",
  "model": "lumi.switch.agl010",
  "class": "zigpy.device.Device"
}

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

Niek, 2025-03-15:
I bought the exact same device and have the same issue. Only the top (physical relay) part of the button is exposed, not the bottom.

Niek, 2025-03-17:
See below the signature of the [Aqara WS-K07E](https://www.zigbee2mqtt.io/devices/WS-K07E.html#aqara-ws-k07e), the one with only 1 button:
```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4447,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0012",
        "0x0702",
        "0x0b04",
        "0xfcc0"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0xfcc0"
      ],
      "output_clusters": []
    },
    "3": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0xfcc0"
      ],
      "output_clusters": []
    },
    "4": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0012",
        "0xfcc0"
      ],
      "output_clusters": []
    },
    "5": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0xfcc0"
      ],
      "output_clusters": []
    },
    "21": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x000c"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "Aqara",
  "model": "lumi.switch.agl009",
  "class": "zigpy.device.Device"
}
```

kokrp, 2025-04-03:
@Niek , I'm a newby and don't know what your remark above means. I guess it means the relevant code is available but not yet 'migrated' to ZHA? Thanks in advance for the help 😊

zeratul021, 2025-04-22:
I can offer help with alpha/beta testing. O have WS-K08D)

ftheomunhoz, 2025-05-01:
I'm down to helping this as well, let me know if you need it. I have H2 and I'm pretty interested in seeing this in.

lonevvolf, 2025-05-18:
Based on the PR for the quirk for the US version [here](https://github.com/zigpy/zha-device-handlers/pull/4008), I have extended it with support for the H2 WS-K07D.  It seems to work for me, but I don't want to add it to the current PR until the US one is done.  In any case, here's the quirk code if you want to try it out.  I'm not sure why my switch is not reporting Scenes or Groups support, but it isn't at the moment.  I was able to automate the bottom button to toggle the light, the same as the top one, so guests can press either one.

```
"""Aqara H2 single rocker switch quirks. Also see opple_switch.py for similar switches."""

from zigpy.profiles import zha
from zigpy.zcl.clusters.general import (
    AnalogInput,
    Basic,
    Groups,
    Identify,
    MultistateInput,
    OnOff,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks.const import (
    ARGS,
    ATTR_ID,
    BUTTON,
    BUTTON_1,
    BUTTON_2,
    CLUSTER_ID,
    COMMAND_DOUBLE,
    COMMAND_HOLD,
    COMMAND_RELEASE,
    COMMAND_SINGLE,
    DEVICE_TYPE,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PRESS_TYPE,
    PROFILE_ID,
    VALUE,
    ZHA_SEND_EVENT,
)
from zhaquirks.xiaomi import (
    AnalogInputCluster,
    BasicCluster,
    MeteringCluster,
    OnOffCluster,
    XiaomiCustomDevice,
)
from zhaquirks.xiaomi.aqara.opple_remote import MultistateInputCluster
from zhaquirks.xiaomi.aqara.opple_switch import OppleSwitchCluster

class PowerMeasurementCluster(AnalogInputCluster):
    """Custom cluster for power measurement."""

    PRESENT_VALUE_ATTRIBUTE = 0x0055

    async def bind(self):
        """Bind cluster."""
        result = await super().bind()
        await self.configure_reporting(
            self.PRESENT_VALUE_ATTRIBUTE,
            0,  # minimum reporting interval
            600,  # maximum reporting interval
            1,  # reportable change
        )
        return result

    def _update_attribute(self, attrid, value):
        super()._update_attribute(attrid, value)
[... 457 lines cut ...]
            },
            21: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    PowerMeasurementCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }

    device_automation_triggers = {
        (COMMAND_SINGLE, BUTTON_1): {
            ENDPOINT_ID: 1,
            CLUSTER_ID: 18,
            ARGS: {BUTTON: 1, ATTR_ID: 0x0055, PRESS_TYPE: COMMAND_SINGLE, VALUE: 1},
        },
        (COMMAND_SINGLE, BUTTON_2): {
            ENDPOINT_ID: 2,
            CLUSTER_ID: 18,
            ARGS: {BUTTON: 2, ATTR_ID: 0x0055, PRESS_TYPE: COMMAND_SINGLE, VALUE: 1},
        },
    }
```

lonevvolf, 2025-06-08:
Just received my two button version, I'll try to install it today and add it to the quirk.

crypto512, 2025-06-08:
> Just received my two button version, I'll try to install it today and add it to the quirk.

Great to know , i’ve currently the two button and ordered the one button. I hope a final version will be available soon 🙏

lonevvolf, 2025-06-08:
This seems to work for both EU versions.  Since it looks liked the other PR is stalled, I will try to have this added as a complete first version later on.

```
"""Aqara H2 single rocker switch quirks. Also see opple_switch.py for similar switches."""

from zigpy.profiles import zha
from zigpy.zcl.clusters.general import (
    AnalogInput,
    Basic,
    Groups,
    Identify,
    MultistateInput,
    OnOff,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks.const import (
    ARGS,
    ATTR_ID,
    BUTTON,
    BUTTON_1,
    BUTTON_2,
    BUTTON_3,
    CLUSTER_ID,
    COMMAND_DOUBLE,
    COMMAND_HOLD,
    COMMAND_RELEASE,
    COMMAND_SINGLE,
    DEVICE_TYPE,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PRESS_TYPE,
    PROFILE_ID,
    VALUE,
    ZHA_SEND_EVENT,
)
from zhaquirks.xiaomi import (
    AnalogInputCluster,
    BasicCluster,
    MeteringCluster,
    OnOffCluster,
    XiaomiCustomDevice,
)
from zhaquirks.xiaomi.aqara.opple_remote import MultistateInputCluster
from zhaquirks.xiaomi.aqara.opple_switch import OppleSwitchCluster

class PowerMeasurementCluster(AnalogInputCluster):
    """Custom cluster for power measurement."""

    PRESENT_VALUE_ATTRIBUTE = 0x0055

    async def bind(self):
        """Bind cluster."""
        result = await super().bind()
        await self.configure_reporting(
            self.PRESENT_VALUE_ATTRIBUTE,
            0,  # minimum reporting interval
            600,  # maximum reporting interval
            1,  # reportable change
        )
        return result

    def _update_attribute(self, attrid, value):
[... 642 lines cut ...]
            CLUSTER_ID: 18,
            ARGS: {BUTTON: 4, ATTR_ID: 0x0055, PRESS_TYPE: COMMAND_RELEASE, VALUE: 255},
        },
        (COMMAND_SINGLE, BUTTON_3): {
            ENDPOINT_ID: 5,
            CLUSTER_ID: 18,
            ARGS: {BUTTON: 5, ATTR_ID: 0x0055, PRESS_TYPE: COMMAND_SINGLE, VALUE: 1},
        },
        (COMMAND_DOUBLE, BUTTON_3): {
            ENDPOINT_ID: 5,
            CLUSTER_ID: 18,
            ARGS: {BUTTON: 5, ATTR_ID: 0x0055, PRESS_TYPE: COMMAND_DOUBLE, VALUE: 2},
        },
        (COMMAND_HOLD, BUTTON_3): {
            ENDPOINT_ID: 5,
            CLUSTER_ID: 18,
            ARGS: {BUTTON: 5, ATTR_ID: 0x0055, PRESS_TYPE: COMMAND_HOLD, VALUE: 0},
        },
        (COMMAND_RELEASE, BUTTON_3): {
            ENDPOINT_ID: 5,
            CLUSTER_ID: 18,
            ARGS: {BUTTON: 5, ATTR_ID: 0x0055, PRESS_TYPE: COMMAND_RELEASE, VALUE: 255},
        }
    }
```

lonevvolf, 2025-06-09:
> [@lonevvolf](https://github.com/lonevvolf) Tested it with 2 button and it associate fine. I've tried 3 times with the single button EU model without success. It end up in Interview complete / configuring.

I had the same, but to be honest, I have this often with other devices as well.  I will experiment a bit more - I also see that the double press is not working.  I'm not sure if this is even possible to get working, and this is my first quirk, so it's a process to understand what's going on.

crypto512, 2025-06-10:
> I had the same, but to be honest, I have this often with other devices as well. I will experiment a bit more - I also see that the double press is not working. I'm not sure if this is even possible [...]

I  was successful with one of them but the second one after more than 10 tries no. Before applying the quirks pairing was successful at the same location immediately. I don’t know how to debug that.

lonevvolf, 2025-06-14:
Ok, I couldn't get the double press working, but I think this is because you would need the Aqara hub, which I don't have.  I tried adding the devices again, and they were added successfully.  Please test - if this works, I will create a new pull request to the official project.

```
"""Aqara H2 single rocker switch quirks. Also see opple_switch.py for similar switches."""

from zigpy.profiles import zha
from zigpy.zcl.clusters.general import (
    AnalogInput,
    Basic,
    Groups,
    Identify,
    MultistateInput,
    OnOff,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks.const import (
    ARGS,
    ATTR_ID,
    BUTTON,
    BUTTON_1,
    BUTTON_2,
    BUTTON_3,
    BUTTON_4,
    CLUSTER_ID,
    COMMAND_DOUBLE,
    COMMAND_HOLD,
    COMMAND_RELEASE,
    COMMAND_SINGLE,
    DEVICE_TYPE,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PRESS_TYPE,
    PROFILE_ID,
    VALUE,
    ZHA_SEND_EVENT,
)
from zhaquirks.xiaomi import (
    AnalogInputCluster,
    BasicCluster,
    MeteringCluster,
    OnOffCluster,
    XiaomiCustomDevice,
)
from zhaquirks.xiaomi.aqara.opple_remote import MultistateInputCluster
from zhaquirks.xiaomi.aqara.opple_switch import OppleSwitchCluster

class PowerMeasurementCluster(AnalogInputCluster):
    """Custom cluster for power measurement."""

    PRESENT_VALUE_ATTRIBUTE = 0x0055

    async def bind(self):
        """Bind cluster."""
        result = await super().bind()
        await self.configure_reporting(
            self.PRESENT_VALUE_ATTRIBUTE,
            0,  # minimum reporting interval
            600,  # maximum reporting interval
            1,  # reportable change
        )
        return result

[... 603 lines cut ...]
    }

    device_automation_triggers = {
        (COMMAND_SINGLE, BUTTON_1): {
            ENDPOINT_ID: 1,
            CLUSTER_ID: 18,
            ARGS: {BUTTON: 1, ATTR_ID: 0x0055, PRESS_TYPE: COMMAND_SINGLE, VALUE: 1},
        },
        (COMMAND_SINGLE, BUTTON_2): {
            ENDPOINT_ID: 4,
            CLUSTER_ID: 18,
            ARGS: {BUTTON: 4, ATTR_ID: 0x0055, PRESS_TYPE: COMMAND_SINGLE, VALUE: 1},
        },
        (COMMAND_SINGLE, BUTTON_3): {
            ENDPOINT_ID: 2,
            CLUSTER_ID: 18,
            ARGS: {BUTTON: 2, ATTR_ID: 0x0055, PRESS_TYPE: COMMAND_SINGLE, VALUE: 1},
        },
        (COMMAND_SINGLE, BUTTON_4): {
            ENDPOINT_ID: 5,
            CLUSTER_ID: 18,
            ARGS: {BUTTON: 4, ATTR_ID: 0x0055, PRESS_TYPE: COMMAND_SINGLE, VALUE: 1},
        },
    }
```

lonevvolf, 2025-06-17:
Please go to the Zigbee info for the device and post the signature.

peschmei, 2025-06-17:
@lonevvolf very much so. I connected N L and L1 but L2 is not connected so there might be even more to it when all wires are connected?

lonevvolf, 2025-06-17:
@peschmei  Please try this version - I added a new handler for Neutral:
```
"""Aqara H2 single rocker switch quirks. Also see opple_switch.py for similar switches."""

from zigpy.profiles import zha
from zigpy.zcl.clusters.general import (
    AnalogInput,
    Basic,
    Groups,
    Identify,
    MultistateInput,
    OnOff,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks.const import (
    ARGS,
    ATTR_ID,
    BUTTON,
    BUTTON_1,
    BUTTON_2,
    BUTTON_3,
    BUTTON_4,
    CLUSTER_ID,
    COMMAND_DOUBLE,
    COMMAND_HOLD,
    COMMAND_RELEASE,
    COMMAND_SINGLE,
    DEVICE_TYPE,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PRESS_TYPE,
    PROFILE_ID,
    VALUE,
    ZHA_SEND_EVENT,
)
from zhaquirks.xiaomi import (
    AnalogInputCluster,
    BasicCluster,
    MeteringCluster,
    OnOffCluster,
    XiaomiCustomDevice,
)
from zhaquirks.xiaomi.aqara.opple_remote import MultistateInputCluster
from zhaquirks.xiaomi.aqara.opple_switch import OppleSwitchCluster

class PowerMeasurementCluster(AnalogInputCluster):
    """Custom cluster for power measurement."""

    PRESENT_VALUE_ATTRIBUTE = 0x0055

    async def bind(self):
        """Bind cluster."""
        result = await super().bind()
        await self.configure_reporting(
            self.PRESENT_VALUE_ATTRIBUTE,
            0,  # minimum reporting interval
            600,  # maximum reporting interval
            1,  # reportable change
        )
        return result

    def _update_attribute(self, attrid, value):
[... 769 lines cut ...]
    }

    device_automation_triggers = {
        (COMMAND_SINGLE, BUTTON_1): {
            ENDPOINT_ID: 1,
            CLUSTER_ID: 18,
            ARGS: {BUTTON: 1, ATTR_ID: 0x0055, PRESS_TYPE: COMMAND_SINGLE, VALUE: 1},
        },
        (COMMAND_SINGLE, BUTTON_2): {
            ENDPOINT_ID: 4,
            CLUSTER_ID: 18,
            ARGS: {BUTTON: 4, ATTR_ID: 0x0055, PRESS_TYPE: COMMAND_SINGLE, VALUE: 1},
        },
        (COMMAND_SINGLE, BUTTON_3): {
            ENDPOINT_ID: 2,
            CLUSTER_ID: 18,
            ARGS: {BUTTON: 2, ATTR_ID: 0x0055, PRESS_TYPE: COMMAND_SINGLE, VALUE: 1},
        },
        (COMMAND_SINGLE, BUTTON_4): {
            ENDPOINT_ID: 5,
            CLUSTER_ID: 18,
            ARGS: {BUTTON: 4, ATTR_ID: 0x0055, PRESS_TYPE: COMMAND_SINGLE, VALUE: 1},
        },
    }
```

peschmei, 2025-06-18:
@lonevvolf Quirk is applied successfully now.
Unfortunately I am still missing the additional buttons. So far only the 2 relays are listed hence I cannot automate anything to the downward pressing of the H2 switches.
When managing the device I can see a variaty of commands for the 2 OnOff Clusters and issuing them switches the relays correctly.
 
I guess either the 4x OppleSwitchClusters or the 2x MultiSateInput Clusters should be responsible for handling the additional states.
(my guess is the MultiStateInput). However there are no available commands for this input so I can only read the values. I will attach the debug information regarding this device in its current state below. Maybe it is insightful.

[AqaraH2Debuginfo_filtered.txt](https://github.com/user-attachments/files/20800466/AqaraH2Debuginfo_filtered.txt)

peschmei, 2025-06-18:
Update: I cannot See additional Buttons in the zigbee Device overview but they are Available when creating automations. They also work correctly as triggers. good enough for me!

crypto512, 2025-06-18:
I've tested the new version that is working better during association. Unfortunately my automation using button_3 is not working anymore. In the device I can see fired some event for single 4 and 5.

lonevvolf, 2025-06-18:
I redid the numbering of the buttons to match the US one. So button 2 is the bottom of the left button and button 4 is the bottom of the right button.

peschmei, 2025-06-18:
After further testing I have to acknowledge that only Button_2_single seems to fire the event correctly.
The upper buttons can be automated using the actual switch entity but the lower right event is not firing at all. As well as the button events for 1 and 3 (those should be the upper ones right?).

However I also see:
`
Switchname 5 Single event was fired
Switchname Lichtschalter 4 Single event was fired
`
in the device log in ZHA. Which does not fit the button_2_single and whatever number 4 and 5 should be..

btw this is an easy automation for checking:
`
alias: TEST ONLY
description: ""
triggers:
  - device_id: 3b8a9264ca3656e6b4ac362cf0ea216a
    domain: zha
    type: single
    subtype: button_1
    trigger: device
    id: button_1 single
  - device_id: 3b8a9264ca3656e6b4ac362cf0ea216a
    domain: zha
    type: single
    subtype: button_2
    trigger: device
    id: button_2 single
  - device_id: 3b8a9264ca3656e6b4ac362cf0ea216a
    domain: zha
    type: single
    subtype: button_3
    trigger: device
    id: button_3_single
  - device_id: 3b8a9264ca3656e6b4ac362cf0ea216a
    domain: zha
    type: single
    subtype: button_4
    trigger: device
    id: button_4_single
conditions: []
actions:
  - action: persistent_notification.create
    metadata: {}
    data:
      message: "Trigger was: {{ trigger.id }}"
      title: Switch was pressed
mode: single
`

lonevvolf, 2025-06-19:
@peschmei That's a good idea.  You can change it to decoupled mode in the attributes - I will have to test to see if they fire correctly once decoupled.

crypto512, 2025-06-20:
@lonevvolf Thanks last linked works with button 4.

lonevvolf, 2025-06-22:
Just FYI, I tried decoupled mode, but no event is triggered for the top buttons, even when decoupled.  I have submitted the quirk to the main repo, let's see how my first attempt at a quirk goes.  :)

kokrp, 2025-06-24:
Thanks for all your efforts! Very much appreciated!
I don't have any experience with this.  Is there a description somewhere explaining how to apply the quirk so I can try to help with testing?

AndreiPustovalov, 2025-08-15:
> Just FYI, I tried decoupled mode, but no event is triggered for the top buttons, even when decoupled. I have submitted the quirk to the main repo, let's see how my first attempt at a quirk goes. :)

Thanks for the quirk! I tested it with 4 buttons switch with neutral.
I changed endpoints 1 and 2 from `MultistateInput` to `MultistateInputCluster` ([here](https://github.com/lonevvolf/zha-device-handlers/blob/dev/zhaquirks/xiaomi/aqara/switch_h2.py#L781) and [here](https://github.com/lonevvolf/zha-device-handlers/blob/dev/zhaquirks/xiaomi/aqara/switch_h2.py#L798)) and top buttons started to work in decoupled mode.

lonevvolf, 2025-08-16:
Thank you - I have added the suggested change to the PR.

I do hope to get back to this at some point soon, but it's also disappointing that it hasn't gotten merged yet.  Not sure what needs to be done to get it moving.

ZekasXD, 2025-08-25:
> Thanks for the quirk! I tested it with 4 buttons switch with neutral. I changed endpoints 1 and 2 from `MultistateInput` to `MultistateInputCluster` [...]

Hi @AndreiPustovalov, can you explain what you did exactly to get it working? I have an Aqara H2 switch with 2 buttons installed and connected to ZHA and using the quirk. I manage to set it to decoupled mode, and now only the bottom button sends single click events, nothing else. The top button doesn't show any events at all and the bottom button doesn't seem to do double or long presses either. But I'd be happy to just have the 2 buttons working at all right now, even with double/long events.

Mazoku01, 2025-09-07:
Any news about feature request?

CarbonCollins, 2025-09-20:
I’m not who you were after but I did just give the new quirk a go on my H2 with 2 channels and 2 buttons and a neutral. 

I ran out of time before I could test all features on it but so far I got:

Both relay switches seem to work correctly and also lock on when turning the lock left and lock right switches. I’m not sure what the Relay and Decoupled modes were intended to do as they did not seem to make any changes to the operation when I tried it.

Power monitoring seems to have worked correctly as I was getting a relatively live power reading (within a few seconds) and the summation delivered looked to increase over time.

I have not tested the lower buttons yet though as I had to leave the house nor have I tested the multi switch toggles.

I have both the 1 and 2 Channel switches in both neutral and no neutral mode so I can also run other tests if needed 🙂 i can get back to you with the lower buttons when I get some time today or tomorrow but though I would at least post here with my initial test 😃

CarbonCollins, 2025-09-20:
Ok i’ve done a bit more testing now, the Relay and Detached modes work as you explained for both buttons. 

I noticed that button numbering was:
1 - Top Left
2 - Top Right
4 - Bottom Left
5 - Bottom Right

While testing the button events I accidentally reset the device so had to re pair it 😅

Flip led and led indicator both worked as expected but only the leds on buttons 1 & 2

Identify worked and flashed all 4 leds

All 4 buttons emitted single events in both relay and detached mode when clicking them.

When enabling multi click i still got all 4 buttons emitting single events but only buttons 4 and 5 emitted hold and release events which took several seconds to come in. After a while i stopped receiving button events but that may have been after I accidentally reset the device…

I did do a device reconfigure which reported a lot of orange crosses in a lot of the columns. I’m assuming these are for things like voltage and current measurements so if it’s usefull i can send that over so you have a full list of them. Just let me know

lonevvolf, 2025-09-20:
@CarbonCollins All the behavior (except the sudden stopping of events) is expected behavior as far as I am aware.  I am currently trying to get the unit tests to pass, but am having a bit of trouble on the last bit.  I've opened a bug with the team, but if someone has experience with this, maybe they can check it:

https://github.com/zigpy/zha-device-handlers/pull/4141

lonevvolf, 2025-09-22:
@ZekasXD Can you please confirm if you have installed it with neutral or without?  Without neutral will not show energy readings.

CarbonCollins, 2025-09-22:
I can add with the power metering on my side I originally had mine wired up without a neutral and later added on. Once that happened though I needed to reset the device before power metering began to work. Ontop of that it took some time before the value began to update as I had low power leds on them so it took a while (like an hour) with the lights on before it started to tick up 🙂 make sure you have had the lights on for a while as it is measuring in kwh :)

ZekasXD, 2025-09-22:
Thanks for thet tip, but it's been hooked up like this for over a month now and the lights have been on multiple full evenings on full brightness. There are 5 LED spots wired to it which are 5 watts each.

CarbonCollins, 2025-09-22:
In that case it should have gone up in that time 🙂

CarbonCollins, 2025-09-22:
Ill give it a go when I get back from work this evening, I may even have some time to set up one of my single channel ones to verify it all works there too.

CarbonCollins, 2025-09-22:
ben doing a bit of messing with the quirk and also looking at the other H2 Dimmer quirk that was recently merged

Looks like we can make use of the DeviceTemperatureCluster. I added to my quirk and it seems to report the device temperature. I was also trying to get the voltage reported without much luck but noticed that https://github.com/zigpy/zha-device-handlers/pull/4202/files made some changes for the agl011 to update attribute names for voltage ect and am wondering if we need to do the same for agl010?

ludwig6666, 2025-09-22:
> I have just updated the quirk code here: https://github.com/lonevvolf/zha-device-handlers/blob/dev/zhaquirks/xiaomi/aqara/switch_h2.py
> 
> I removed the US hardware support for now, since I can't test it. [@peschmei](https://github.com/peschmei) If you could test with the neutral installed variant, it would be much appreciated. I'm not sure how it will work with the extra clusters.

I've lost the events with this last update :-(   lumi.switch.agl004 (Aqara Smart Light Switch H2 (2-in-1 Neutral, 2 Buttons, 1 Channel))

ludwig6666, 2025-10-02:
@lonevvolf would you be able to do it? thanks a lot

lonevvolf, 2025-10-02:
Thanks for the feedback

> My theory btw is that double click and hold actions are never meant for the physical buttons 1 & 2

This is my understanding as well.

> sure! I can test it. I have 1, 2 and 3 channel versions of them. Just let me know what do you need I try. Thanks a lot!

@ludwig6666 Sure, I can try creating a new branch where we can play around with this - I will _try_ to do it this weekend, but it all depends on time.  Do you have them installed with neutral or without?

ludwig6666, 2025-10-03:
> Thanks for the feedback
>
>
> This is my understanding as well.
>
>
> [@ludwig6666](https://github.com/ludwig6666) Sure, I can try creating a new branch where we can play around with this - I will _try_ to [...]

I have them already installed, ready to test from my side 👍 , with neutral, 1, 2 and 3 channels.

ludwig6666, 2025-10-06:
> [@ludwig6666](https://github.com/ludwig6666) Please test the file from: https://[raw.githubusercontent.com/lonevvolf/zha-device-handlers/refs/heads/us-test/zhaquirks/xiaomi/aqara/switch_h2.py](https://raw.githubusercontent.com/lonevvolf/zha-device-handlers/refs/heads/us-test/zhaquirks/xiaomi/aqara/switch_h2.py)
> 
> I have added the agl004 (should be the single rocker version) to be the same as the existing 009. There might be a few features missing, but I hope it should at least find it and work.

I did some chabges and now a lot of thing works    """Aqara H2 (EU) rocker switch quirks. Also see opple_switch.py for similar switches."""

from zigpy import types
from zigpy.profiles import zha
from zigpy.quirks import CustomCluster
from zigpy.quirks.v2 import QuirkBuilder
from zigpy.zcl.clusters.general import AnalogInput, Identify
from zigpy.zcl.foundation import BaseAttributeDefs, DataTypeId, ZCLAttributeDef

from zhaquirks.const import (
    BUTTON_1,
    BUTTON_2,
    BUTTON_3,
    BUTTON_4,
    COMMAND,
    COMMAND_DOUBLE,
    COMMAND_HOLD,
    COMMAND_RELEASE,
    COMMAND_SINGLE,
    ZHA_SEND_EVENT,
)
from zhaquirks.xiaomi import XiaomiAqaraE1Cluster
from zhaquirks.xiaomi.aqara.opple_remote import (
    COMMAND_1_DOUBLE,
    COMMAND_1_HOLD,
    COMMAND_1_RELEASE,
    COMMAND_1_SINGLE,
    COMMAND_2_DOUBLE,
    COMMAND_2_HOLD,
    COMMAND_2_RELEASE,
    COMMAND_2_SINGLE,
    COMMAND_4_DOUBLE,
    COMMAND_4_HOLD,
    COMMAND_4_RELEASE,
    COMMAND_4_SINGLE,
    COMMAND_5_DOUBLE,
    COMMAND_5_HOLD,
    COMMAND_5_RELEASE,
    COMMAND_5_SINGLE,
)
from zhaquirks.xiaomi.aqara.opple_switch import MultistateInputCluster

class AqaraOperationMode(types.enum8):
    """Aqara operation mode attribute values."""

    Decoupled = 0x00
    Relay = 0x01

class AqaraPowerOnMode(types.enum8):
    """Aqara power on mode attribute values."""

    On = 0x00
    Previous = 0x01
    Off = 0x02
    Inverted = 0x03

class PowerMeasurementCluster(CustomCluster, AnalogInput):
    """Custom cluster for power measurement."""

    PRESENT_VALUE_ATTRIBUTE = 0x0055

    async def bind(self):
        """Bind cluster."""
        result = await super().bind()
        await self.configure_reporting(
            self.PRESENT_VALUE_ATTRIBUTE,
[... 237 lines cut ...]
        fallback_name="Lock relay",
    )
    .switch(
        AqaraManuSpecificCluster.AttributeDefs.multi_click.name,
        AqaraManuSpecificCluster.cluster_id,
        off_value=1,
        on_value=2,
        endpoint_id=4,
        translation_key="multi_click",
        fallback_name="Multi click",
    )
    .device_automation_triggers(
        {
            (COMMAND_HOLD, BUTTON_1): {COMMAND: COMMAND_1_HOLD},
            (COMMAND_SINGLE, BUTTON_1): {COMMAND: COMMAND_1_SINGLE},
            (COMMAND_DOUBLE, BUTTON_1): {COMMAND: COMMAND_1_DOUBLE},
            (COMMAND_RELEASE, BUTTON_1): {COMMAND: COMMAND_1_RELEASE},
            (COMMAND_HOLD, BUTTON_2): {COMMAND: COMMAND_4_HOLD},
            (COMMAND_SINGLE, BUTTON_2): {COMMAND: COMMAND_4_SINGLE},
            (COMMAND_DOUBLE, BUTTON_2): {COMMAND: COMMAND_4_DOUBLE},
            (COMMAND_RELEASE, BUTTON_2): {COMMAND: COMMAND_4_RELEASE},
        }
    )
    .add_to_registry()
)

EvgeniGanchev, 2025-11-10:
When should this be released? Or the above will be a workaround for the time being

nickpanaiotov, 2025-12-12:
Just dropping to say that the lonevvolf fork works perfectly for me. I've had some fun time figuring out how to add it as a custom quirk, but this is not a problem with the code. Mine is the four buttons eu version, I have no neutral wire so the power consumption is not tested.
The level of customisation is amazing you can lock the relay switch while is been turned on and after that to use the same button to toggle smart lights without interfering with the circuit.

ieberl1, 2026-01-02:
FWIW
I have the [H2 US version alg007](https://www.aqara.com/us/product/dimmer-switch-h2-us/) (3 switch dimmer) and can confirm the version located [here](https://github.com/orgs/home-assistant/discussions/1720) works. I tested On/Off/Dimming/Automatons/Scene. I don't have a neutral so I can't test power monitoring.

Not sure if you want to add that version or not.

Twanfox, 2026-01-12:
I recently acquired a [Aqara H2 2-button 1-channel switch model WS-K02E](https://www.aqara.com/en/product/light-switch-h2-us/)  and identified as lumi.switch.agl004 and installed it with neutral. 

I sadly don't know much about quirk development, so it took me a while to figure out what all I needed to do. I configured my HA instance to accept custom quirks, and installed the definition from [here](https://github.com/zigpy/zha-device-handlers/issues/3957#issuecomment-3464269505) (naming it as a .py file in my quirks folder) only to have it not identify correctly. I tried others, but in flopping around back and forth I didn't understand well what I was doing nor why I didn't see the second switch. It took me adding the Manufacturer name to the previously linked quirk file (code block below) before that one (supposedly the 'best' and right one?) would identify properly and show all the additional features. Still no switch for the bottom button though.

`Line 261    .applies_to("Aqara", "lumi.switch.agl004")`. 

It finally dawned on me I was getting confused on how the switch for the wireless (button 2) was to be read. Since there's no backing relay, it technically isn't a switch, but a button that triggers a zha_event (which the device now does, where before it was just a 'turn_on' event from the top relay switch and nothing from the bottom button). Can confirm that is now usable in automations and I can replace an Aqara mini switch doing that same purpose. 

It doesn't illuminate the LED, but I suppose it wouldn't have a relay state to look at so if I want that, then I probably need to program the automation to turn it on. I alas do not know if this is possible or how to do it, but probably. 

Since I installed the switch with neutral, it seems to be recording power data fine, though there is that same noted delay in responsiveness in getting it updated in HA. Not critical but I did happen to notice the delay.