zigpy/zha-device-handlers #3962: [Device Support Request] Avatto ZWSMD-4 (4 channel relay controller)

charliewallis, 2025-03-14:
### Problem description

Current firmware version 0x0000004e

### Solution description

ZHA inclusion

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xed00",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE284_f5efvtbv",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

charliewallis, 2025-03-14:
Working custom Quirk for the Avatto ZWS

[Avatto ZWSMD-4 Quirk.txt](https://github.com/user-attachments/files/19251533/Avatto.ZWSMD-4.Quirk.txt)

MD-4 attached.

robertio, 2025-05-20:
Hi,
first of all many Thanks to Charliewallis for this solution!!

[image]

How to apply this: 
1. Create the custom_zha_quirks Directory
ssh to the home assistant device. search your HA config directory (in my case it was ) : 
`mkdir -p  /var/lib/homeassistant/homeassistant/custom_zha_quirks`

2. Save the Python Quirk File
`nano /var/lib/homeassistant/homeassistant/custom_zha_quirks/ts0601_tze284_quad_switch.py`
 
3. Create or Update __init__.py
`nano /var/lib/homeassistant/homeassistant/custom_zha_quirks/__init__.py`
Add this:
"""Custom quirks for ZHA integration."""
import zhaquirks
zhaquirks.setup()

4. Update configuration.yaml
`zha:
  custom_quirks_path: /config/custom_zha_quirks`

5. Restart Home Assistant from the UI or run 
`docker restart homeassistant`

robertio, 2025-05-25:
The Device has a firmware setting to switch auto off (1 sec delay after you switch device), just hold power button AND Switch 1 button at the same time for 5 seconds, and this set the relays to "momentary" mode. this is used for garage doors and some other devices needing only 1 sec impulse.
Normal mode when you set Switch on and off from Homeassistant can be set same way. push power button and 1st switch button for 5 seconds. Blue led will blinking and then stay blue.