zigpy/zha-device-handlers #3963: [Device Support Request] Schneider Electric Wiser 230v smoker alarm not recognized as router

siepkes, 2025-03-14:
### Problem description

The Schneider Electric Wiser 230v smoker alarm (W599501 / 550B1028) is not recognized as router but as an end device. This is probably due to the fact the device also has a battery, of which it also reports the status, aside from being 230v powered.

Other then the above it works pretty great with Home assistant. It's actually a pretty great some detector because it checks all these boxes:

* Supports Zigbee.
** Works without Internet connection with Homeassistant
* Has a backup battery.
* Has backup RF communication with other smokedetectors.
* Detects both smoke and heat.
* Is a smoke detector from a reputable brand.

This issue was reported before as https://github.com/zigpy/zha-device-handlers/issues/3170.

### Solution description

Make the 230v version of the wiser smoke detector a Zigbee router.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

tannyl, 2025-06-15:
I sadly don't think this is something any of the Zigbee gateways are able to change.

I have captured the Zigbee traffic between SEs own Wiser hub and one of these smoke alarms. At no point does the smoke alarm send any information about it having 230V connected or not, what the supply voltage is and so on.

It seems like the Wiser smoke detector itself does not actually "know" that it is powered by 230V nor if 230V is actually connected. It looks like they just took the battery version and added a 230V power-supply circuit in parallel to the battery circuit (I am just simplifying here).

So unless SE releases new firmware which makes it "230V aware" (or someone makes custom firmware) **AND** SE actually already have circuitry build into them that feeds information to the micro controllers about the 230V, I don't see this changing, which is a same, since I think it is actually pretty important to know if the 230V to your smoke alarm is actually working.

puddly, 2025-06-15:
Indeed, the node descriptor for the device has `is_receiver_on_when_idle=False` and `logical_type=<LogicalType.EndDevice: 2>`. There's no way to change the way the device itself functions on the network, unfortunately.

siepkes, 2025-06-16:
Thanks for looking in to it folks! 

> since I think it is actually pretty important to know if the 230V to your smoke alarm is actually working.

Definitely. 

I do think at some level the device is "aware" because if it's powered by 230V the status LED is continuously lid green. The battery ones don't do that. I should probably test that by cutting the power. 

Oddly enough I don't think the [manual](https://www.productinfo.schneider-electric.com/wiser_eu/wiser-smoke-alarm-230-v_device-user-guide_wiser_se/English/Wiser%20Smoke%20Alarm%20230%20V_WSE_Device%20user%20guide.pdf ) mentions this LED state. I see no mention of what it means if the LED is continuously lid green.