zigpy/zha-device-handlers #3966: Another ts0601 soil sensor device

medovarsky, 2025-03-18:
Hello,

I have got new quirky QT-07S TS0601 type soil sensor device by _TZE284_myd45weu, which is not showing its values.
Could it be added to the list? Surely, I'd be glad to test.

It seems the same or similar to existing _TZE20x devices to my untrained eye:
https://github.com/zigpy/zha-device-handlers/blob/5c4e08d94091a4c5f37cea7b21fd918474e98f82/zhaquirks/tuya/ts0601_sensor.py#L377

[zha-01JMQFNM0W3E8YWW3APXK5Z0BM-_TZE284_myd45weu TS0601-70d8b760bc6dc02448c97a0713ee8bf7(3).json](https://github.com/user-attachments/files/19302625/zha-01JMQFNM0W3E8YWW3APXK5Z0BM-_TZE284_myd45weu.TS0601-70d8b760bc6dc02448c97a0713ee8bf7.3.json)

Thank you in forward,

Lubos Medovarsky

embedded-solutions-art, 2025-04-12:
I have bought the same sensor two weeks ago. It doesn't work with the mentioned script.

nullaegy, 2025-04-12:
There is a solution to our problem: 
https://community.home-assistant.io/t/troubleshooting-tze284-temp-humid-sensor-custom-quirk-in-zha/828046/8

To sum it up:
I've found a these info in the forum and some fiddling with the config helped me. 
I've glued some parts together and it's finally works. 
quick fix with home-assistant in docker:
create a file:
`/home-assistant/config/custom_zha_quirks/soil-sensor_TZE284_myd45weu.py`
with this content:
```
"""Tuya temp and humidity sensors."""

import copy

from zigpy.quirks.v2 import EntityPlatform, EntityType
from zigpy.quirks.v2.homeassistant import PERCENTAGE, UnitOfTemperature, UnitOfTime
from zigpy.quirks.v2.homeassistant.sensor import SensorDeviceClass
import zigpy.types as t
from zigpy.zcl import foundation

from zhaquirks.tuya import (
    TUYA_SET_TIME,
    TuyaPowerConfigurationCluster2AAA,
    TuyaTimePayload,
)
from zhaquirks.tuya.builder import TuyaQuirkBuilder, TuyaTemperatureMeasurement
from zhaquirks.tuya.mcu import TuyaMCUCluster

class TuyaTempUnitConvert(t.enum8):
    """Tuya temperature unit convert enum."""

    Celsius = 0x00
    Fahrenheit = 0x01

(
    TuyaQuirkBuilder("_TZE284_myd45weu", "TS0601")
    .tuya_temperature(dp_id=5, scale=100)
    .tuya_battery(dp_id=15)
    .tuya_soil_moisture(dp_id=3)  # Adicionando escala para umidade do solo
    .skip_configuration()
    .add_to_registry()
)
```

next to your docker-compose.yml there is a config folder somewhere and you can find a `configuration.yaml` and you can add this part to that config:
```
zha:
  custom_quirks_path: /config/custom_zha_quirks/
```

After this you can `docker-compose up -d` and your sensor will work, at least mine did (try to remove and re add it if you need) :] good luck and happy measuring :]

embedded-solutions-art, 2025-04-12:
@nullaegy your solutions work! I have only added one line to the quirk that you proposed in the constructor body:
` .applies_to("_TZE284_myd45weu", "TS0601")`

Quirk implementation should be stored in _/root/config/custom_zha_quirks/_TZE284_myd45weu.py_.
Zigbee Home Assistance quirks path should be added without quotes in _/root/config/configuration.yaml_:
```
zha:
  custom_quirks_path: /config/custom_zha_quirks

```

I have tested it on raspberry pi 3, HA version 15.1 core 2025.4.1, 
[image]

Thanks!

medovarsky, 2025-04-13:
Hello @nullaegy, @embedded-solutions-art,

Thank you a lot! Confirming that your solution works with my HA Green.

derekcentrico, 2025-05-02:
There seems to be a pending commit that officially fixes this:  https://github.com/zigpy/zha-device-handlers/pull/4038

closed 2025-05-03 by TheJulianJES