zigpy/zha-device-handlers #3969: [Device Support Request] ZigBee 24G MmWave Radar Human Body Presence Motion Sensor

noobipaul, 2025-03-20:
### Problem description

ZigBee 24G MmWave Radar Human Body Presence Motion Sensor - not showing entities.

### Solution description

support added

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[[aliexpress](https://de.aliexpress.com/item/1005008034398184.html?spm=a2g0o.order_list.order_list_main.5.4a9e5c5f6YIQrG&gatewayAdapt=glo2deu)]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

[{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xed00",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE284_iadro9bf",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}]

```

[... 629 lines cut ...]
</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

gkk127, 2025-03-21:
Looking for same device support on identical presence sensor in home assistant, TS0601 by _TZE284_iadro9bf. Shall keep an eye on the response here.

noobipaul, 2025-03-24:
This is far from perfect, I sure there many of the configurations that don't work, Illuminance and Target distance also doesn't work, I am no coder but I've managed to hack this around to get the presence detection working for this sensor, and that's all I currently wanted.

`"""BlitzWolf IS-3/Tuya motion rechargeable occupancy sensor."""

import asyncio
from typing import Any

from zigpy.quirks.v2 import EntityPlatform, EntityType
from zigpy.quirks.v2.homeassistant import LIGHT_LUX, UnitOfLength, UnitOfTime
from zigpy.quirks.v2.homeassistant.binary_sensor import BinarySensorDeviceClass
from zigpy.quirks.v2.homeassistant.sensor import SensorDeviceClass, SensorStateClass
import zigpy.types as t
from zigpy.zcl.clusters.measurement import OccupancySensing
from zigpy.zcl.clusters.security import IasZone

from zhaquirks.tuya import TuyaLocalCluster
from zhaquirks.tuya.builder import TuyaQuirkBuilder

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class TuyaMotionWithReset(IasZone, TuyaLocalCluster):
    """Tuya local IAS motion cluster with reset."""

    _CONSTANT_ATTRIBUTES = {
        IasZone.AttributeDefs.zone_type.id: IasZone.ZoneType.Motion_Sensor
    }
    reset_s: int = 15

    def __init__(self, *args, **kwargs):
        """Init."""
        super().__init__(*args, **kwargs)
        self._loop = asyncio.get_running_loop()
        self._timer_handle = None

    def _turn_off(self) -> None:
        """Reset IAS zone status."""
        self._timer_handle = None
        self.debug("%s - Resetting Tuya motion sensor", self.endpoint.device.ieee)
        self._update_attribute(IasZone.AttributeDefs.zone_status.id, 0)

    def _update_attribute(self, attrid: int | t.uint16_t, value: Any) -> None:
        """Catch zone status updates and potentially schedule reset."""
        if (
            attrid == IasZone.AttributeDefs.zone_status.id
            and value == IasZone.ZoneStatus.Alarm_1
        ):
            self.debug("%s - Received Tuya motion event", self.endpoint.device.ieee)
            if self._timer_handle:
                self._timer_handle.cancel()
            self._timer_handle = self._loop.call_later(self.reset_s, self._turn_off)

        super()._update_attribute(attrid, value)

class TuyaSelfCheckResult(t.enum8):
    """Tuya self check result enum."""

    Checking = 0x00
    CheckSuccess = 0x01
    CheckFailure = 0x02
    Others = 0x03
    CommFault = 0x04
    RadarFault = 0x05

class TuyaPresenceState(t.enum8):
    """Tuya presence state enum."""

    Unoccupied = 0x00
    Presence = 0x01
    Peaceful = 0x02
[... 250 lines cut ...]
    )
    .tuya_number(
        dp_id=112,
        attribute_name="block_time",
        type=t.uint16_t,
        device_class=SensorDeviceClass.DURATION,
        unit=UnitOfTime.SECONDS,
        min_value=0,
        max_value=10,
        step=0.1,
        multiplier=0.1,
        translation_key="block_time",
        fallback_name="Block time",
    )
    # 113 parameter_setting_result, z2m doesn't expose
    # 114 factory_parameters, z2m doesn't expose
    .tuya_enum(
        dp_id=115,
        attribute_name="sensor_mode",
        enum_class=TuyaMotionSensorMode,
        translation_key="sensor_mode",
        fallback_name="Sensor mode",
    )
    .add_to_registry()
)`

gkk127, 2025-03-24:
Fantastic, at least now we can get some function out of the device. Hoping quirk will come in time, but this is great. Thanks.

LundinPlace, 2025-05-05:
Working like a charm! Thanks 👍

josiasdx, 2025-05-16:
Only presence detection works... but ... "works" (good start!) Thank you

vothar, 2025-05-22:
seem that there is a full working version for Z2M
https://github.com/Koenkk/zigbee-herdsman-converters/issues/8939

dd32, 2025-06-20:
I've acquired a Tuya Gateway, and unfortunately it appears that target distance isn't exposed by the device there, so it seems to be more or less a super basic PIR-like device, just using mmWave.

| Tuya App Details | Tuya App Settings |
| --- | --- |
| [image] | [image] |

Setting a timeout less than 10 seconds also seemed to cause it to treat it as a 10s delay. 

The dp_ids from the Tuya debug console were:
```
{
"1":"sensor state",
"9":"Trigger Sensitivity",
"12":"Illuminance value",
"101":"exist_Sensitivity",
"102":"hold_time",
}
```

PR incoming.

dd32, 2025-06-20:
Here's the custom quirk for testing without the PR..
```
"""ZigBee 24G MmWave Radar Human Body Presence Motion Sensor _TZE284_iadro9bf - https://github.com/zigpy/zha-device-handlers/issues/3969"""

import asyncio

from zigpy.quirks.v2.homeassistant import UnitOfTime
from zigpy.quirks.v2.homeassistant.sensor import SensorDeviceClass
import zigpy.types as t
from zigpy.zcl.clusters.measurement import OccupancySensing

from zhaquirks.tuya import TuyaLocalCluster
from zhaquirks.tuya.builder import TuyaQuirkBuilder

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class TuyaMotionPresenceSensitivity(t.enum8):
    """Tuya motion presence sensitivity enum."""

    Low = 0x00
    Medium = 0x01
    High = 0x02

class TuyaSensitivityMode(t.enum8):
    """Tuya sensitivity mode enum."""

    Low = 0x00
    Medium = 0x01
    High = 0x02

(
    TuyaQuirkBuilder("_TZE284_iadro9bf", "TS0601")
    # "1":"sensor state" - Working
    .tuya_dp(
        dp_id=1,
        ep_attribute=TuyaOccupancySensing.ep_attribute,
        attribute_name=OccupancySensing.AttributeDefs.occupancy.name,
        converter=lambda x: x == 0,
    )
    .adds(TuyaOccupancySensing)

    # "9":"Trigger Sensitivity" - Low/Medium/High
    .tuya_enum(
        dp_id=9,
        attribute_name="trigger_sensitivity",
        enum_class=TuyaSensitivityMode,
        translation_key="trigger_sensitivity",
        fallback_name="Trigger sensitivity",
    )

    # "12":"Illuminance value" - 3lux
    .tuya_illuminance(dp_id=12)

    # "101":"exist_Sensitivity" - Low/Medium/High
    .tuya_enum(
        dp_id=101,
        attribute_name="exist_Sensitivity",
        enum_class=TuyaMotionPresenceSensitivity,
        translation_key="presence_sensitivity",
        fallback_name="Presence sensitivity",
    )

    ## "102":"hold_time" Default 30s (10-600s)
    .tuya_number(
        dp_id=102,
        attribute_name="hold_time",
        type=t.uint16_t,
        device_class=SensorDeviceClass.DURATION,
        unit=UnitOfTime.SECONDS,
        min_value=10,
        max_value=600,
        step=1,
        translation_key="fading_time",
        fallback_name="Fading time",
    )
    .skip_configuration()
    .add_to_registry()
)
```

thojust, 2025-07-06:
I used the seoncd quirk that @dd32 dd32 made, the illuminace value works perfectly, but for some reason the presence is just always clear regardless of actual human presence

generaltorjan, 2025-07-06:
@thojust try to set the fading time to a value 10 and set the presence sensitivity "high or medium or low" and the trigger sensitivity as well , when i did it start working for me , good luck

fingertrouble, 2026-05-24:
For those having issues if you are using the square presence sensor with a slightly different name:

"TS0601 by _TZE204_iadro9bf"  (NOT  _TZE204_iadro9bf)

Use this quirk from @dd32 https://github.com/zigpy/zha-device-handlers/issues/3969#issuecomment-2989957516 and rename TZE284_iadro9bf to _TZE204_iadro9bf on line 31.

It then suddenly comes to life!