zigpy/zha-device-handlers #3971: [Device Support Request] Tuya SDM01V1.5 / Zemismart 3P+N energy monitor

validide, 2025-03-21:
### Problem description

Hi,

I have a [Zemismart 3P+N energy monitor (Tuya SDM01V1.5)], Zigbee model TS0601 and manufacutrer _TZE204_gomuk3dc  (https://www.zigbee2mqtt.io/devices/SDM01V1.5.html) and this is not working with the fix from  #3571. 

### Solution description

I have implemented the fix from [Z2M](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts#L11196) and all of the sensors seem to be working.

The ones that are not working are:
* apparent_power
* power_factor
* update_frequency - I have not yet fully tested this

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4098,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0b04",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE204_gomuk3dc",
  "model": "TS0601"
}

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>
The file to large to paste here so I have attached it.
[diag.json](https://github.com/user-attachments/files/19394821/diag.json)

</details>

### Logs

<details><summary>Logs</summary>

```python

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

class Tuya3PhaseElectricalMeasurementV1_5(ElectricalMeasurement, TuyaLocalCluster):
    """Tuya Electrical Measurement cluster."""

    _CONSTANT_ATTRIBUTES = {
        ElectricalMeasurement.AttributeDefs.ac_current_divisor.id: 1000,
        ElectricalMeasurement.AttributeDefs.ac_voltage_divisor.id: 10,
        ElectricalMeasurement.AttributeDefs.ac_frequency_divisor.id: 100,
    }

(
    TuyaQuirkBuilder("_TZE204_gomuk3dc", "TS0601")
    .applies_to("_TZE284_gomuk3dc", "TS0601")
    .applies_to("_TZE200_gomuk3dc", "TS0601")
    .tuya_sensor(
        dp_id=1,
        attribute_name="energy_consumed",
        type=t.uint32_t,
        divisor=100,
        state_class=SensorStateClass.TOTAL_INCREASING,
        device_class=SensorDeviceClass.ENERGY,
        unit=UnitOfEnergy.KILO_WATT_HOUR,
        fallback_name="Total energy",
    )
    .tuya_sensor(
        dp_id=23,
        attribute_name="energy_produced",
        type=t.uint32_t,
        divisor=100,
        state_class=SensorStateClass.TOTAL_INCREASING,
        device_class=SensorDeviceClass.ENERGY,
        unit=UnitOfEnergy.KILO_WATT_HOUR,
        translation_key="energy_produced",
        fallback_name="Energy produced",
    )
    .tuya_dp(
        dp_id=29,
        ep_attribute=Tuya3PhaseElectricalMeasurementV1_5.ep_attribute,
        attribute_name="total_active_power",
    )
    .tuya_dp(
        dp_id=32,
        ep_attribute=Tuya3PhaseElectricalMeasurementV1_5.ep_attribute,
        attribute_name="ac_frequency",
    )
    .tuya_sensor(
        dp_id=50,
        attribute_name="power_factor",
        type=t.uint8_t,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.POWER_FACTOR,
        unit=PERCENTAGE,
        translation_key="total_power_factor",
        fallback_name="Total power factor",
    )
    # Update the frequency at which the device reports its data
    # did not manage to test this fully
    # .tuya_number(
    #     dp_id=102,
    #     attribute_name="update_frequency",
    #     type=t.uint16_t,
    #     device_class=SensorDeviceClass.DURATION,
    #     unit=UnitOfTime.SECONDS,
    #     min_value=5,
    #     max_value=3600,
    #     step=1,
    #     translation_key="update_frequency",
    #     fallback_name="Update frequency",
    #     access=foundation.ZCLAttributeAccess.Write,
    # )
    .tuya_dp(
        dp_id=103,
        ep_attribute=Tuya3PhaseElectricalMeasurementV1_5.ep_attribute,
        attribute_name="rms_voltage_ph_a",
    )
    .tuya_dp(
        dp_id=104,
        ep_attribute=Tuya3PhaseElectricalMeasurementV1_5.ep_attribute,
        attribute_name="rms_current_ph_a",
    )
    .tuya_dp(
        dp_id=105,
        ep_attribute=Tuya3PhaseElectricalMeasurementV1_5.ep_attribute,
        attribute_name="active_power_ph_a",
    )
    .tuya_dp(
        dp_id=108,
        ep_attribute=Tuya3PhaseElectricalMeasurementV1_5.ep_attribute,
        attribute_name="power_factor_ph_a",
    )
    .tuya_sensor(
        dp_id=109,
        attribute_name="energy_consumed_ph_a",
        type=t.uint32_t,
        divisor=100,
        state_class=SensorStateClass.TOTAL_INCREASING,
        device_class=SensorDeviceClass.ENERGY,
        unit=UnitOfEnergy.KILO_WATT_HOUR,
        translation_key="energy_ph_a",
        fallback_name="Energy phase A",
    )
    .tuya_sensor(
        dp_id=110,
        attribute_name="energy_produced_ph_a",
        type=t.uint32_t,
        divisor=100,
        state_class=SensorStateClass.TOTAL_INCREASING,
        device_class=SensorDeviceClass.ENERGY,
        unit=UnitOfEnergy.KILO_WATT_HOUR,
        translation_key="energy_produced_ph_a",
        fallback_name="Energy produced phase A",
    )
    .tuya_dp(
        dp_id=112,
        ep_attribute=Tuya3PhaseElectricalMeasurementV1_5.ep_attribute,
        attribute_name="rms_voltage_ph_b",
    )
    .tuya_dp(
        dp_id=113,
        ep_attribute=Tuya3PhaseElectricalMeasurementV1_5.ep_attribute,
        attribute_name="rms_current_ph_b",
    )
    .tuya_dp(
        dp_id=114,
        ep_attribute=Tuya3PhaseElectricalMeasurementV1_5.ep_attribute,
        attribute_name="active_power_ph_b",
    )
    .tuya_dp(
        dp_id=117,
        ep_attribute=Tuya3PhaseElectricalMeasurementV1_5.ep_attribute,
        attribute_name="power_factor_ph_b",
    )
    .tuya_sensor(
        dp_id=118,
        attribute_name="energy_consumed_ph_b",
        type=t.uint32_t,
        divisor=100,
        state_class=SensorStateClass.TOTAL_INCREASING,
        device_class=SensorDeviceClass.ENERGY,
        unit=UnitOfEnergy.KILO_WATT_HOUR,
        translation_key="energy_ph_b",
        fallback_name="Energy phase B",
    )
    .tuya_sensor(
        dp_id=119,
        attribute_name="energy_produced_ph_b",
        type=t.uint32_t,
        divisor=100,
        state_class=SensorStateClass.TOTAL_INCREASING,
        device_class=SensorDeviceClass.ENERGY,
        unit=UnitOfEnergy.KILO_WATT_HOUR,
        translation_key="energy_produced_ph_b",
        fallback_name="Energy produced phase B",
    )
    .tuya_dp(
        dp_id=121,
        ep_attribute=Tuya3PhaseElectricalMeasurementV1_5.ep_attribute,
        attribute_name="rms_voltage_ph_c",
    )
    .tuya_dp(
        dp_id=122,
        ep_attribute=Tuya3PhaseElectricalMeasurementV1_5.ep_attribute,
        attribute_name="rms_current_ph_c",
    )
    .tuya_dp(
        dp_id=123,
        ep_attribute=Tuya3PhaseElectricalMeasurementV1_5.ep_attribute,
        attribute_name="active_power_ph_c",
    )
    .tuya_dp(
        dp_id=126,
        ep_attribute=Tuya3PhaseElectricalMeasurementV1_5.ep_attribute,
        attribute_name="power_factor_ph_c",
    )
    .tuya_sensor(
        dp_id=127,
        attribute_name="energy_consumed_ph_c",
        divisor=100,
        type=t.uint32_t,
        state_class=SensorStateClass.TOTAL_INCREASING,
        device_class=SensorDeviceClass.ENERGY,
        unit=UnitOfEnergy.KILO_WATT_HOUR,
        translation_key="energy_ph_c",
        fallback_name="Energy phase C",
    )
    .tuya_sensor(
        dp_id=128,
        attribute_name="energy_produced_ph_c",
        type=t.uint32_t,
        divisor=100,
        state_class=SensorStateClass.TOTAL_INCREASING,
        device_class=SensorDeviceClass.ENERGY,
        unit=UnitOfEnergy.KILO_WATT_HOUR,
        translation_key="energy_produced_ph_c",
        fallback_name="Energy produced phase C",
    )
    .adds(Tuya3PhaseElectricalMeasurementV1_5)
    # .skip_configuration() # Not sure why we would need to skip configuration
    .add_to_registry()
)

```

</details>

### Additional information

I have implemented the quirk on a fork of the repo, would it be ok to create a pull request and discuss about any necessary code changes there?

IG-BB, 2025-03-28:
@validide
I have exactly the same model, but I can't get it to run. The quirk doesn't work for me at all. even with the built-in signature. It didn't match my model is like in the quirk _TZE204_gomuk3dc.

I'd be glad to have the same values as in your screenshot.

validide, 2025-03-28:
Hi,

The quirk file I am using is the one from here https://github.com/validide/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_din_power.py#L436 .

Can you share some more device info, the device signature?

validide, 2025-03-29:
Can you elaborate on the "implausible"?

crazyracer98, 2025-05-04:
Hi @validide,
I have the `_TZE284_gomuk3dc` and for me, the following sensors do not seem to work:
- rms_voltage_ph_a
- rms_current_ph_a
- active_power_ph_a
- power_factor_ph_a
- total_active_power

In HA / ZHA it just shows "Unknown" as the value.

validide, 2025-05-05:
Unfortunatly I have the device installed in a remote location and I do not have direct access la to it right now. I will try to look into it as soon as possible.

It might be that the quirck is not correct and the fact it is working for me might be the result of some missconfiguration (I played with multiple quirks before finding this one).

validide, 2025-05-06:
@crazyracer98, I think I fixed all of the sensors appart for `total_active_power`.

You can find the code [here](https://github.com/validide/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_din_power.py#L436) (I am using this exact file).

validide, 2025-05-07:
I managed to fix the aparent power also. All the entities work for me now. @crazyracer98 can you also try?

crazyracer98, 2025-05-08:
Hi @validide 
I can confirm that phase 1 works now :D

For me, `dp_id=29` seems to be the total power.
Additionally, `attribute_name="update_frequency"` snaps to a lower limit of 68.

I adjusted the quirk a bit:
<details>
<summary>Quirk:</summary>

```python
"""
    Tuya Din Power Meter.
    From: https://github.com/zigpy/zha-device-handlers/issues/3971
"""

from zigpy.profiles import zha
from zigpy.quirks.v2 import SensorDeviceClass, SensorStateClass
from zigpy.quirks.v2.homeassistant import PERCENTAGE, UnitOfEnergy, UnitOfPower, UnitOfTime
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import Bus, LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaLocalCluster,
    TuyaManufClusterAttributes,
    TuyaOnOff,
    TuyaSwitch,
)
from zhaquirks.tuya.builder import TuyaQuirkBuilder

TUYA_TOTAL_ENERGY_ATTR = 0x0211
TUYA_CURRENT_ATTR = 0x0212
TUYA_POWER_ATTR = 0x0213
TUYA_VOLTAGE_ATTR = 0x0214
TUYA_DIN_SWITCH_ATTR = 0x0101

SWITCH_EVENT = "switch_event"

"""Hiking Power Meter Attributes"""
HIKING_DIN_SWITCH_ATTR = 0x0110
HIKING_TOTAL_ENERGY_DELIVERED_ATTR = 0x0201
HIKING_TOTAL_ENERGY_RECEIVED_ATTR = 0x0266
HIKING_VOLTAGE_CURRENT_ATTR = 0x0006
HIKING_POWER_ATTR = 0x0267
HIKING_FREQUENCY_ATTR = 0x0269
HIKING_POWER_FACTOR_ATTR = 0x026F
HIKING_TOTAL_REACTIVE_ATTR = 0x026D
HIKING_REACTIVE_POWER_ATTR = 0x026E

class Tuya3PhaseElectricalMeasurementV1_5(ElectricalMeasurement, TuyaLocalCluster):
    """Tuya Electrical Measurement cluster."""

    _CONSTANT_ATTRIBUTES = {
        ElectricalMeasurement.AttributeDefs.ac_current_divisor.id: 1000,
        ElectricalMeasurement.AttributeDefs.ac_voltage_divisor.id: 10,
        ElectricalMeasurement.AttributeDefs.ac_frequency_divisor.id: 100,
    }

[... 173 lines cut ...]
        divisor=100,
        type=t.uint32_t,
        state_class=SensorStateClass.TOTAL_INCREASING,
        device_class=SensorDeviceClass.ENERGY,
        unit=UnitOfEnergy.KILO_WATT_HOUR,
        translation_key="energy_ph_c",
        fallback_name="Energy phase C",
    )
    .tuya_sensor(
        dp_id=128,
        attribute_name="energy_produced_ph_c",
        type=t.uint32_t,
        divisor=100,
        state_class=SensorStateClass.TOTAL_INCREASING,
        device_class=SensorDeviceClass.ENERGY,
        unit=UnitOfEnergy.KILO_WATT_HOUR,
        translation_key="energy_produced_ph_c",
        fallback_name="Energy produced phase C",
    )
    .adds(Tuya3PhaseElectricalMeasurementV1_5)
    # .skip_configuration() # Not sure why we would need to skip configuration
    .add_to_registry()
)
```
</details>

validide, 2025-05-08:
You might be right about the `dp_id=29`and the numbers add-up.

I do not know about the update frequency (maybe a different version), I will leave that commented out for now.

532910, 2025-08-31:
dup #2757

lattice0, 2025-09-14:
Any news on the reporting interval for this device? The 1P + N version (_TZE200_qhlxve78) has support for setting the interval and it seems to work, but the 3P + N (_TZE200_v9hkz2yn) does not and I have 3 of them. They are flooding my network. They also show no bindings that I can set the interval.

chrfin, 2025-09-24:
I just restartet my HA and now I'm "back" to only miss Phase A for active power/current:

[image]

@crazyracer98 This seems to be the sensors you mentioned in your first post. I used the quirk from your post above. Is that the one you use and works, or did you use the last code from @validide ?

chrfin, 2025-09-28:
...and I'm back to admit that I'm stuipd 🤓:
[image]
Because it lists "Energy" as "Energie" and "Energie Phase A", "Energie Phase B" and "Energie Phase C" I expected the actual power to be "Leistung" for all combined and "Leistung Phase A", "Leistung Phase B" and "Leistung Phase C" so I was missing the "Leistung Phase A", BUT the combined power is "Gesamtleistung" (=Total power in German) and "Leistung" is actually the actual power of Phase A 😵...

Maybe there is a way to fix this in the quirk so it shows up as expected?