zigpy/zha-device-handlers #3972: [Device Support Request]  Zigbee WZ-SPI RGBW LED Strip SPI Controller

THK-artjom, 2025-03-21:
### Problem description

Device can be found here:
https://www.iskydance.com/index.php?c=product_show&a=index&id=2433

[image]

### Solution description

Currently the device is not support. It would be great if someone could help me integrating it.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_8fffc3kb",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

nikakoy131, 2025-04-25:
@THK-artjom Thank you! Great work! Works fine for me
Only one issue - typo in code of custom quirk, no closing ")"

**BTW**
Is this possible to implement some "scenes" in this ZHA implemetation? Like in https://github.com/https://github.com/Koenkk/zigbee2mqtt/issues/23991 ? 

**Fixed version of custom quirk** with bigger steps (10) in brightnes and color (in this case, somewhy, ha show them as sliders)

```python

import zigpy.types as t

from zhaquirks.tuya.builder import TuyaQuirkBuilder

"""Work Mode"""
class WorkMode(t.enum8):
    colour = 0x00
    white = 0x01
    scene = 0x02

(
    TuyaQuirkBuilder("_TZE204_8fffc3kb", "TS0601")
    .tuya_onoff(dp_id=1)
    .tuya_enum(
        dp_id=2,
        attribute_name="work_mode",
        enum_class=WorkMode,
        translation_key="work_mode",
        fallback_name="Work Mode",
    )
    .tuya_number(
        dp_id=3,
        attribute_name="bright_value",
        fallback_name="Brightness",
        translation_key="brightness",
        type=t.uint16_t,
        min_value=10,
        max_value=1000,
        multiplier=1,
        step=10,
    )
    .tuya_number(
        dp_id=4,
        attribute_name="temp_value",
        fallback_name="Light Temperature",
        translation_key="light_temp",
        type=t.uint16_t,
        min_value=0,
        max_value=1000,
        multiplier=1,
        step=10,
    )
    .skip_configuration()
    .add_to_registry()
)
```

puddly, 2025-08-22:
Quirks in issue comments really aren't kept track of. If you'd like to contribute one, please turn it into a pull request!

Rudis1261, 2026-04-03:
Hey, please see this, maybe try it out and add a comment if it helped you get your device to work, I only have the one strip type to test with sadly:
https://github.com/zigpy/zha-device-handlers/pull/4901