zigpy/zha-device-handlers #3973: [Device Support Request]  MOES Tuya ZHA smart curtain switch TS0601 _TZE204_5slehgeo

jsnieuwenhuis, 2025-03-22:
### Problem description

I added the curtain switch (the one with the panel with 3 touch buttons) with ZHA. It doens't show any configuration (like opening, closing, stop), only shows the firmware cxonfiguration.

I tried adding this quirk and included my model numbers:
https://github.com/zigpy/zha-device-handlers/commit/0b97948867e623078ab306d919af48d97994e901

I see the quirk in the device info. But it didn't give any configurations. Can anyone help me solve this?

### Solution description

I would like to see the open,. stop, close buttons (and bakclight) in Home Assistant.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

[Paste the device signature here]

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

jsacra2003, 2025-04-14:
Any fixes??

matte80000, 2025-04-16:
Same issue. Waiting on any solutions as I've already fixed this into the wall and painted around it.

jsacra2003, 2025-04-21:
I managed to fix it by just installing zigbee 2 mqtt, you probably have to upadate your zigbee dongle.
https://www.youtube.com/watch?v=sfsZF0R0HtE
try following one of this guides you will also have to remove the current dongle config before doing this so good luck.

jsnieuwenhuis, 2025-04-23:
I read that works, but that would mean for me changing almost 50 Zigbee devices from ZHA tot mqtt......

For now i bought a Tuya Zigbee bridge. And i load it into HA with the Tuya integration. That works. But I would rather have them directly connected to HomeAssistant.

jsacra2003, 2025-04-23:
Yes I had to do It, but if you keep the entities name your automations will remain the same. I am enjoying the zigbee2Mqtt it has way more functionalities

marcin478, 2025-05-30:
I developed ZHA quirk for this device:
```
from zhaquirks.tuya.ts0601_cover import TuyaMoesCover0601

class MoesCurtainSwitch(TuyaMoesCover0601):
    signature = {
        "models_info": [
            ("_TZE204_5slehgeo", "TS0601"),
        ],
        "endpoints": {
            1: {
                "profile_id": 0x0104,
                "device_type": 0x0051,
                "input_clusters": [
                    0x0000,
                    0x0004,
                    0x0005,
                    0xEF00,
                ],
                "output_clusters": [
                    0x000A,
                    0x0019,
                ],
            },
            242: {
                "profile_id": 0xA1E0,
                "device_type": 0x0061,
                "input_clusters": [],
                "output_clusters": [0x0021],
            },
        },
    }

    replacement = TuyaMoesCover0601.replacement
```

jsnieuwenhuis, 2025-06-28:
Thanks marcin478! I installed in at it works. It’s directly connected with ZHA now. And I have the open/close controls. 

There is only one but: the other controls (light, backlight and turn engine) are not there. Especially the turn engine option is important for me, because now the buttons are turned around (upper button is down) and the status is turned around as well (when it is open it says it is closed).  

Is there any way you can fix this? Thank you very much.

jsnieuwenhuis, 2025-07-08:
I changed the wires and that is working correctly now. Thanks. 

I do notice some bugs. When opening/closing. One way it goes to 0% and than to 255% counts down a little and than jumps to 0%. 

The other way around when it comes to 100% it counts up to 120 switching to opening/closing each 1 percent up. 

That’s only visible in the slider in HA during opening/closing. The actual screen is opening and closing correctly.

TheFreaker86, 2025-07-18:
This is my quirk I developed with support from ChatGPT:

```
from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, Groups, Scenes, Time, Ota
from zhaquirks import Bus
from zhaquirks.const import (
    PROFILE_ID,
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    OUTPUT_CLUSTERS,
    MODELS_INFO,
)
from zhaquirks.tuya import (
    TuyaManufacturerWindowCover,
    TuyaWindowCoverControl,
)

class PatchedTuyaWindowCoverControl(TuyaWindowCoverControl):
    """Tuya Cover Control mit Positionsauswertung und Kalibrierung."""

    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)
        # interne Kalibrierung
        self._top_position = 100
        self._bottom_position = 0

    def cover_event(self, dp, value):
        """Intercept Tuya DPs für Position + Kalibrierung + simuliertes Stop."""

        if dp == 0x04:
            # Kalibrierung: max/min beobachten
            if value > self._top_position:
                self._top_position = value
            if value < self._bottom_position:
                self._bottom_position = value

            # Reale Position berechnen (linear zwischen min/max)
            try:
                span = self._top_position - self._bottom_position
                if span > 0:
                    calibrated_value = int((value - self._bottom_position) * 100 / span)
                else:
                    calibrated_value = value  # Fallback

                self._update_attribute(0x0008, calibrated_value)
                self.debug("Kalibrierte Position: %s %% (Rohwert: %s)", calibrated_value, value)
            except Exception as e:
                self.debug("Fehler bei Kalibrierung: %s", e)

            # Simulierter Stop bei Endlagen
            if value == self._top_position or value == self._bottom_position:
                self.cover_bus.listener_event("cover_stop")

        # Weitergabe an Standard
        super().cover_event(dp, value)

class TZE204_5slehgeo(CustomDevice):
    """Quirk für _TZE204_5slehgeo mit Kalibrierung und Stop-Erkennung."""

    tuya_cover_inverted_by_default = False

    def __init__(self, *args, **kwargs):
        self.cover_bus = Bus()
        self.tuya_cover_command = {
            0: 0,  # open
            1: 2,  # stop
            2: 1,  # close
        }
        super().__init__(*args, **kwargs)

    signature = {
        MODELS_INFO: [("_TZE204_5slehgeo", "TS0601")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: 0x0104,
                DEVICE_TYPE: 0x0051,
                INPUT_CLUSTERS: [0x0000, 0x0004, 0x0005, 0xEF00],
                OUTPUT_CLUSTERS: [0x000A, 0x0019],
            },
            242: {
                PROFILE_ID: 0xA1E0,
                DEVICE_TYPE: 0x0061,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [0x0021],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: 0x0104,
                DEVICE_TYPE: 0x0202,
                INPUT_CLUSTERS: [
                    Basic,
                    Groups,
                    Scenes,
                    TuyaManufacturerWindowCover,
                    PatchedTuyaWindowCoverControl,
                ],
                OUTPUT_CLUSTERS: [
                    Time,
                    Ota,
                ],
            },
        },
    }

```
It includes a simulated positioning reporting entity (x % open, opened, closed) because my setup has a shutter motor with built in limit switches. The switch itself does not know when the shutter is fully opened or closed. Feel free to ask ChatGPT for your own modifications based on this code.

marcin478, 2025-08-09:
@Fr33L4nc3r  I migrated to my Zigbee network to Z2M, where this device works out of the box...

Fr33L4nc3r, 2025-08-10:
@marcin478 Thanks for the suggestion.
Out of curiosity and just to make sure before doing a migration, since you're on Z2M, when you press a button on the switch, is there any event fired that you can rely on in an automation trigger ?
Basically, I'm trying to see if this device could act as a nice touch panel switch (so no wires connected to open / close output)