zigpy/zha-device-handlers #3978: [BUG] _TZE200_ya4ft0w4 target distance not working

Subcode, 2025-03-24:
### Bug description

I bought some new _TZE200_ya4ft0w4's and added them to homeassistant.
it looks like the Target distance never changes or gets updated, Presence Detection and Illuminance work fine.

### Steps to reproduce

- Add _TZE200_ya4ft0w4 to network
- Check Target Distance

### Expected behavior

I would expect Target distance to show the correct value.

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4660,
    "maximum_buffer_size": 108,
    "maximum_incoming_transfer_size": 0,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 0,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0400",
        "0x0406",
        "0xef00"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "_TZE200_ya4ft0w4",
  "model": "TS0601",
  "class": "zigpy.quirks.v2.CustomDeviceV2"
}

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

Subcode, 2025-03-24:
Form what i see, in Zigbee2MQTT it does work, but there the device has a switch to enable the feature.

[image]

TheJulianJES, 2025-04-01:
The Z2M "Find switch" should correspond to the ZHA "LED indicator" indicator, I think.
Does it make a difference if you turn that on?

Maybe it's just a bad name on our side..?

Bruners, 2025-05-05:
I have added a pull request for a different device that seems to work similar to this device and might fix this bug

https://github.com/zigpy/zha-device-handlers/pull/4044