zigpy/zha-device-handlers #3987: [Device Support Request] TS0601 door _TZE200_8whfphjv

VoordeMensen, 2025-03-27:
### Problem description

request for device support for TS0601 door _TZE200_8whfphjv window blinder motor. I've added the model to ts0601_garage.py but to no avail. 

### Solution description

a supported driver

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_8whfphjv",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

{
[... 647 lines cut ...]
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaGarageManufCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

</details>

VoordeMensen, 2025-03-30:
Adding the model to ts0601_cover.py fixed it. I could've thought of that myself.
Oh well I think I did in the end. Anyway. Thanks!

codetwice, 2025-05-29:
I also got a curtain roller, according to the seller the motor brand is Raex but when I plug it into ZHA it reports as `TS0601
by _TZE200_8whfphjv`

I tried applying your quirk and the up/down/stop buttons did show up on Home Assistant but don't work. They give me
`Failed to perform the action cover/close_cover. Failed to close cover: <Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>`

My Zigbee info looks like this:
```
Zigbee info

IEEE: a4:c1:38:44:2c:b3:41:f9
Nwk: 0x281d
Device Type: EndDevice
LQI: 21
RSSI: Unknown
Last seen: 2025-05-29T09:24:48
Power source: Battery or Unknown
Quirk: ts0601_cover.TuyaMoesCover0601
```

How can I further debug this problem?