zigpy/zha-device-handlers #3990: [BUG] _TZE200_ya4ft0w4 Motion sensitivity and Presence sensitivity get reset

Itay1787, 2025-03-30:
### Bug description

Hi, I noticed that Motion sensitivity and Presence sensitivity reset every time on _TZE200_ya4ft0w4
I can configure it, and after about 15 minutes, it resets back to 0 - I just now tried to set it and the Presence sensitivity gets resat after one minute, the motion still set like I set it (for now...)
I tried configuring it through the "Configuration" in the device settings, and also through the "Manage Zigbee device"
It resets in both
In the past, I used to use the quirk where it would also reset, but less frequently
I would appreciate help with this

### Steps to reproduce

Change the Motion sensitivity and Presence sensitivity to a value greater than 0 
Then wait for it to reset

### Expected behavior

to stay on the set value that I set it

### Screenshots/Video

[image]

[image]

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4660,
    "maximum_buffer_size": 108,
    "maximum_incoming_transfer_size": 0,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 0,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0400",
        "0x0406",
        "0xef00"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "_TZE200_ya4ft0w4",
  "model": "TS0601",
  "class": "zigpy.quirks.v2.CustomDeviceV2"
[... 738 lines cut ...]
            },
            "unsupported_attributes": []
          }
        },
        "out_clusters": {}
      }
    }
  }
}

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

TheJulianJES, 2025-04-01:
It seems like this might be a device issue. Related issue when using Zigbee2MQTT:
- https://github.com/koenkk/zigbee2mqtt/issues/26672

We should also follow that issue to see if Z2M comes up with any workarounds or fixes.

Itay1787, 2025-04-02:
> It seems like this might be a device issue. Related issue when using Zigbee2MQTT:
> 
>     * [Tuya ZY-M100-24GV3 Move sensitivity and Presence sensitivity are set to state 0 Koenkk/zigbee2mqtt#26672](https://github.com/Koenkk/zigbee2mqtt/issues/26672)
> 
> 
> We should also follow that issue to see if Z2M comes up with any workarounds or fixes.

Okay, interesting.
If it was a problem with the device and only happened to me, I would ask the seller for a replacement, but if it's a problem with the device itself and not just one broken device, that's a problem...
And I already thought I found a motion sensor that works relatively well and isn't expensive... Luckily, I didn't order more of those (I hope there's a solution for this soon)
I have a Tuya Hub. I wonder if it also behaves like this in its native application and resets its values. I'll check and update here after more tests on that.
But what's strange is that it didn't behave this way at first...
I also have a Hub that should arrive soon for Z2M. I'll try to add it to Z2M after resetting it, maybe I'll have better luck there - but as it seems, I don't expect anything different.

DelBierzo, 2025-04-10:
Hi! 
The same thing happens to me, ( Tuya ZY-M100-24GV3 | TS0601 | _TZE200_ya4ft0w4 )
Some times after 15 minutes, and others 60 minutes, **motion sensitivity** and **presence sensitivity** it resets back to 0
If **Find switch**, is turned off this doesn't happen.

gekkehenkie11, 2025-05-22:
Yeah I found the problem. The "multiplier=0.1" line should be deleted for:
    .applies_to("_TZE200_ya4ft0w4", "TS0601")
    .applies_to("_TZE204_ya4ft0w4", "TS0601")

So it should look like this:

    .tuya_number(
        dp_id=102,
        attribute_name="presence_sensitivity",
        type=t.uint16_t,
        min_value=0,
        max_value=10,
        step=1,
        translation_key="presence_sensitivity",
        fallback_name="Presence sensitivity",
    )

After that it works fine. Also if someone is going to patch this please also change the name for id 101, this is not a LED, but this is a switch for the distance measurement (you can turn distance measurement on/off with this switch). Works fine but name should be changed. Thanks

DelBierzo, 2025-05-23:
This is how I have it fixed in my quirk. But **when a power loss occurs, it resets to 0**; the firmware doesn't save the configuration. Firmware problem!
With the help of an automation, when Home Assistant restarts or when this value changes to 0, it can be returned to the required value.

Bruners, 2025-05-31:
> [@TheJulianJES](https://github.com/TheJulianJES)
> 
> Yeah I found the problem. The "multiplier=0.1" line should be deleted for: .applies_to("_TZE200_ya4ft0w4", "TS0601") .applies_to("_TZE204_ya4ft0w4", "TS0601")
> 
> So it should look like this:
> 
> ```
> .tuya_number(
>     dp_id=102,
>     attribute_name="presence_sensitivity",
>     type=t.uint16_t,
>     min_value=0,
>     max_value=10,
>     step=1,
>     translation_key="presence_sensitivity",
>     fallback_name="Presence sensitivity",
> )
> ```
> 
> After that it works fine. Also if someone is going to patch this please also change the name for id 101, this is not a LED, but this is a switch for the distance measurement (you can turn distance measurement on/off with this switch). Works fine but name should be changed. Thanks

Could you do a test with my pull request on this sensor to see if they are the same?
https://github.com/zigpy/zha-device-handlers/pull/4044

gekkehenkie11, 2025-06-05:
@DelBierzo Well what makes it worse is that the sensor schedules a reboot every 3 days, causing the settings to lose!!

I looked into all this, Istudied the manual which can be found here:
https://www.gigadevice.com.cn/Public/Uploads/uploadfile/files/20250427/GD32E23x_User_Manual_Rev2.4.pdf

So the right place to store user variables to flash seems to be in the 16 bytes "Option byte Block". I dumped the block with the debugger but it's not used (holding 0xA5, 5A, FF, 00, FF, FF, FF, FF, FF, FF etc). Which makes sense since I could not find any references to it in the firmware code.

So then I thought, in theory it's possible that they save variables to the main flash, although that would be pretty stupid, so I changed vars, dumped the whole main flash again and compared, was exactly the same as my other dumps. So that's also not it. Then there's a 3 kb flash block for the boot loader, seems highly unlikely that it was stored there (if that's even possible, I assume it's not) but I checked again, changed vars, dumped it again and was the same.

So in short at this point I start to think that the sensor does not even store settings AT ALL. Which is pretty stupid, especially combined with a scheduled power cycle every 3 days, LOL. The thing I don't understand though is how it's possible that the "fading time" setting actually does seem to survive a power cycle ...

Anyway, although it's possible to modify the firmware to store the setttings to that "option byte block", and restore them on a reboot, that's actually a lot of work. Maybe if there's some example assembly code to use that it would be easier. But if not and you'd have to start from scratch figuring it all out, it's a quite a bit of work.

As a work around I guess it's best to set up an automation in HA that changes settings back if they change to a different value.

gekkehenkie11, 2025-06-05:
@Bruners hey mate, I'm not sure what you want me to do exactly? I don't own that sensor.

gekkehenkie11, 2025-06-05:
BTW @Bruners just out of curiosity, do your motion sensitiy and presence sensitivity values in HA also get reset to 1 when you unplug and plug in the sensor? I just noticed that with the original tuya app this doesn't happen. Not sure if that's because the tuya app just sends out the values to the sensor when they connect.

gekkehenkie11, 2025-06-05:
@Bruners And most importantly I just did a quick log and only found values 0, 1, 2 for DP 1. So in which case it would seem that the DP code I was using was correct and yours is not suitable for my sensor. Then again, I will log some more time to see if other value's pop up. But honestly if that was the case I would have most likely found 'weird' behavior in the presence sensor and that's not been the case ...

gekkehenkie11, 2025-06-05:
@Bruners I logged my data for more than an hour and only have seen values 0, 1 and 2 for DP 1. So I'm pretty confident that your sensor is different than mine.

DelBierzo, 2025-06-06:
This is my automation for the sensor. I don't remember where I saw it, but I've modified it to my liking.

When reboot or whenever change any of the values, it resets them to my default values.

I also have it on another sensor that works well, but I make sure they are always correct.
With this it works perfectly and doesn't give me headaches. 

```
alias: Invalid Value Error For Tuya Presence Sensor
description: ""
triggers:
  - trigger: homeassistant
    event: start
  - type: turned_on
    device_id: b6df106ff3ffb3ae53c45bc256f68758
    entity_id: 6034e3e730f0cc014bb6eaf9daba0353
    domain: switch
    trigger: device
  - trigger: numeric_state
    entity_id:
      - number.sensor_presencia_tuya_estudio_alcance_maximo
    below: 4.5
  - trigger: numeric_state
    entity_id:
      - number.sensor_presencia_tuya_estudio_alcance_maximo
    above: 4.5
  - trigger: numeric_state
    entity_id:
      - number.sensor_presencia_tuya_estudio_alcance_minimo
    above: 0.75
  - trigger: numeric_state
    entity_id:
      - number.sensor_presencia_tuya_estudio_alcance_minimo
    below: 0.75
    enabled: true
  - trigger: numeric_state
    entity_id:
      - number.sensor_presencia_tuya_estudio_sensibilidad_al_movimiento
    below: 7
  - trigger: numeric_state
    entity_id:
      - number.sensor_presencia_tuya_estudio_sensibilidad_al_movimiento
    above: 7
  - trigger: numeric_state
    entity_id:
      - number.sensor_presencia_tuya_estudio_sensibilidad_de_presencia
    below: 7
  - trigger: numeric_state
    entity_id:
      - number.sensor_presencia_tuya_estudio_sensibilidad_de_presencia
    above: 7
  - trigger: numeric_state
    entity_id:
      - number.sensor_presencia_tuya_estudio_tiempo_de_desvanecimiento
    below: 10
  - trigger: numeric_state
    entity_id:
      - number.sensor_presencia_tuya_estudio_tiempo_de_desvanecimiento
    above: 10
conditions: []
actions:
  - alias: Distance switch
    choose:
      - conditions:
          - condition: device
            type: is_on
            device_id: b6df106ff3ffb3ae53c45bc256f68758
            entity_id: 6034e3e730f0cc014bb6eaf9daba0353
            domain: switch
        sequence:
[... 82 lines cut ...]
        sequence:
          - device_id: b6df106ff3ffb3ae53c45bc256f68758
            domain: number
            entity_id: 0c67aeb27ebd3d14571a27777f66fbd9
            type: set_value
            value: 7
  - alias: Fade-out time
    choose:
      - conditions:
          - condition: or
            conditions:
              - condition: numeric_state
                entity_id: number.sensor_presencia_tuya_estudio_tiempo_de_desvanecimiento
                below: 10
              - condition: numeric_state
                entity_id: number.sensor_presencia_tuya_estudio_tiempo_de_desvanecimiento
                above: 10
        sequence:
          - device_id: b6df106ff3ffb3ae53c45bc256f68758
            domain: number
            entity_id: 3caab84a40da648f19ea91067a951966
            type: set_value
            value: 10
mode: restart
```

gekkehenkie11, 2025-06-06:
Yeah it's super simple to set it up with the automation builder. You only need to specify: When (the entity) motion sensitivity  for your sensor is below value X, then do action Set value (device) motion sensitivity.

What caught me off guard is that you if try it manually and then change the value 1 it sometimes LOOKS like nothing happened. But if you refresh the page you'll see it worked.

gekkehenkie11, 2025-06-06:
BTW thinking about this, I think you don't even need to do the 'home assistant on boot' part. The sensor periodically sends all settings via zigbee. So HA will receive the settings after a reboot quite soon anyway and then correct them.

Bruners, 2025-06-10:
> [@Bruners](https://github.com/Bruners) I just checked your code. I then hooked up my sensor to a zigbee gateway in combination with the original tuya app to see if the settings are right. First of all interesting find that the minimum for both sensitivities should be 1, not 0 ! I never realized that that the original range is 1-10 instead of 0-10. My sensor worked with setting it to '0' but I have no idea what it exactly did in that mode LOL.
> 
> Some comments:
> 
> * DP 1: I havent checked the different Presence states. I guess I could find them in the firmware code. Did you ensure this is correct (TuyaPresenceStateV02) and why did you comment "needs more work" ?
>   *DP 2: I see in my original Tuya app that it calls DP 2 "motion sensitivity", while your code calls it "radar sensitivity", maybe then better to stick to the original name? Although both are fine of course IMHO.
>   *DP 3: min distance, in my original tuya app it's a range between 0.00 and 8.25 with 0.75 steps. Isn't this the case for your sensor? So the 'base' DP actually seems fine here and I don't think we need to override it for my sensor.
>   *DP 4: max distance, same thing as above.
>   *DP 9: hmm I'll need to look into this ...
>   *DP 104: Is there any use for this info? I couldn't really think of anything useful. I saw that there was no code to process in my current implemention and I didn't miss it. So I actually figured, let's kill those messages in the firmware, to make the sensor less spammy: https://github.com/gekkehenkie11/ZY-M100-Sensor-patching So in other words, I can't check this for you since my (patched) sensor doesn't send this info anymore LOL.
>   *DP 105: the range in the original app is 1-15000
> 
> Other than that some interesting tweaks! But I think it's important to make sure what I said about DP 3, DP 4 and DP 105. If it's different in your sensor then I guess we should use different code.
> 
> _EDIT_ I verified DP 9, nice find, this seems to be correct! I never really paid much attention to it, in fact, I quickly disabled the switch. But yeah now the values actually make sense :)

I don't have a Tuya hub so i can't check the sensor with it, the code is based on data from z2m and other sensors in zigpy. My sensor did not like setting the values to 0, had to reset it and pair it again to make it report readings again. I would say the correct should be 1-10 then. 

DP1: This is why i have written "needs more work" since i don't exactly know what the states imply since the sensor is not really giving me any reliable test cases that makes sense, both 1-2 are shown both when moving and when still. I have had it report 3 and 4 once in a wierd case, but never managed to trigger it again. What does the Tuya app say for the states?

DP2: Motion sensitiviy is probably the correct term, i have yet been able to tell the difference in how the sensor reports movement.

DP3/DP4: I will test my sensor with theese steps, this is where the z2m code is different between _TZE204_gkfbdvyx and _TZE200_ya4ft0w4.

DP101 Turns distance tracking on or off on my device, I have read that some devices has been spammy with this even if it is turned off. I have not noticed anything from mine (unless i'm looking on the wrong location).

DP102: What does the Tuya app call this? Radar sensitivity?

DP104: Depends, my first guess was that it is supposed to say if it detects the Presence of people or not, as in not moving. I have not really found it usefull since DP1 is there as reporting the same info?

DP105: What is it called in the Tuya app? not really sure what it actually does, is it the time for the sensor to reset if movement or Presence is detected?

gekkehenkie11, 2025-06-25:
@Bruners hey mate I bought some new sensors and they came as _TZE200_gkfbdvyx TS0601. I tried your code and it seems mostly to mimic the original Tuya app, except for DP 1, which shows in the app as either "none, presence, move", so I'm pretty sure that needs changing. Also for DP 105 the values in the app are:
        min_value=5,
        max_value=15000,
        step=5,

And the, like said above, better use "Motion sensitivity" for DP2 as the name. Other than that it all seems perfect! 
BTW if you're going to merge that, could you also add my sensor? Since yours is TZE204, and mine TZE200. Thanks!

rrooggiieerr, 2025-07-15:
Ok, thanks for confirming!

I created a pull request to change the name to 
"Distance tracking": https://github.com/zigpy/zha-device-handlers/pull/4187

And an other one to fix the Presence sensitivity multiplier: https://github.com/zigpy/zha-device-handlers/pull/4188