zigpy/zha-device-handlers #3993: [BUG] Tuya Water Valve (TS0049 by _TZ3210_0jxeoadc)

vikitor87, 2025-04-01:
### Bug description

As I mentioned in the aforementioned thread, after a 2025 HA core update, the ZigBee device went from working correctly with a custom quirk to not working correctly with the native quirk.

The same device, with the model name "ru41azca," continues using the custom quirk and works fine.

With the native quirk, the device detects more entities but doesn't respond to their use. All it does is display status changes when you press the device's physical button, but it also doesn't support irrigation time values.

### Steps to reproduce

1. Settings
2. Devices & Services
3. ZigBee Home Automation
4. Devices
5. TS0049 by TZ3210_0jxeoadc
6. ZigBee info
7. Quirk: zhaquirks.tuya.builder.EchantedDeviceV2

### Expected behavior

The device should display the control switch, firmware, and battery level. The ru41azca model using the ts0049_irrigationvalve.py custom quirk works correctly and responds to instructions. The irrigation time can be modified from the advanced management.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

[image]

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0006",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
[... 886 lines cut ...]
              }
            },
            "unsupported_attributes": []
          }
        }
      }
    }
  }
}

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

TheJulianJES, 2025-04-01:
Thanks for the report. I'll prepare another quirks version that should fix this, hopefully still for HA Core 2025.4.0.
You'll see a "Bump ZHA to ..." in the HA changelog if there's another beta where it's already included.

Ping me if this issue somehow still isn't resolved then, so this can be reopened in that case.

---

Also, please check if the battery sensor works correctly. Just from looking at the code, it seems like it won't.
Please create a new issue if that's the case. You can just copy/paste most of this. Just to make sure we're tracking that issue then.

And lastly, please open an issue where you provide the signature for the `ru41azca` model you mentioned. Please also provide the custom quirk file you're using for that, even if it's the same as for this one, but the exact file you're using wasn't linked here.
If the v2 quirk generally works, we should add that model as well.

closed 2025-04-01 by TheJulianJES

pnneil, 2025-04-02:
Hi, 

Thanks for your work on this, the quirk has been an "orphan" for some time :)
One thing I think I have noticed is that the code doesn't seem to reference DP110 but it is used in the original custom quirk.  when using that quirk I observed that this is modified from b'\x00\x00' to b'\x00\x01' when the switch is turned on manually, not in HA though.  (I observed this by reading the "dp_110 (id: 0xef02)" attribute of the "TuyaValveFamilyCluster (Endpoint id: 1, Id: 0xef00, Type: in)" cluster.)
I hope this is helps, if not you know what to do :)
Cheers.

TheJulianJES, 2025-04-02:
The previous quirks I can find for this device just mapped datapoint 110 to a fake ZCL attribute, but didn't do anything more with it, so I don't think we need it.

pnneil, 2025-04-02:
I have gone back to the original quirk I cobbled together and from memory it wouldn't without it.

See attached.

[TS0049_valve_original.txt](https://github.com/user-attachments/files/19561403/TS0049_valve_original.txt)

Unless of course I am doing something wrong and can't break it now :)

813648, 2025-04-03:
It is back in active life with the update 2025.4. Thank you very much!!!

pnneil, 2025-04-06:
Yes thanks a lot. I have just installed 2025.4.1 and all is working well.

MBRetrofit, 2025-07-28:
For me is not working. I am not in the location so I can not re-pair

[image]

 and the _TZ3210_0jxeoadc only respond Interviewing and display a big error when I press Reconfigure.