zigpy/zha-device-handlers #3998: [Device Support Request] Smart Radiator Thermostat (_TZE200_cxakecfo)

jrfbal, 2025-04-02:
### Problem description

Hello,
Bought a Radiator Thermostat branded Girier from Aliexpress (https://aliexpress.com/item/1005008013900913.html?spm=a2g0o.order_list.order_list_main.46.6245caa4Z6mAja&gatewayAdapt=glo2bra), it shows in HA but does not have any entities showing.
It seems that is possible to integrate via zigbee2mqtt, but I prefer not to go via that route (https://github.com/Koenkk/zigbee2mqtt/discussions/26202)

### Solution description

Device pairs with HA but no entities show.

### Screenshots/Video

[image]

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0402",
        "0x0405",
        "0x040d",
        "0x042a",
        "0x042b",
        "0x042e",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_c2fmom5z",
  "model": "TS0601",
  "class": "zigpy.quirks.v2.CustomDeviceV2"
}
[... 947 lines cut ...]
</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

JCWardle, 2025-05-14:
Hey jrfbal - just wanted to report I've bought the same device and have the same issue.

schabau, 2025-07-15:
Hello,
In case anyone else wants to do it this way, I downloaded the tuya_trv.py file here, which already supports the very similar Avatto ME168 ("_TZE200_p3dbf6qs" ; "_TZE200_rxntag7i"), and added line 403.

`.applies_to("_TZE200_cxakecfo", "TS0601") # model: 'ME168', vendor: 'Girier'`

I also added line 457.

`.tuya_battery(dp_id=6)`

This is basically how the thermostat with ZHA works, and the battery is also displayed in the overview.

Best regards,
Detlef Paschke

[image]

jrfbal, 2025-07-15:
Hello,
Thank you for your feedback @schabau, unfortunately not only it doesn't show the battery correctly for me, but I also cannot adjust the target temperature with your fix, just the offset.
Thank you

schabau, 2025-07-15:
I had to re-link the thermostats.
I removed the thermostats and re-linked them, and then everything worked.

closed 2026-01-20 by github-actions[bot]