zigpy/zha-device-handlers #4015: [Device Support Request] Tuya Vertical Dream Blinds

t4chik0ma, 2025-04-17:
### Problem description

Can anyone help make a quirk for this thing? Vibed my way through gettings these blinds accepting open/close/stop+dim left/dim right in zigbee2mqtt, though am struggling to convert it into the format needed to make it work in HA w/ ZHA - every time I re-pair in HA it comes back in with zero entities to control. 

### Solution description

Working dream blinds control via ZHA

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_wzre8hu2",
  "model": "TS0601",
  "class": "dreamblindsquirkopenai.DreamBlinds"
}

```

</details>
[... 164 lines cut ...]
                await entity.command(
                    'manuSpecificTuya',
                    'dataRequest',
                    { seq, dpValues: [{ dp, datatype: 4, data: [dpVal] }] },
                    { disableDefaultResponse: true }
                );
                return { dim: value };
            },
        },
    ],
    exposes: [
        e.enum('state', exposes.access.STATE_SET, ['OPEN', 'CLOSE', 'STOP']),
        e.numeric('position', exposes.access.STATE_SET)
            .withValueMin(0)
            .withValueMax(100)
            .withUnit('%'),
        e.enum('dim', exposes.access.STATE_SET, ['left', 'right', 'stop']),
    ],
};

module.exports = definition;

```

</details>

t4chik0ma, 2025-04-17:
Blinds instruction sheet for the blinds motor in case it's useful to anyone

[image]
[image]

Ptitdyno, 2025-04-22:
Hello t4chik0ma

I just bought a TZE204_wzre8hu2 curtain motor and thanks to your code provided I was able to integrate it in Zigbee2mqtt

I don’t know how to help you with your problem because me under Jeedom/ Z2m ca works even after a redemarrage

my problem is that the position does not work, I have no up from 0 to 100

Would you know where this could come from?

Thank you

[image]

t4chik0ma, 2025-05-11:
Some progress... got open/close/stop controls working now. Don't have tilt working just yet. 

[image]

[image]

`from zigpy.profiles import zha
from zigpy.zcl.clusters.general import Basic, Groups, Scenes, Time, Ota
from zhaquirks import CustomDevice
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaManufCluster,
    TuyaWindowCover,
    TuyaWindowCoverControl,
    TuyaManufacturerWindowCover,
)

class TuyaDreamBlindsCustom(TuyaWindowCover):
    """Working quirk for Dreamblinds (_TZE204_wzre8hu2 TS0601)."""

    signature = {
        MODELS_INFO: [
            ("_TZE204_wzre8hu2", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: 0x0104,
                DEVICE_TYPE: 0x0051,
                INPUT_CLUSTERS: [
                    0x0000,
                    0x0004,
                    0x0005,
                    0xef00,  # Tuya manufacturer cluster
                ],
                OUTPUT_CLUSTERS: [
                    0x000a,
                    0x0019,
                ],
            },
            242: {
                PROFILE_ID: 0xa1e0,
                DEVICE_TYPE: 0x0061,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [0x0021],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic,
                    Groups,
                    Scenes,
                    TuyaManufacturerWindowCover,  # Handles DP 0 (position)
                    TuyaWindowCoverControl,       # Provides open/close/stop
                ],
                OUTPUT_CLUSTERS: [
                    Time,
                    Ota,
                ],
            },
            242: {
                PROFILE_ID: 0xa1e0,
                DEVICE_TYPE: 0x0061,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [0x0021],
            },
        },
    }
`

t4chik0ma, 2025-05-11:
> Hello t4chik0ma
>
> I just bought a TZE204_wzre8hu2 curtain motor and thanks to your code provided I was able to integrate it in Zigbee2mqtt
>
> I don’t know how to help you with your problem because me [...]

Not sure on that one sorry. Slider control in z2m isn't one I had tried for long to get working.

Ptitdyno, 2025-05-12:
Hello thank you for your return,

I don’t see how to adapt it for Z2m

good luck

t4chik0ma, 2026-10-20:
### Full Solution: Working Continuous Tilt / Dimming + ACK Flood Fix

Coming back to this after digging into the raw Zigbee clusters and live Tuya DP packet sniffers in Home Assistant 2026.10. We finally have 100% working open/close/stop traverse **and full continuous tilt angle control (0° to 180° / 0% to 100%)** working seamlessly in ZHA!

---

#### 1. The Key Discoveries & Datapoints
* **Continuous Tilt is DP 21 (`0x15`, Type 2 VALUE)**:
  * The motor accepts and reports the tilt angle directly in degrees: `0` to `180`.
  * `0°`: Dimmed completely in direction A (closed).
  * `90°`: Perfectly centered (slats perpendicular / sheer open for daylight).
  * `180°`: Dimmed completely in direction B (closed).
  * We map this linearly to Home Assistant's `0%` – `100%` tilt percentage (`tilt_pct = round(angle / 1.8)` and `angle = round(target_pct * 1.8)`).
* **DP 20 (`0x14`, Type 4 ENUM)**:
  * This is merely a directional motion status indicator (`0` = dimming left, `1` = dimming right, `2` = stopped) sent while the motor runs. Relying on DP 20 for position clobbers the real angle. DP 21 is the true angle attribute.
* **Fixing the 5Hz Motor Radio Retransmit Flood**:
  * In standard ZHA handling, if incoming Tuya reports do not receive an explicit default response frame, these motors get trapped in a ~200ms retransmit loop that spams the Zigbee mesh coordinator.
  * Adding `self.send_default_rsp(hdr, foundation.Status.SUCCESS)` in `handle_cluster_request` immediately ACKs reports and halts the retransmit storms.
* **Unlocking Tilt Controls in Home Assistant (`supported_features: 255`)**:
  * ZHA masks out tilt capabilities unless the `WindowCovering` cluster returns a covering type with tilt (e.g. `Tilt_blind_tilt_and_lift` = `8`). Overriding `get()` on `TuyaDreamBlindsControl` ensures HA exposes both lift and tilt controls (`supported_features: 255`).

---

#### 2. Working Custom Quirk

Save this file as `dreamblinds.py` inside your Home Assistant `/config/custom_zha_quirks/` directory:

```python
"""Quirk for Tuya Vertical Dream Blinds (_TZE204_wzre8hu2 TS0601)."""

import logging
from typing import Any
from zigpy.profiles import zha
from zigpy.zcl import foundation
from zigpy.zcl.clusters.closures import WindowCovering
from zigpy.zcl.clusters.general import Basic, Groups, Scenes, Time, Ota
from zhaquirks import CustomDevice
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    COVER_EVENT,
    TUYA_GET_DATA,
    TUYA_SET_DATA,
    TUYA_SET_DATA_RESPONSE,
    TuyaManufCluster,
    TuyaWindowCover,
    TuyaWindowCoverControl,
    TuyaManufacturerWindowCover,
)

_LOGGER = logging.getLogger(__name__)

class InvertedTuyaManufacturerWindowCover(TuyaManufacturerWindowCover):
    """Invert position, handle DP 21 continuous angle tilt, and ACK incoming reports."""

    def handle_cluster_request(
        self,
        hdr: foundation.ZCLHeader,
        args: list[Any],
        *,
        dst_addressing: Any = None,
    ) -> None:
        if hdr.command_id in (TUYA_GET_DATA, TUYA_SET_DATA_RESPONSE):
            tuya_payload = args[0]
[... 130 lines cut ...]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic,
                    Groups,
                    Scenes,
                    InvertedTuyaManufacturerWindowCover,
                    TuyaDreamBlindsControl,
                ],
                OUTPUT_CLUSTERS: [
                    Time,
                    Ota,
                ],
            },
            242: {
                PROFILE_ID: 0xa1e0,
                DEVICE_TYPE: 0x0061,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [0x0021],
            },
        },
    }
```

Happy to open a PR to merge this into `zhaquirks/tuya/ts0601_cover.py` (or as a standalone quirk) if the maintainers would like it merged upstream!

t4chik0ma, 2026-10-20:
Official PR submitted upstream: https://github.com/zigpy/zha-device-handlers/pull/5478 🎉