zigpy/zha-device-handlers #4018: [BUG] Unknown battery

NSJ84, 2025-04-19:
### Bug description

TS0207 _TZ3210_tgvtvdoc Solar Rain sensor shows unknown battery status.

### Steps to reproduce

Developer tools
States
State: unknown

### Expected behavior

It should show the battery status in %

### Screenshots/Video

State: unknown

state_class: measurement
battery_quantity: 1
battery_size: Other
unit_of_measurement: "%"
device_class: battery
friendly_name: _TZ3210_tgvtvdoc TS0207 Batteri

### Device signature

{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0402",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0400",
        "0x0500",
        "0xef00"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0006",
        "0x000a",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "_TZ3210_tgvtvdoc",
  "model": "TS0207",
  "class": "zigpy.quirks.v2.CustomDeviceV2"
}

### Diagnostic information

[zha-01JS1MGNR3XFRT5P7HXDQ10B05-_TZ3210_tgvtvdoc TS0207-9ea00f9543212a5017c7ac6146c402c6.json](https://github.com/user-attachments/files/19821181/zha-01JS1MGNR3XFRT5P7HXDQ10B05-_TZ3210_tgvtvdoc.TS0207-9ea00f9543212a5017c7ac6146c402c6.json)

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

JAudi23, 2026-11-09:
Perfect, this procedure worked for me, and I see the battery percentage again.
For the record, I did not have to physically press the button on the sensor device.  I clicked reconfigure from the device page, and it picked up the battery info after a minute or so of reconfiguring.

I'll monitor over the next few days to confirm nothing else is impacted.

gianlucaromito, 2026-11-09:
Tried right now. It works! No phisical press, just reconfigure. Thank you :)