zigpy/zha-device-handlers #4024: [Device Support Request] Aqara Smart Pet Feeder C1 - Unable to configure onboard scheduling

Managarmr719, 2025-04-23:
### Problem description

There's currently no way to configure the onboard schedule through ZHA.
It has a toggle between "Manual" and "Schedule" but it doesn't actually have a way to configure said schedule.
I'm currently using an automation as a workaround but this makes it so I'm dependant on HA to make it work.
The device has battery backup so ideally, it should work even if the power went out or if for some reason there's an issue with the zigbee connection.

### Solution description

Have an option to configure the onboard schedule.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

[Paste the device signature here]

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

iamjoshk, 2025-06-11:
I really want the onboard scheduling to work, too. I've been working on this myself and while I have made some progress, it's pretty rough at the moment. Hope to share a PR for the quirk soon.

BernardoCBranco, 2025-12-10:
The issue has not been resolved, it still exists on the latest version.