zigpy/zha-device-handlers #4026: [Device Support Request] Gosund Model:TS17 / TS0601 by _TZE204_d7lpruvi

zsoltking, 2025-04-24:
### Problem description

I solved it for myself.
Now I'm just sharing it with the community.
My solution is in the Custom quirk.
Device signature also added.

### Solution description

Please add the device into the tuya_sensor.py

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0402",
        "0x0405",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE204_d7lpruvi",
  "model": "TS0601",
  "class": "zigpy.quirks.v2.CustomDeviceV2"
}

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

(
    TuyaQuirkBuilder("_TZE200_a8sdabtg", "TS0601")  # Variant without screen, round
    .applies_to("_TZE200_qoy0ekbd", "TS0601")
    .applies_to("_TZE200_znbl8dj5", "TS0601")
    .applies_to("_TZE200_zppcgbdj", "TS0601")
    .applies_to("_TZE204_s139roas", "TS0601")
    .applies_to("_TZE200_s1xgth2u", "TS0601")  # Nedis ZBSC30WT
    .applies_to("_TZE204_d7lpruvi", "TS0601")  # Gsound ST17
    .tuya_temperature(dp_id=1, scale=10)
    .adds(TuyaTemperatureMeasurement)
    .tuya_humidity(dp_id=2)
    .tuya_battery(dp_id=4)
    .skip_configuration()
    .add_to_registry()
)

```

</details>

### Additional information

Working well for me.

zsoltking, 2025-07-18:
I had some time to check it, and yes I missed the battery part.
I created a new config (it's converting raw data to 100%, 50% and 25%):
```
(
    TuyaQuirkBuilder("_TZE204_d7lpruvi", "TS0601")
    .tuya_temperature(dp_id=1, scale=10)
    .adds(TuyaTemperatureMeasurement)
    .tuya_humidity(dp_id=2)
    .tuya_dp(
        dp_id=3,
        ep_attribute=TuyaPowerConfigurationCluster2AAA.ep_attribute,
        attribute_name="battery_percentage_remaining",
        converter=lambda x: {0: 50, 1: 100, 2: 200}[x],
    )
    .adds(TuyaPowerConfigurationCluster2AAA)
    .skip_configuration()
    .add_to_registry()
)
```
The battery config is copied from another device. My sensor using not 2 but 3 AAA batteries, but it's show 100% correctly.

lu-krin, 2025-08-02:
> I had some time to check it, and yes I missed the battery part. I created a new config (it's converting raw data to 100%, 50% and 25%):
>
> ```
> (
>     TuyaQuirkBuilder("_TZE204_d7lpruvi", "TS0601") [...]

Thank you, I confirm this works! Would you be able, by any chance, to also solve the reset issue which is apparently known with this device? Basically pressing the backlight button at the top causes the device to lose connection and the attempts to re-connect take a long time and incredibly reduce the battery.

closed 2026-02-06 by github-actions[bot]