zigpy/zha-device-handlers #4028: [Device Support Request]  TS0601 by _TZE284_a14rjslz  3P4W ZigBee smart meter

rseste, 2025-04-25:
### Problem description

I bought this device on [aliexpress](https://pt.aliexpress.com/item/1005007984838422.html?gps-id=pcStoreLeaderboard&scm=1007.22922.271278.0&scm_id=1007.22922.271278.0&scm-url=1007.22922.271278.0&pvid=05eb4ffe-d2c1-41ab-941c-3cfbcc812ed8&_t=gps-id%3ApcStoreLeaderboard%2Cscm-url%3A1007.22922.271278.0%2Cpvid%3A05eb4ffe-d2c1-41ab-941c-3cfbcc812ed8%2Ctpp_buckets%3A668%232846%238109%231935&pdp_ext_f=%7B%22order%22%3A%22141%22%2C%22eval%22%3A%221%22%2C%22sceneId%22%3A%2212922%22%7D&pdp_npi=4%40dis%21BRL%21266.96%21239.76%21%21%21325.76%21292.57%21%402101e9a217456182869418500e266e%2112000043163896937%21rec%21BR%21%21ABXZ&spm=a2g0o.store_pc_home.smartLeaderboard_2008782814175.1005007984838422&gatewayAdapt=glo2bra) 
I was able to make changes on the quirk signature to make ZHA recognize it but it does not show any values, the entity sensors seams to be mapped wrongly. (only Summation delivered fields has values

 

### Solution description

I would like to know if someone have a correct quirt for this device? 
if not how complex is to discover the correct sensor/properties/entities exposed by the device? 
I can try to do that but I'm new to ZHA and quirks Any help to reverse engineering is welcome.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4098,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0702",
        "0x0b04",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "16": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0006"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "_TZE284_a14rjslz",
  "model": "TS0601",
  "class": "ts0601_din_power.HikingPowerMeter"
}

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

IgorMilavec, 2025-05-25:
Related implementation for zigbee2mqtt: https://github.com/Koenkk/zigbee2mqtt/discussions/25937

ssnspp, 2025-12-23:
I have similar device, same model_info, but different brand.
I tried create ZHA quirk file for my device.

```
"""
ZHA Quirk for Ourtop 3-Phase Energy Meter (ATMS10013Z3).
Logic based on Zigbee2MQTT PhaseVariant2 implementation.
"""

"""
This quirk created with help of Gemini
Ourtop 3-phase Energy Meter
In home assistant will appear 3 AC Frequency sensor, but this device will not produce AC Frequency data.
also will appear 3 power factor sensor, but this device will produce only 1 power factor data.
this device also produce 1 energy data, so can't be used as 3 1-phase energy meter,
eventough will produce 3 voltage, current, and power data.
"""

import logging
from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
#import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering
from zhaquirks.tuya import TuyaLocalCluster
from zhaquirks.tuya.mcu import TuyaMCUCluster

_LOGGER = logging.getLogger(__name__)

# DP Mapping from Z2M
DP_ENERGY_FORWARD = 1   # Forward Active Energy
DP_PHASE_A = 6          # Packed: V, A, P (8 bytes)
DP_PHASE_B = 7          # Packed: V, A, P (8 bytes)
DP_PHASE_C = 8          # Packed: V, A, P (8 bytes)
DP_ENERGY_REVERSE = 23  # Produced Energy
DP_ENERGY_TOTAL = 24    # Produced + Consumed
DP_POWER_TOTAL = 29     # Total Active Power
DP_REACTIVE_TOTAL = 30  # Total Reactive Power
DP_PF_TOTAL = 50        # Overall Power Factor

class OurtopManufCluster(TuyaMCUCluster):
    """Tuya MCU Cluster with Z2M Packed Payload Decoding."""

    def handle_cluster_request(self, hdr, args, *extra_args, **kwargs):
        if hdr.command_id != 0x02:
            return super().handle_cluster_request(hdr, args, *extra_args, **kwargs)

        dps = []
        try:
            if hasattr(args, "data") and hasattr(args.data, "datapoints"):
                dps = args.data.datapoints
            elif isinstance(args, list) and hasattr(args[0], "datapoints"):
                dps = args[0].datapoints
        except Exception:
            return

        _LOGGER.debug("Extracted DPs for processing: %s", dps)

        for status in dps:
            try:
                dp = status.dp
                data = status.data.payload

                if dp == DP_ENERGY_FORWARD:
                    val = int(data)

                    cluster = self.endpoint.device.endpoints[1].in_clusters.get(Metering.cluster_id)
                    if cluster:
                        cluster._update_attribute(0x0000, val)
[... 75 lines cut ...]
            }
        },
    }

    replacement = {
        "endpoints": {
            1: { # Phase A + Totals
                "input_clusters": [Basic.cluster_id, Groups.cluster_id, Scenes.cluster_id, 
                                 OurtopManufCluster, ElectricalMeasurementPhase, MeteringCluster],
                "output_clusters": [Time.cluster_id, Ota.cluster_id],
            },
            2: { # Phase B
                "profile_id": zha.PROFILE_ID,
                "device_type": zha.DeviceType.SMART_PLUG,
                "input_clusters": [ElectricalMeasurementPhase],
            },
            3: { # Phase C
                "profile_id": zha.PROFILE_ID,
                "device_type": zha.DeviceType.SMART_PLUG,
                "input_clusters": [ElectricalMeasurementPhase],
            },
        },
    }

```