zigpy/zha-device-handlers #4032: [Device Support Request] Aqara Dimmer Switch H2 EU

BramMusters, 2025-04-28:
### Problem description

[Product URL](https://eu.aqara.com/en-eu/products/aqara-dimmer-switch-h2-eu)

The dimmer is recognized and I can control the on/off state, but I cannot control brightness and other stuff.

Zigbee2mqtt seems to have full support for this:
https://www.zigbee2mqtt.io/devices/KD-R01D.html
https://github.com/Koenkk/zigbee-herdsman-converters/pull/8896

### Solution description

1. Brightness
2. Any other useful entities.

### Additional information

This is the first time I'm filling this for Home Assistant related stuff, so excuse me if there is not enough info.

Luskendeilder, 2025-06-05:
@lersveen I have 3 and can test. 
Have you seen this?
https://github.com/sprut/Hub/issues/3932

Are you working on dimming?
[image]

lersveen, 2025-06-06:
> @lersveen I have 3 and can test. 
> Have you seen this?
> https://github.com/sprut/Hub/issues/3932
> 
> Are you working on dimming?
> [image]

Great! Will get around to testing further this weekend myself.

Have not seen that, looks like it has a lot of useful info to make this quirk full featured!

Oh, and dimming is absolutely first priority. Hopefully works as soon as I get the quirk to be applied.

Luskendeilder, 2025-06-07:
@lersveen Cool, let me know when its updated so i can test it before release:)

Todilo, 2025-06-27:
What is the status of this?

lersveen, 2025-07-01:
I haven't gotten around to work more on this, thankful for any contributions.

closed 2025-08-24 by TheJulianJES

pizzaisdavid, 2025-08-27:
Noob question, but since the improved user interface was merged, will the changes be home the default home assistant eventually?

How to know roughly when? Will I need to reset my device or will the thing just change after updating one date?