zigpy/zha-device-handlers #4039: [Device Support Request] Vimar 14593 Mains_Power_Outlet_v1.0

Stefano0042, 2025-05-01:
### Problem description

It exposes standard cluster structure, but during pairing mode sensor active_power as part of Electrical Measurement cluster 0x0b04 is not detected, it will appear spontaneously few hours late, but the update rate is still problematic. 
Was ok on HA core 2025.2.x / python ver. 12.2
Currently: HA core 2025.4.4 / python ver. 13.3

### Solution description

Indeed there are not state_changed event related on the bus.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4660,
    "maximum_buffer_size": 108,
    "maximum_incoming_transfer_size": 127,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 127,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "10": {
      "profile_id": "0x0104",
      "device_type": "0x0009",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0b04"
      ],
      "output_clusters": []
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "Vimar",
  "model": "Mains_Power_Outlet_v1.0",
  "class": "zigpy.device.Device"
}

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

### Additional information

No error present in the log.
No custom quirks.

piki1979, 2025-06-12:
Hi,
I have the same problem. Just updated to 2025.6.0 and reconfigured device.
I see active_power with green mark in reconfiguration details but no identity is created.

[image]

Maybe I miss something.

Thank you.

mattiazeni, 2025-08-30:
I think the issue might be that the active_power value is in endpoint 10 on the Vimar device, instead of endpoint 1 as other devices.

[image]
[image]

puddly, 2025-08-30:
Can you attach diagnositics JSON for the device?

Stefano0042, 2025-10-02:
@puddly hello might you kindly change the label to "bug"?

Stefano0042, 2025-10-04:
From diagnostic:

"id": "0x050b",
"name": "active_power",
"zcl_type": "int16",
"value": 33

but, it is no longer detected as a sensor.