zigpy/zha-device-handlers #4040: [BUG] Hue Wall switch module RDM001 quirk not loaded

tundratier, 2025-05-01:
### Bug description

I’m having trouble with the wall swithces, as well. I have the old RDM001, and it does not load the quirk when pairing the device. So I don’t have the options available to change the button modes. It even says so within the log file:

```
2025-04-27 13:06:27.169 DEBUG (MainThread) [zha] (ZHAGatewayProxy) handling event protocol for event: RawDeviceInitializedEvent(device_info=RawDeviceInitializedDeviceInfo(ieee=00:17:88:01:0c:c2:6f:d6, nwk=0xD9BE, pairing_status=<DevicePairingStatus.INTERVIEW_COMPLETE: 2>, model=‘RDM001’, manufacturer=‘Signify Netherlands B.V.’, signature={‘manufacturer’: ‘Signify Netherlands B.V.’, ‘model’: ‘RDM001’, ‘node_desc’: {‘logical_type’: <LogicalType.EndDevice: 2>, ‘complex_descriptor_available’: 0, ‘user_descriptor_available’: 0, ‘reserved’: 0, ‘aps_flags’: 0, ‘frequency_band’: <FrequencyBand.Freq2400MHz: 8>, ‘mac_capability_flags’: <MACCapabilityFlags.AllocateAddress: 128>, ‘manufacturer_code’: 4107, ‘maximum_buffer_size’: 82, ‘maximum_incoming_transfer_size’: 128, ‘server_mask’: 11264, ‘maximum_outgoing_transfer_size’: 128, ‘descriptor_capability_field’: <DescriptorCapability.NONE: 0>}, ‘endpoints’: {1: {‘profile_id’: 260, ‘device_type’: <DeviceType.NON_COLOR_SCENE_CONTROLLER: 2096>, ‘input_clusters’: [0, 1, 3, 64512], ‘output_clusters’: [25, 0, 3, 4, 6, 8]}}}), event_type=‘zha_gateway_message’, event=‘raw_device_initialized’)
2025-04-27 13:06:27.169 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for Signify Netherlands B.V. RDM001 (00:17:88:01:0c:c2:6f:d6)
2025-04-27 13:06:27.170 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class ‘zhaquirks.philips.wall_switch.PhilipsWallSwitch’>
2025-04-27 13:06:27.170 DEBUG (MainThread) [zigpy.quirks] Fail because device_type mismatch on at least one endpoint
```

My HA Versions are:

Core: 2025.4.4
Supervisor: 2025.04.0
Operating System: 15.2
Frontend: 20250411.0

I tried to replace the quirk file with the latest from the Github Link posted above but the contents are identical. It seems to use the latest version.

### Steps to reproduce

1. Add new device through ZHA
2. Set wall switch module into pairing mode
3. Complete pairing
4. Wall Switch will be detected as generic zigbee device without any manufacturer-specific options available

### Expected behavior

The quirk for the wall switch module is loaded to enable setting manufacturer-specific options.

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4107,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 128,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 128,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0830",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0xfc00"
      ],
      "output_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0008",
        "0x0019"
      ]
    }
  },
  "manufacturer": "Signify Netherlands B.V.",
  "model": "RDM001",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

tundratier, 2025-05-02:
Looking into https://github.com/zigpy/zha-device-handlers/blob/c2f64fd0a34fbac677bdac7b1a5fe069c69dd0a5/zhaquirks/philips/wall_switch.py and the signature of my wall switch module, it seems that it is identifying itself as an RDM001, while using the clusters of the RDM004. Can anyone more involved with the zigpy quirk code validate this theory?
The WSM was connected to a Hue bridge until recently. Maybe an OTA update updated the clusters of the RDM001 switch to be consistent with the newer RDM004?

RalfCGM, 2025-05-23:
Most likely related to: https://github.com/zigpy/zha-device-handlers/issues/4053

I don't know quirks very well, but my guestimation is that our rdm001 are a premature version, in need for an own qustom quirk. Device signature as far as understand is a little different that other quirks God rdm001 and rdm004

But can't interpret the data enough to confirm.

RalfCGM, 2025-05-23:
Did anyone try using the quirk from: https://github.com/zigpy/zha-device-handlers/issues/3830?

RalfCGM, 2025-05-23:
Quirk from #3830 is working for me.

closed 2025-08-24 by TheJulianJES