zigpy/zha-device-handlers #4048: [Device Support Request] Tuya mmWave 24Ghz radar, _TZE204_lbbg34rj TS0601

fweffort, 2025-05-06:
### Problem description

Can we please have support for this one?
Have been replacing applies_to and rebooting for a while now.... 

### Solution description

Please

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_lbbg34rj",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

```

[... 671 lines cut ...]
</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

puddly, 2025-05-06:
> Have been replacing applies_to and rebooting for a while now....

What quirk are you using?

fweffort, 2025-05-06:
none for now... couldnt make it work.... 
I have been using quirks for 3 days tho.... I may be doing something really wrong....

fweffort, 2025-05-06:
> > Have been replacing applies_to and rebooting for a while now....
> 
> What quirk are you using?

none for now... couldnt make it work....
I have been using quirks for 3 days tho.... I may be doing something really wrong....