zigpy/zha-device-handlers #4051: [Device Support Request] Tuya TS0601 _TZE20C_zka46xbw

dwhitnall, 2025-05-08:
### Problem description

I've managed to get my fist Zigbee device and connected to my HA.
It's the Tuya TS0601 Garage door unit, after some searching I came across the one that is already here.
I tried to use that but since the signature was different it wasn't recognized.  I added the signiture to the quirks file and tried again but now face this error. 
DEBUG (MainThread) [zigpy.quirks] Fail because device_type mismatch on at least one endpoint
Since this is my first attempt of anything zigbee and I'm struggling to understand the nuances of the quirks file I'm kinda stumbling in the dark.
I'm hoping it's an easy simple thing.

### Solution description

Updated quirks fil

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0402",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x0500",
        "0x1000",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
[... 1085 lines cut ...]
### Logs

<details><summary>Logs</summary>

```python

2025-05-08 19:16:24.616 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZE20C_zka46xbw TS0601 (a4:c1:38:47:04:c1:e8:e0)
2025-05-08 19:16:24.616 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'ts0601_garage.TuyaGarageSwitchTO'>
2025-05-08 19:16:24.616 DEBUG (MainThread) [zigpy.quirks] Fail because device_type mismatch on at least one endpoint

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

Znubixu, 2025-08-27:
I also have the same device And I can't find a way to make it work in Home Assistant. Tried to copy the code from another device but it doesn't pick the quirk when added to HA. It defaults to the standard and despite the fact that there is a function to make it work the device itself doesn't respond.

dwhitnall, 2025-09-07:
> I also have the same device And I can't find a way to make it work in Home Assistant. Tried to copy the code from another device but it doesn't pick the quirk when added to HA. It defaults to the [...]

Znubixu, sorry for the late reply I've been working away.  I ended up creating my own custom quirk that got it working reliably.
I put the following in my zhaquirks directory.

from zhaquirks.tuya.builder import TuyaQuirkBuilder

(
    TuyaQuirkBuilder("_TZE20C_zka46xbw", "TS0601")
    .applies_to("_TZE20C_zka46xbw", "TS0601")
    .tuya_onoff(dp_id=1)
    .tuya_binary_sensor(
        dp_id=3,
        attribute_name="Sensor_State",
        translation_key="Sensor_State",
        fallback_name="Sensor State",
    )
    .skip_configuration()
    .add_to_registry()
)

I do hope it works for others as well it was an interesting learning curve which has pretty much pushed me into getting more zigbee devices.

Znubixu, 2025-09-07:
Hi dwhitnall, thank you vey much for your reply. I've created a new a file in zhaquirks directory as instructed. Removed the device, restarted Home assistant and added the device again. It picked the new file and it's now working.