zigpy/zha-device-handlers #4059: [Device Support Request] TS0601 _TZE284_libht6ua Cover motor

ZHonry, 2025-05-10:
### Problem description

The blind motor is detected but no controls are present.

### Solution description

I would like it to have cover controls as per my other blind motors.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4098,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xed00",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE284_libht6ua",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[zha-01JTTFK4TDDMWV5QC9AFMYSXPM-_TZE284_libht6ua TS0601-b6de1f8b259181af418a7e461e138594 (1).json](https://github.com/user-attachments/files/20138178/zha-01JTTFK4TDDMWV5QC9AFMYSXPM-_TZE284_libht6ua.TS0601-b6de1f8b259181af418a7e461e138594.1.json)

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[log.log](https://github.com/user-attachments/files/20138182/log.log)

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

caddish, 2025-06-03:
> I had Claude help me get this working as a qwerk. I have this [gist](https://gist.github.com/rderewianko/0084e2b5a3cc8c00f6dcc42dcdcfbbc4)

This works as a custom querks ! But I can't figure out how to calibrate the limits

rderewianko, 2025-06-04:
What I found is that some of the instructions for the moes ones worked for these
https://media.home2link.nl/Moes/ZC-LS.pdf

What I did, was hit up, let the controller get to its top position, set the blind manually to the position i wanted, then hit both buttons, let the blind go down, hit the stop button that it would set the bottom.

caddish, 2025-06-04:
I also managed to calibrate it by pressing left and right on the remote for 3sec then clicking open and stop when satisfied

Sadly the control is messed up in zha.
Sometime clicking up or down does nothing but clicking stop will make it start. And during operation I get multiple like 10x "was opened, was closed" 

I have some basic programming knowledge so I tried playing with the querk but I just can't figure out what's what

Davej817, 2025-06-04:
I also tried to get Claude to solve the issue and got the following code. The stop and up buttons are swapped so that they function correctly. I haven't yet got the calibration to work, but it is included in the quirk.

`"""Tuya blind controller quirk for _TZE284_libht6ua TS0601 - Enhanced ZCL Integration."""
import logging
from typing import Any, Dict

from zigpy.profiles import zha
from zigpy.zcl import foundation
from zigpy.zcl.clusters.closures import WindowCovering
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaWindowCover,
    TuyaManufacturerWindowCover,
    TuyaWindowCoverControl,
    TuyaPowerConfigurationCluster,
)

_LOGGER = logging.getLogger(__name__)

class TuyaBlindControllerManufCluster(TuyaManufacturerWindowCover):
    """Enhanced manufacturer cluster that coordinates with ZCL attributes."""
    
    def __init__(self, *args, **kwargs):
        """Initialize the manufacturer cluster."""
        super().__init__(*args, **kwargs)
        self._last_stable_position = None
        self._position_samples = []
        self._ignore_until = 0
        self._calibration_mode = False
        self._known_data_points = {}
        
    async def bind(self):
        """Bind and set up initial configuration."""
        result = await super().bind()
        
        # Try to read the current ZCL position attributes
        try:
            window_covering = self.endpoint.window_covering
            if window_covering:
                # Read current position from ZCL attributes
                current_pos = await window_covering.read_attributes(['current_position_lift_percentage'])
                if current_pos and 'current_position_lift_percentage' in current_pos[0]:
                    pos_value = current_pos[0]['current_position_lift_percentage']
                    if pos_value is not None and 0 <= pos_value <= 100:
                        self._last_stable_position = pos_value
                        _LOGGER.info("Initial position read from ZCL: %s%%", pos_value)
        except Exception as e:
            _LOGGER.debug("Could not read initial ZCL position: %s", e)
            
        return result
    
    async def command(
        self,
        command_id: int,
        *args,
        manufacturer: int | None = None,
        expect_reply: bool = True,
        tsn: int | None = None,
        **kwargs,
    ):
        """Override command to manage position reporting during movements."""
[... 405 lines cut ...]
#      command_type: server
#      params: {}
#
# 2. Set calibration time:
#    Service: zha.issue_zigbee_cluster_command
#    Data:
#      ieee: [your device IEEE address]
#      endpoint_id: 1
#      cluster_id: 61184
#      cluster_type: in  
#      command: set_calibration_time
#      command_type: server
#      params:
#        seconds: 30
#
# 4. Monitor the Home Assistant logs for "POTENTIAL CALIBRATION DP" messages
#    to identify which data points your specific device uses.
#
# 5. The device may need to be manually moved to full open/close positions
#    during the calibration process.
#
# 6. Try different DP numbers if the first attempts don't work:
#    - Calibration time: DP 7, 8, 9, 10
#    - Motor speed: DP 5, 6, 14, 15  
#    - Calibration trigger: DP 16, 17, 18, 101`

Davej817, 2025-06-04:
Sorry about the formatting. The comments all changed after I clicked 'comment'

caddish, 2025-06-04:
Great, can you try pasting the querks again with correct formatting ? 🙏🙏🙏🙏

rderewianko, 2025-06-04:
Heh, i figured out why it was reversed, cause when you hit both buttons at the same time briefly, for some reason on the device itself it reverses the buttons.

Here's what i know.
To conect to zigbee, power on, press the power button two times rapidly.

To set the limits on the device i've done this, however i haven't gotten this to be consistent yet.
1. Start at the top
2. press both up and down
3. Wait to go down, then press stop
Test

Davej817, 2025-06-04:
> Heh, i figured out why it was reversed, cause when you hit both buttons at the same time briefly, for some reason on the device itself it reverses the buttons.
>
> Here's what i know. To conect to [...]

Well done on finding the calibration.

rderewianko, 2025-06-04:
Okay, I had a bit more time to play with my qwerk.

I've got it showing blind open percentage.. although its not 100% correct, its close enough.

I've got it looking at firmware

https://gist.github.com/rderewianko/db23d7a4dea1784b5fdddb0c7531c9e4

rderewianko, 2025-06-08:
Testing this

If i use: 
``` 
service: cover.set_cover_position
  target:
    entity_id: cover.your_actual_entity_name
  data:
    position: 75

e.g.

action: cover.set_cover_position
target:
  device_id: 1e95e6fd4eadf1c2a6eee5126acfd7ae
data:
  position: 100

```
That works

However using the issue command to zigbee fails. Which I don't feel like troubleshooting when the first command works

```
  service: zha.issue_zigbee_cluster_command
  data:
    ieee: "fc:4d:6a:ff:fe:63:c4:c7"
    endpoint_id: 1
    cluster_id: 258
    cluster_type: in
    command: 5
    command_type: server
    params:
      lift_percentage: 75
```

Davej817, 2025-06-09:
Can I ask, how you found your id? Im using the following but nothing happens, Your ID seems to be a string of characters. 
`action: cover.set_cover_position
target:
  device_id: cover.tze284_libht6ua_ts0601
data:
  position: 100`

Davej817, 2025-06-09:
Using GUI worked. Thanks

caddish, 2025-06-09:
FYI I ended up using the base tuya cover script, only editing the input cluster based on my signature 

Then only using cover "set position" 100 or 0 to open or close 

On my dashboard I have an open and close button with a tap action set to cover "set position" 100 or 0

And not using open/close/stop from the control since it's not reliable

[image]

```
show_name: true
show_icon: true
type: button
entity: light.bulb1_light
tap_action:
  action: perform-action
  perform_action: cover.set_cover_position
  target:
    entity_id: cover.blind
  data:
    position: 100
```

```
"""Tuya based cover and blinds."""

from zigpy.profiles import zha
from zigpy.zcl.clusters.general import Basic, Groups, Identify, OnOff, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaManufacturerWindowCover,
    TuyaManufCluster,
    TuyaWindowCover,
    TuyaWindowCoverControl,
)
from zigpy.zcl.clusters.closures import WindowCovering

class TuyaZemismartSmartCover0601_23(TuyaWindowCover):
    """Tuya Zemismart blind cover motor."""

    signature = {
        #  "node_descriptor": {
        #    "logical_type": 2,
        #    "complex_descriptor_available": 0,
        #    "user_descriptor_available": 0,
        #    "reserved": 0,
        #    "aps_flags": 0,
        #    "frequency_band": 8,
        #    "mac_capability_flags": 128,
        #    "manufacturer_code": 4098,
        #    "maximum_buffer_size": 82,
        #    "maximum_incoming_transfer_size": 82,
        #    "server_mask": 11264,
        #    "maximum_outgoing_transfer_size": 82,
        #    "descriptor_capability_field": 0
        #  },
        #  "endpoints": {
        #    "1": {
        #      "profile_id": "0x0104",
        #      "device_type": "0x0051",
        #      "input_clusters": [
        #        "0x0000",
        #        "0x0004",
        #        "0x0005",
        #        "0xed00",
        #        "0xef00"
        #      ],
        #      "output_clusters": [
        #        "0x000a",
        #        "0x0019"
        #      ]
        #    }
        #  },
        #  "manufacturer": "_TZE284_libht6ua",
        #  "model": "TS0601",
        #  "class": "zigpy.device.Device"
        MODELS_INFO: [
            ("_TZE284_libht6ua", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: 0x0104,
                DEVICE_TYPE: 0x0051,
                INPUT_CLUSTERS: [
                    0x0000,  # Basic
                    0x0004,  # Groups
                    0x0005,  # Scenes
                    0xed00,  # Tuya manufacturer cluster
                    0xef00,  # Additional Tuya cluster
                ],
                OUTPUT_CLUSTERS: [
                    0x000a,  # Time
                    0x0019,  # OTA
                ],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerWindowCover,
                    TuyaWindowCoverControl,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            }
        }
    }

```

rderewianko, 2025-09-26:
When you press the physical buttons does it move to the top/bottom position?

muratozbayraktar, 2025-09-26:
> When you press the physical buttons does it move to the top/bottom position?

Same reaction, same way.