zigpy/zha-device-handlers #4062: [Device Support Request] TS0041 _TZ3000_8rppvwda Tuya 1 smart remote button

D4VID0x2, 2025-05-11:
### Problem description

This model of the smart button has a slightly different signature and the provided ts0041.py quirk doesn't bind to it.

### Solution description

Add a new entry to the ts0041.py quirk

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4742,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0020"
      ],
      "output_clusters": [
        "0x0004",
        "0x0006",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_8rppvwda",
  "model": "TS0041",
  "class": "zigpy.device.Device"
}

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

"""Tuya 1 Button Remote."""

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, Groups, Identify, OnOff, Ota, PollControl, PowerConfiguration

from zhaquirks.const import (
    BUTTON_1,
    COMMAND,
    DEVICE_TYPE,
    DOUBLE_PRESS,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    LONG_PRESS,
    MODEL,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SHORT_PRESS,
)
from zhaquirks.tuya import (
    TuyaNoBindPowerConfigurationCluster,
    TuyaSmartRemoteOnOffCluster,
)

class TuyaSmartRemote0041PollControl(CustomDevice):
    """Tuya 1-button remote device with poll control"""

    signature = {
        MODEL: "TS0041",
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    PollControl.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Groups.cluster_id,
                    OnOff.cluster_id,
                    Ota.cluster_id,
                ],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.REMOTE_CONTROL,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaNoBindPowerConfigurationCluster,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    PollControl.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Groups.cluster_id,
                    TuyaSmartRemoteOnOffCluster,
                    Ota.cluster_id,
                ],
            },
        },
    }

    device_automation_triggers = {
        (SHORT_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: SHORT_PRESS},
        (LONG_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: LONG_PRESS},
        (DOUBLE_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: DOUBLE_PRESS},
    }

```

</details>

### Additional information

Two events are captured on each button press:
```yaml
  command: press_type
  args:
    - 0
  params:
    press_type: 0
```
and
```yaml
  command: remote_button_short_press
  args: []
  params: {}
```
where arg/type 0 is short, 1 is double and 2 is long.
Either event can be used to control something but be aware that the other event is also captured.

seba-home, 2025-05-20:
thx, works perfect!

dinc5150, 2025-06-04:
Life saver! Thanks worked great!

MattWestb, 2026-01-17:
Then you need doing one new class with that cluster configs on the endpoints for getting the quirk being laded.
Its on  very large PR for converting most quirk to version 2 and then the endpoint / cluster settings is not needed only the device ID and manufacture so being more easy to fix.