zigpy/zha-device-handlers #4065: [BUG]

dontinelli, 2025-05-12:
### Bug description

I have some Aqara T1 (with neutral) which for the moment basically work fine. However, I noticed that since 2 May the value for energy consumption of one switch is oscillating between the effective value and a (fixed) lower value (cf. below screenshot):

[image]

The duration the lower value is set is not always the same:

[image]

This odd behavior only applies to one of my T1 switches. Neither the restart of HA nor the reconfiguration of the device over the UI have changed anything.

### Steps to reproduce

Review entity history

### Expected behavior

Value is constantly increasing

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4447,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009",
        "0x000a",
        "0x0702",
        "0x0b04",
        "0xfcc0"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "21": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x000c"
      ],
      "output_clusters": []
    },
    "31": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x000c"
      ],
      "output_clusters": []
    },
    "41": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0012"
      ],
      "output_clusters": []
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.switch.n0agl1",
  "class": "zhaquirks.xiaomi.aqara.switch_t1.SwitchT1Alt3"
}

```

</details>

### Diagnostic information

[zha-e5966a6cb73290616e4ffa5201c869cf-LUMI lumi.switch.n0agl1-160d438d0419fc1aa08ff38e982f540e.json](https://github.com/user-attachments/files/20159964/zha-e5966a6cb73290616e4ffa5201c869cf-LUMI.lumi.switch.n0agl1-160d438d0419fc1aa08ff38e982f540e.json)

<details><summary>Diagnostic information</summary>

</details>

### Logs

The only log entry I found is related to the fact that the value is not strictly increasing.

<details><summary>Logs</summary>

```python

Entity sensor.keller_deckenlampe_energy from integration zha has state class total_increasing, but its state is not strictly increasing. Triggered by state 41.15 (42.26) with last_updated set to 2025-05-10T09:27:00.026109+00:00. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+zha%22

```

</details>

andreibsk, 2025-05-20:
Looks similar to what I noticed here:
https://github.com/zigpy/zha-device-handlers/pull/3653#issuecomment-2566506603

dontinelli, 2025-05-20:
Yes, indeed.