zigpy/zha-device-handlers #4066: [Device Support Request] TS00601 _TZE284_a2xewxoo Tuya 2 Gang Wall Switch

codyli2002hk, 2025-05-12:
### Problem description

This model of the 2gang wall switch has a slightly different signature and I tryed this ts0601_switch.py quirk doesn't bind to it.

### Solution description

Please create device support for this model with all the other available data fields, also change from light switch to normal switch.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xed00",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE284_a2xewxoo",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

puddly, 2025-05-12:
Can you upload the diagnostic information for the device?

codyli2002hk, 2025-05-13:
Yes, I had upload the diagnostic information.

Thank you @puddly

closed 2025-11-18 by github-actions[bot]