zigpy/zha-device-handlers #4067: [Device Support Request] Tuya TS0210 vibration sensor (_TZ3210_kjafhwd2)

joaocpuga, 2025-05-13:
### Problem description

Hello, and thank you for the excellent work on this project!

I'm reaching out to request support for a Tuya vibration sensor, model TS0210, manufactured by _TZ3210_kjafhwd2.

There is already a quirk available for the TS0210 device in the repository, with support for multiple manufacturer IDs and different class signatures. However, my device is not being matched by any of the existing definitions, and as a result, ZHA does not apply the quirk.

I am using Home Assistant OS (version 2025.5.1) with the ZHA integration. 

### Solution description

I kindly ask for a new class to support this additional variant of the vibration sensor, including the appropriate clusters so that it can be correctly recognized.

I have other quirks working properly, but I’m not familiar enough with this system to confidently identify what needs to be changed in the signature to match my specific device.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

[Paste the device signature here]

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2025.5.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.13.3",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/Sao_Paulo",
    "os_name": "Linux",
    "os_version": "6.12.23-haos",
    "supervisor": "2025.05.0",
    "host_os": "Home Assistant OS 15.2",
    "docker_version": "28.0.4",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "documentation": "https://hacs.xyz/docs/use/",
      "version": "2.0.5",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "tapo": {
[... 1619 lines cut ...]
</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

joaocpuga, 2025-05-13:
I’ve added my draft Quirk to the original post by editing it. I still don’t have vibration events working on the entity with this Quirk, but getting the device recognized by signature was a first step. I’ll keep testing some options on my side and remain open to any assistance.

inigortega, 2025-11-15:
Any upgrades to get working on zha and to show battery level? Thanks in advance.