zigpy/zha-device-handlers #4072: [Device Support Request] - TS0601 by _TZE200_gkfbdvyx

albertjh, 2025-05-18:
### Problem description

I have a device with model TS0601 by _TZE200_gkfbdvyx. After pairing it with Home Assistant using ZHA, it does not expose any sensors or entities (no occupancy, illuminance, or other sensors are available). Only ZHA events are fired, but I cannot use the device in automations or see sensor values.

Device details:

Model: TS0601
Manufacturer: _TZE200_gkfbdvyx
Connected via: Generic Zigbee Coordinator (EZSP)
ZHA Integration: Zigbee Home Automation
Home Assistant Version: 
 - Core - 2025.5.2
 - Supervisor 2025.05.1
 - Operating System 15.2
 - Frontend 20250516.0

Screenshot:
[image]

Expected behavior
The device should expose its sensors and/or actuators (e.g. occupancy, presence, illuminance, or other functionalities), similar to other TS0601-based devices. Currently, none appear in Home Assistant.

What I’ve tried

Restarted Home Assistant and re-paired the device.
Checked for custom quirks in the ZHA Device Handlers (similar to #2510 and related issues).
No sensors/entities appear for this device.
Device signature
Please let me know if you need the Zigbee device signature or further diagnostic info; I can provide logs or run any required commands.

Additional context
Related issues:

[#2510](https://github.com/zigpy/zha-device-handlers/issues/2510) (other TS0601 mmWave sensors)
Thank you for your help!

### Solution description

I would like support for the TS0601 device by _TZE200_gkfbdvyx in ZHA.
Currently, after pairing, the device does not expose any sensors or entities in Home Assistant.
Ideally, the solution would provide a custom quirk or device handler so that all the sensors or controls supported by this device are available in Home Assistant (for example, occupancy or presence detection, illuminance, or any other features this device may offer).
If possible, please add full support so that the device works like similar TS0601-based sensors, and its values can be used in automations and the Home Assistant UI.

### Screenshots/Video

[image]
[image]
[image]

### Device signature

IEEE: e4:98:bb:fa:e7:34:95:0d
Nwk: 0x7e1c
Device Type: Router
LQI: 180
RSSI: -66
Visto por última vez: 2025-05-18T11:05:46
Fuente de alimentación: Mains

### Diagnostic information

[zha-01JVFBF3A4G8107ZD3P72EJA6N-_TZE200_gkfbdvyx TS0601-afd2e6b7f3e9494b0c46a0cf88f7883a.json](https://github.com/user-attachments/files/20272059/zha-01JVFBF3A4G8107ZD3P72EJA6N-_TZE200_gkfbdvyx.TS0601-afd2e6b7f3e9494b0c46a0cf88f7883a.json)

albertjh, 2025-05-19:
I've made it work (at least the sensor presence) and add a few params, however, when I change the values it does not store the set up value and after a few seconds it returns to the original value. Also, the luminance value does not update as well. But here you have it:

```python
from typing import Dict, List, Optional, Union
from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import (
    AnalogOutput,
    Basic,
    Groups,
    Scenes,
    Time,
    Ota,
    Identify,
)
from zigpy.zcl.clusters.measurement import (
    OccupancySensing,
    IlluminanceMeasurement,
)
from zhaquirks.const import (
    MODELS_INFO,
    INPUT_CLUSTERS,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    DEVICE_TYPE,
    ENDPOINTS,
)
from zhaquirks.tuya import TuyaLocalCluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaMCUCluster,
)

class TuyaAnalogOutput(TuyaLocalCluster, AnalogOutput):
    """Custom AnalogOutput cluster with workarounds for Tuya quirks."""
    def read_attributes(self, attributes, allow_cache=False, only_cache=False, 
                       manufacturer=None):
        """Override to make present_value less likely to be marked unsupported."""
        if 0x0055 in attributes:  # present_value ID
            # Ensure we have a valid value to return
            if self._attr_cache.get(0x0055) is None:
                self._update_attribute(0x0055, self.attributes_by_name["min_present_value"].value)
        
        return super().read_attributes(attributes, allow_cache, only_cache, manufacturer)

class TuyaMmwRadarSensitivity(TuyaAnalogOutput):
    """Sensitivity control for MMW radar."""
    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)
        self._update_attribute(self.attributes_by_name["description"].id, "Motion sensitivity")
        self._update_attribute(self.attributes_by_name["min_present_value"].id, 1)
        self._update_attribute(self.attributes_by_name["max_present_value"].id, 9)
        self._update_attribute(self.attributes_by_name["resolution"].id, 1)
        self._update_attribute(self.attributes_by_name["engineering_units"].id, 0)
        # Establish a default value to avoid "unsupported" status
        self._update_attribute(self.attributes_by_name["present_value"].id, 5)

class TuyaMmwRadarMinRange(TuyaAnalogOutput):
    """Minimum range control for MMW radar."""
    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)
        self._update_attribute(self.attributes_by_name["description"].id, "Minimum range")
        self._update_attribute(self.attributes_by_name["min_present_value"].id, 0.0)
        self._update_attribute(self.attributes_by_name["max_present_value"].id, 9.5)
        self._update_attribute(self.attributes_by_name["resolution"].id, 0.1)
        self._update_attribute(self.attributes_by_name["engineering_units"].id, 31)
        # Establish a default value to avoid "unsupported" status
        self._update_attribute(self.attributes_by_name["present_value"].id, 0.0)

[... 214 lines cut ...]
            5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [TuyaMmwRadarSensitivity],
                OUTPUT_CLUSTERS: [],
            },
            6: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [TuyaMmwRadarDetectionDelay],
                OUTPUT_CLUSTERS: [],
            },
            7: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.LIGHT_SENSOR,
                INPUT_CLUSTERS: [TuyaIlluminanceMeasurement],
                OUTPUT_CLUSTERS: [],
            },
        }
    }
```

[image]

Anyone can let me know how to properly set up the values and store them and make work the Iluminance measurement?

albertjh, 2025-05-20:
@therealcmj any chance you could help me here? I've shown a few pull requests and seems to be quite similar to mine but I cannot make work the luminance and store the settings. Now they keep stored but when I reboot HA they set up to the previous values.

C00kie-M0de, 2025-06-05:
I have the same device and its not exposing any sensors, hopefully we can get this resolved soon.

C00kie-M0de, 2025-06-11:
> At my place works a quirk from DelBierzo from the link: [#3584 (comment)](https://github.com/zigpy/zha-device-handlers/issues/3584#issuecomment-2848253135) You just need to change the entry: **TuyaQuirkBuilder("_TZE200_ya4ft0w4", "TS0601")** to **TuyaQuirkBuilder("_TZE200_gkfbdvyx", "TS0601")**

That quirk is working correct for now, need to test it out some more, Thank you

rrooggiieerr, 2025-06-15:
> At my place works a quirk from DelBierzo from the link: [#3584 (comment)](https://github.com/zigpy/zha-device-handlers/issues/3584#issuecomment-2848253135) You just need to change the entry: **TuyaQuirkBuilder("_TZE200_ya4ft0w4", "TS0601")** to **TuyaQuirkBuilder("_TZE200_gkfbdvyx", "TS0601")**

I just received this device and this quirk seems to work. What is the _Distance switch_ for?

[image]

proasnet, 2025-07-08:
Thank you, working for me for the first use. I will test it for few days.

rrooggiieerr, 2025-11-05:
TS0601 by _TZE200_gkfbdvyx is now working with default ZHA without a custom quirk, so I think this issue can be closed