zigpy/zha-device-handlers #4074: [Device Support Request] Moes battery powered valve controller for water and gas

fokcuk, 2025-05-19:
### Problem description

Need support for Moes TS0601 by _TZE284_zm8zpwas that is made for Tuya Zigbee 

### Solution description

Please add support for battery powered Moes valve controller

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xed00",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE284_zm8zpwas",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[... 394 lines cut ...]
</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

fokcuk, 2025-05-19:
Also not sure about the battery reading. Have 4xAA and it's voltage is at 6v and just under 6v under load.
But id 7 is showing it at 2 or 1%

doronzuk84, 2025-05-20:
Need this as well... no entities in ZHA, thanks a lot!

agostinoga, 2025-05-25:
I'm sorry, I'm not an expert in Home Assistant. Could you explain how I can add these quirks to my installation? I didn't understand how to write the file that contains the previous code.

miguelvico-netizen, 2025-08-25:
With ZHA I did not get it working, with zigbee2mqtt it works fine!