zigpy/zha-device-handlers #4082: [Device Support Request] _TZE284_fhvpaltk Dual Water Valve SGW08

the-dreamer, 2025-05-21:
### Problem description

The dual water valve _TZE284_fhvpaltk is not supported yet. but is also working with the quirk _TZE284_eaet5qt5 as also mention here: https://github.com/Koenkk/zigbee2mqtt/issues/24927#issuecomment-2840190601 

I have edited the tuya_valve file and added the _TZE284_fhvpaltk and was working with both valves in the device. 

### Solution description

add _TZE284_fhvpaltk to the _TZE284_eaet5qt5 quirk

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

[Paste the device signature here]

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json

[Paste the diagnostic information here]

```

</details>

### Logs

<details><summary>Logs</summary>

```python

[Paste the logs here]

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

[Paste your custom quirk here]

```

</details>

gorstj, 2025-05-23:
I have the same device (SGW08) and using this quirk modification worked for me - both valve confirmed working

I copied the code from https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/tuya_valve.py into a new .py file in my custom-quirk directory and modified around line 364 to be the following.

```
# NEO NAS-WV03B
(
    TuyaQuirkBuilder("_TZE204_rzrrjkz2", "TS0601")
    .applies_to("_TZE204_uab532m0", "TS0601")
    .applies_to("_TZE204_z7a2jmyy", "TS0601")
    .tuya_onoff(dp_id=1)
    .tuya_enum(
```

felixwerner, 2025-06-10:
yes seems to be working for me too. thx

mirakels, 2025-06-11:
Also have this device (_TZE284_fhvpaltk running on Z2M 2.4.0-dev).
The device properly identified but does not act on actions.

When manually operating the device I see log messages appearing showing the corrent states but the device status in the frontend is not updated (strangely enough it is in Home assistant)....

When I operate the device from Z2M or Home assistant, the device does not act on it, as if the zigbee message is not send by Z2M, or not received by the device. 
When for instance setting the countdown, a message appears in the log but nothing happens on the device. When I then manually switch on/of one of the valves the countdown is reset to zero (or 9 or 10 actually).

Any suggestion would be appreciated...

proasnet, 2025-07-30:
@puddly It couldn't. I'm not the author. It's just a quirk copied and modified according to the instructions that I tested.

PR #4256 (2025-08-24): Add Tuya valve SGW08 `_TZE284_fhvpaltk` variant
## Proposed change

should fix #4082

## Additional information

## Device diagnostics

## Checklist

- [x] The changes are tested and work correctly
- [ ] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works
- [x] Device diagnostics data has been attached
closed 2025-08-24 by TheJulianJES